realsenseai/librealsense #5102: C++ pipeline creation crash for non Intel camera

CyanFair, 2019-10-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | Non Intel one | 
| Operating System & Version |   {Win 10 | 
| Platform                                 | PC |
| SDK Version                            | 2.22.0 | 
| Language                            |  C++                         | 
| Segment			| others | 

I want to use SR305 for my project.
In Intel RealSense Viewer I can see video stream from simple webcam(not from Intel), so I thought that I can learn some basics of API by using it(without depth, IR etc. ofcourse, just to test - will it be possible to use it for me with other libraries which I use) before buy SR305 without testing.
But API crashes my program when I just try to create pipeline without any errorcode even in debug mode. So I want to understand - is it because I use not specific camera or I just do something wrong?

And another small issue: it looks like links to files here are not working: https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md

dorodnic, 2019-10-22:
Hi @CyanFair 
Links from sample-data.md seem on-line, maybe these are being blocked by your proxy / VPN? 
Pipeline examples should not be crashing, but I don't think it will work with a 2D camera either (almost all examples are assuming some kind of 3D data)
Downloading the sample data would be better way to try the API, let me know if you are able to reach the links

CyanFair, 2019-10-22:
DNS_PROBE_FINISHED_NXDOMAIN for all these links. It's strange, I havn't issues with other internet pages.
At first I want to test just showing image takken from API from camera in my GUI application, so I don't need 3D on this step. I use qt-widgets for GUI and some libraries for postprocessing of video stream, so I want to test this chain first without using depth vision.
I wanted to download sample data first but, as I said, I cant go through links to them.

CyanFair, 2019-10-22:
About crashes:
the only thing that I use from librealsense now looks like:

in .h file:
`#include <librealsense2/rs.hpp>

class SceneModifier : public QObject
{
    Q_OBJECT
...
    rs2::pipeline *pipe;
...`

in .cpp:
`SceneModifier::SceneModifier(...)
{
...
this->pipe = new rs2::pipeline();  //I tried to use it not but pointer, but as object, the result is the same
...`

And I have crash in this case without any error report. Debugger just says:
`18:35:09: Debugging starts
18:35:09: Debugging has finished`
If I comment this string - program runs succesfully.
If I add breakpoints before this string and run debugger - the result is the same as without breakpoints. I don't understand why.

CyanFair, 2019-10-23:
@dorodnic I got sample materials from these links today, so now I can use them.
But with that I got new problem which I described here: https://github.com/IntelRealSense/librealsense/issues/5108

CyanFair, 2019-10-23:
Even this doesn't give any error message. I can't understand the reason of these crashes.
`try{
    this->pipe = new rs2::pipeline();
    cfg = new rs2::config();
    //cfg.enable_stream(RS2_STREAM_COLOR, 1920, 1080, RS2_FORMAT_BGR8, 30);
    cfg->enable_device_from_file("structured.bag");
    pipe->start(*cfg);
    }
    catch (const rs2::error & e)
    {
        std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    }
    catch (const std::exception& e)
    {
        std::cerr << e.what() << std::endl;
    }`

Just `15:37:53: The program has unexpectedly finished.
15:37:53: The process was ended forcefully.`

CyanFair, 2019-10-24:
Solved my problem.
It looks like program didn't find some dlls on launch and crashed without saying anything.
I tried to launch .exe manualy, not from the IDE and put all needed dlls into the folder. After that it launched without crashing and in the IDE too.

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

Hi @CyanFair, I am glad to hear the problem is resolved.
I will go ahead and close this ticket now unless there is any question on the same topic.
Thank you.

closed 2019-11-09 by RealSenseCustomerSupport