realsenseai/librealsense #5103: Floating/Delayed 3d model using t265 slam

vladHoloArch, 2019-10-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265} | 
| Firmware Version                   | (0.0.18.5715) | 
| Operating System & Version |   Win 10| 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | PC/Android |
| SDK Version                            |  2.23.0                       | 
| Language                            |  C#/unity                        | 
| Segment			|  AR/others                         | 

### Issue Description
Hi,
trying to build an AR Unity based app using t265's slam.
The problem I'm having is coordinating the model to the real world, I'm influencing the camera position directly with the t265's position and rotation and the 3d model looks delayed/floating.
When I'm moving the rig (t265 combined with the device built in camera) it looks like the model is trying to catch up with the image I get from the camera and swings back and forth as if the update rate Isn't quick enough which doesn't make sense.
Even while the rig is stationary the whole model 'floats' around like it's on a ship at sea..
Did anyone come across with this effect?

ogoshen, 2019-10-29:
> it looks like the model is trying to catch up with the image I get from the camera and swings back and forth as if the update rate Isn't quick enough which doesn't make sense.

Actually it does, T265 poses update the entire pose at 60Hz, and the orientation part at 200Hz (the gyro's frequency).

Also, there's the issue of when the pose is updated, not just how many FPS. 
The supplied `RsPoseStreamTransformer` updates a transform with the pose in the `Update` callback, which is too soon in a frame, because by the time rendering started, the device could have gotten a new pose.

Now the easiest to check is to turn VSync off in the quality settings, so Unity renders as fast as it can, minimizing the delay between the pose and rendering, but this can cause undesirable performance.

Better would be to use [Application.onBeforeRender](https://docs.unity3d.com/ScriptReference/Application-onBeforeRender.html) or at least [MonoBehaviour.OnPreCull](https://docs.unity3d.com/ScriptReference/MonoBehaviour.OnPreCull.html).
These are probably that last chance to update the camera's transform just before rendering starts.

Finally, there's the "motion to photon latency" - by the time the image is displayed on screen the pose is already changed... 
So the best you can do seems to be: estimate rendering time, and apply pose prediction based on linear and angular velocities. This could overshoot though, and have the opposite effect of the latency you're now experiencing.

vladHoloArch, 2019-11-04:
@ogoshen Thanks,
I've updated my question now that I have another issue, would be very grateful if you take a look...

ogoshen, 2019-11-04:
> Another problem I've got is that t265 pose does not correspond to d435 640x480 produced RGB colorframe, I know it's the intrinsic I just have now idea how to align them..

This is whole other issue, I'd appreciate if you could split it to a new issue, it might be a long discussion...

I'll just mention that you need *extrinsics* from T265 pose to D435.
For AR, see the *ARBackground* sample in Unity on how to apply intrinsics (+depth) to Unity.

Please also state if the original issue is resolved, this might need to be a PR.

vladHoloArch, 2019-11-04:
@ogoshen 
still working on it, I've applied low-pass filter to smoothen the jitterness and followed your instructions about when to update my pose, I'm using my own really basic java driver (the whole system operates on android) which makes limited to what I can do.
I've opened the new issue regarding the t265 to d435 extrinsics..

RealSenseSupport, 2020-09-21:
Thank you for highlighting the issue outlined regarding T265. We have moved our focus to our next generation of products and consequently, we will not be addressing this issue in the T265.

closed 2020-09-28 by RealSenseSupport