realsenseai/librealsense #5108: C++ compiling error

CyanFair, 2019-10-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Operating System & Version |  Win 10  | 
| Platform                                 | PC  |
| SDK Version                            | 2.22.0                  | 
| Language                            |C++                       | 
| Segment			|  others                    | 

I use qt+opencv+librealsense libraries, when I do this:
`   this->pipe = new rs2::pipeline();
    cfg = new rs2::config();
    cfg->enable_device_from_file("structured.bag");
    pipe->start(cfg);`

My includes:
`#include <QObject>
#include <Qt3DCore/QEntity>
#include <Qt3DExtras/Qt3DWindow>
#include "paintedtextureimage.h"         //another my class
#include <librealsense2/rs.hpp>
#include <opencv2/opencv.hpp>`

And this gives:
`...\Intel RealSense SDK 2.0\include\librealsense2\hpp\rs_frame.hpp:1131: error: expression cannot be used as a function
             on_frame_function(frame{ fref });
             ^~~~~~~~~~~~~~~~~`

ev-mp, 2019-10-23:
@CyanFair , you're trying to pass a pointer to an instance
>cfg = new rs2::config(); 
...
pipe->start(cfg);

while The [API expects either a reference or a callback](https://github.com/IntelRealSense/librealsense/blob/659860bb9a0eca56814c97d90a162b3cdfb2af82/include/librealsense2/hpp/rs_pipeline.hpp#L421-L442).
Since you pass in `rs2::config *` which is different from `rs2::config&` the compiler does not recognizes it and tries to match with another override for `pipe.start(Callback)` , which also fail.

You can either go for
>pipe->start(*cfg);

or
>rs2::config cfg;
cfg.enable_device_from_file("structured.bag"); 
pipe.start(cfg);

CyanFair, 2019-10-23:
@ev-mp oh, such stupid mistake from me. Thanks a lot!
And again I have this issue: https://github.com/IntelRealSense/librealsense/issues/5102
It just crashes in a moment after the start.

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

Closing this ticket as the issue should resolved. Thank you.

closed 2019-11-01 by RealSenseCustomerSupport