realsenseai/librealsense #5114: Alternative download location for firmware?

mikepurvis, 2019-10-23:
Currently librealsense downloads the latest firmware from an S3 bucket at configure time, here:

https://github.com/IntelRealSense/librealsense/blob/4f2eac248530ef75fb38b95002522aa032165638/common/fw/CMakeLists.txt#L11-L21

This behaviour can be turned off by passing `-DINTERNET_CONNECTION=0` to CMake, but it would be nice to instead be able to point to a local/elsewhere mirror for these assets. This could be as simple as setting the two lines to something like:

set(FW_URL_PREFIX "http://realsense-hw-public.s3-eu-west-1.amazonaws.com" CACHE STRING)
set(D4XX_FW_URL "${FW_URL_PREFIX}/Releases/RS4xx/FW")
set(SR3XX_FW_URL "${FW_URL_PREFIX}/Releases/SR300/FW")

So that a user would have the option of passing an alternative base URL in with `-DFW_URL_PREFIX=http://path/to/my/mirror`.

MartyG-RealSense, 2019-10-23:
Aside from the firmware download page linked to from the SDK 'Releases' page, the files are also available at the link below, which was the download location before Intel changed to the site used on the Releases page.

https://downloadcenter.intel.com/download/29078/Latest-Firmware-for-Intel-RealSense-D400-Product-Family

mikepurvis, 2019-10-23:
Oh yeah, I'm aware they can be had elsewhere. My issue is with the librealsense build needing external network access at configure time. For some CI environments this is unacceptable, so I'd like it to be configurable.

radfordi, 2019-10-23:
@mikepurvis, separating the base URL into a variable that can be overridden is reasonable.  Submit a PR.   Alternatively, these files can be placed in the build directory prior to running CMake and if their SHA1 hashes match, then they will not be downloaded.

HenryHuYu, 2019-12-07:
I m working on installing libreasense on a Jetson nano. Suffering from errors showed below. I check every possible solution and didnt work out. I tried have a proxy but it also didnt work. 

/home/drone/librealsense/common/fw-update-helper.cpp:15:10: fatal error: common/fw/D4XX_FW_Image.h

Failed to identify Internet connection
-- Info: REALSENSE_VERSION_STRING=2.25.0
-- Setting Unix configurations
-- Info: Building with CUDA requires CMake v3.8+
-- CUDA_LIBRARIES: /usr/local/cuda/include /usr/local/cuda/lib64/libcudart_static.a;-pthread;dl;/usr/lib/aarch64-linux-gnu/librt.so;/usr/local/cuda/lib64/libcusparse.so;/usr/local/cuda/lib64/libcublas.so
-- No internet connection, disabling BUILD_WITH_TM2
-- using RS2_USE_V4L2_BACKEND
-- pybind11 v2.2.1
-- GLFW 3.3 not found; using internal version
-- Using X11 for window creation
-- Could NOT find apriltag (missing: APRILTAG_INC APRILTAG_LIB) 
-- Unable to find apriltag library, skipping pose-apriltag example
-- No internet connection, disabling IMPORT_UVC_FW
-- Configuring done
-- Generating done

Could you please share a mirror link? besides,  as for mentioned, could I get it from  https://downloadcenter.intel.com/download/29078/Latest-Firmware-for-Intel-RealSense-D400-Product-Family ?
and then how can I set it and  install librealsense properly?

MartyG-RealSense, 2019-12-07:
The website JetsonHacks has a guide for installing Librealsense on the Jetson Nano.  Have you seen this guide please?

https://www.jetsonhacks.com/2019/05/16/jetson-nano-realsense-depth-camera/

If that guide does not work for you, there is an alternative method for installing Librealsense when the usual installation method does not work.

https://support.intelrealsense.com/hc/en-us/community/posts/360037657234/comments/360009799533

HenryHuYu, 2019-12-09:
hi, I finally get it done by having a robust vpn. but I got some other problems. 
Firstly, when I install librealsense-ros, it sait that the minimum of libreasense version > 2.29, but the scripts offered by the JetsonHack is 2.25, I change the script to 2.30.0 and delete the build file but it didnot work. could u help me with that (How to upgrade librealsense?)

MartyG-RealSense, 2019-12-09:
If you are using the latest version of the RealSense ROS wrapper (the latest at the time of writing this) then SDK version 2.29.0 should be used with it.  This is because development of the ROS wrapper is not synced with development of the main SDK, so the SDK version that the RealSense ROS wrapper supports can lag behind.

https://github.com/IntelRealSense/realsense-ros/releases

HenryHuYu, 2019-12-09:
Thanks for your reply. I am trying to install a SDK version > 2.29 from source(Jetson Nano). But I followed the instructions of JetsonHack and it automatically install the version 2.25 which causes trouble when install realsense-ros.

How can I rebuild the source code and install a SDK version (2.29)?  I goes to /librealsense/build and do make uninstall but showed the following error.

-- Uninstalling "/usr/local/lib/librealsense2.so.2.25.0"
CMake Error at cmake_uninstall.cmake:16 (MESSAGE):
  Problem when removing "/usr/local/lib/librealsense2.so.2.25.0"

CMakeFiles/uninstall.dir/build.make:57: recipe for target 'CMakeFiles/uninstall' failed
make[3]: *** [CMakeFiles/uninstall] Error 1
CMakeFiles/Makefile2:106: recipe for target 'CMakeFiles/uninstall.dir/all' failed
make[2]: *** [CMakeFiles/uninstall.dir/all] Error 2
CMakeFiles/Makefile2:113: recipe for target 'CMakeFiles/uninstall.dir/rule' failed
make[1]: *** [CMakeFiles/uninstall.dir/rule] Error 2
Makefile:177: recipe for target 'uninstall' failed
make: *** [uninstall] Error 2

MartyG-RealSense, 2019-12-09:
You can obtain SDK 2.29.0's source code from Intel's official SDK 'Releases' page.  If you scroll to the bottom of the page, there is a list of files titled 'Assets' where the source code can be downloaded from.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.29.0

The official documentation on uninstalling an SDK version is here:

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#uninstalling-the-packages

RealSenseCustomerSupport, 2020-03-19:
----------------------------------------------

Hi @mikepurvis aHenryHuYund Do you need further help on this issue?
If we don't hear from you in 7 days, we will close this issue.

PR #6098 (2020-03-22): Make firmware URL overrideable
Change as proposed in #5114. This permits the firmware binaries to be mirrored on-site for more restrictive CI environments that do not permit external network access at build time.

Closes: #5114
closed 2020-04-20 by RealSenseCustomerSupport