realsenseai/librealsense #5117: C# .NET Problem loading json files

KarlAbson, 2019-10-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 05.11.15.00 | 
| Operating System & Version |   {Win 10  | 
| SDK Version                            |  { 2.29 }                          | 
| Language                            |  {C# }                          | 

### Issue Description
 On loading a json file a System.Exception occurs. This appears to be the case both for json files downloaded and those created through Realsense viewer.

The full error is:

hr returned: HResult 0x8007001f: "A device connected to the system does not work."
rs2_load_json(dev:07BDB9F8, json_content:11B39ED0, content_size:3595)'

Issues with loading json files in .Net were reported back in Dec 2018 in #2854.
This job was closed as fixed?

Is a simple way of changing the advanced settings available in .Net yet? This is really needed.

Many thanks,
Karl

KarlAbson, 2019-10-24:
The code being used is as follows:

var adv = AdvancedDevice.FromDevice(pp.Device);
string path = System.IO.Directory.GetCurrentDirectory();
string full = System.IO.Path.Combine(path, "default.json");
adv.JsonConfiguration = System.IO.File.ReadAllText(full);

MartyG-RealSense, 2019-10-24:
The C# wrapper still does not have access to the Advanced Mode instructions, as far as I know.

 A C# code example for loading a json file can be found here:

https://github.com/IntelRealSense/librealsense/issues/2854#issuecomment-444803962 

In the case of this example, the filename '1.json' refers to the "Default" visual preset json.   Substitute in your own json file-name.

The easiest way to create your own json file is to define settings in the RealSense Viewer and then export them as a json using an icon on the toolbar at the top of the Viewer's options side-panel.

KarlAbson, 2019-10-24:
I would like a response from Intel regarding this. The problem of loading these files was reported over a year ago. Accessing the advanced features of the camera is necessary for professional applications.

MartyG-RealSense, 2019-10-24:
I had some recollection of this subject being discussed earlier this year on the Intel Support forum and dug out the link.  An Intel support team member advised accessing Advanced Mode in C# via a script called **NativeMethods.cs**

https://support.intelrealsense.com/hc/en-us/community/posts/360033065534-How-to-retrieve-advanced-mode-options-in-C-

KarlAbson, 2019-10-25:
I would rather investigate and fix the problem with the above code than search for a workaround at the moment. It could well be that the workaround suffers from the same problem. The following code should work:

var adv = AdvancedDevice.FromDevice(pp.Device);
string path = System.IO.Directory.GetCurrentDirectory();
string full = System.IO.Path.Combine(path, "HighResHighDensityPreset.json");
adv.JsonConfiguration = System.IO.File.ReadAllText(full);

Indeed I tried this in one of the sample projects and it works there. In my project however I get the following error message:

System.Exception: 'hr returned: HResult 0x8007001f: "A device connected to the system does not work." rs2_load_json(dev:084AA598, json_content:1101ABC0, content_size:3595)'

My project is naturally more complex as I need to be able to start / stop capture without memory leaks as well as analyse images etc. It carries things out in much the same way however in the same order.  Understanding the above error message, why, where and how it happens would help.

RealSenseCustomerSupport, 2019-11-13:
----------------------------------------------

Hi @KarlAbson,

A simple C# app with json file loading has no the issue. Do you have any librealsense log there when you run your app with this loading exception issue?
Will also further discuss internally to have a simple way to configure the advanced parameters directly in C#.

KarlAbson, 2019-11-13:
Hi @RealSenseCustomerSupport,

Thanks for the reply. I managed to fix this. The problem was related to the following case:
[C# Wrapper - samples crash on launch](https://github.com/IntelRealSense/librealsense/issues/5122)

Simply there are a number of issues with the C# wrapper. The dlls are named wrong. He cannot find the realsense2.dll as it is now called realsense2d.dll. Also the C# projects are set to AnyPC by default. There are no mention of these things in the read me so it took time to identify the problem.

MartyG-RealSense, 2021-09-11:
Adding a note that I am now the Intel RealSense team member handling this case.

closed 2023-07-18 by MartyG-RealSense