realsenseai/librealsense #5119: Memory leak C++

OneQuestion, 2019-10-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i  | 
| Firmware Version                   | 2.29 | 
| Operating System & Version |   Linux (Ubuntu 16) | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 |   |
| SDK Version                            |                          | 
| Language                            |  C++                         | 
| Segment			|                           | 

### Issue Description
Hello, I have found a possible memory leak in this simple code. I ran it for 7 hours and RAM usage went from ~95MB to ~200MB. Do I need to delete or free something?

```
#include <librealsense2/rs.hpp>
#include <iostream>
int main()
{
    rs2::pointcloud pc;
    rs2::points points;
    rs2::pipeline pipe;
    rs2::config cfg;
    cfg.enable_device_from_file("20190904_132610.bag");
    pipe.start(cfg);
    auto frames = pipe.wait_for_frames();
    auto depth = frames.get_depth_frame();
    points = pc.calculate(depth);
    auto vertices = points.get_vertices();
    auto tex_coords = points.get_texture_coordinates();
    long int i = 0;
    while (1)
    {
	i ++;
	std::cout<<i<<std::endl;
        frames = pipe.wait_for_frames();
        depth = frames.get_depth_frame();
        points = pc.calculate(depth);
        vertices = points.get_vertices();
        tex_coords = points.get_texture_coordinates();
    }
}
```

MartyG-RealSense, 2019-10-24:
There is a known issue of a memory leak in the point cloud processing block that is listed in the SDK release notes for the current version (2.29.0).

https://github.com/IntelRealSense/librealsense/issues/2860

dorodnic, 2020-07-06:
@ev-mp - if I recall correctly, this was addressed, right?

ev-mp, 2020-07-07:
> @ev-mp - if I recall correctly, this was addressed, right?

The recent fix was for the Occlusion filter.

ev-mp, 2020-07-07:
Duplicate of #2860

RealSenseSupport, 2021-08-04:
Fix is in the development branch via https://github.com/IntelRealSense/librealsense/pull/9553

closed 2021-08-04 by RealSenseSupport