realsenseai/librealsense #5120: How can I get the normal from the SDK or realsense-viewer

ss2lyf, 2019-10-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435i } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) &ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |   2.29                           | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hi, I want to get the normal of each point , how can I get the normal from the SDK or realsense-viewer, could you show me an example?The export_to_ply generate the mesh file , how can I get the raw point cloud, I mean each point from the depth map ,RGB and normal information. I think it is different from the mesh. The mesh is cacluated by the some surface algrithm. Vertices dosent equal to the raw point. Am I right? 
This may relate to my issue.
#4906

dorodnic, 2019-10-24:
Upcoming release of the RealSense Viewer is going to include an option to include normals in PLY exporter, but generally speaking, I would export PLY and calculate normals externally using MeshLab.

cdb0y511, 2019-10-25:
Hi,@dorodnic 
Compute normal of each point by meshlab or cloudcompare is pretty slow, especially when the point cloud is dense. I think some stereo algorithm ,like patchmatch, compute normal and depth simultaneously.  I guess the normal calculation is done by the VPU of D400. When can I get this function? Dose the development branch have include this function yet?
I find #4906 have been approved.
Thanks.

lramati, 2019-10-28:
As you can see by inspecting #4906 ([specifically here](https://github.com/IntelRealSense/librealsense/pull/4906/files#diff-883f598ceaa739d7473cdc28d7995e23R129)) normals are calculated in the viewer and not on the D400's VPU. This is why we recommend using Meshlab or similar as their code is likely more optimized for calculating vertex normals on large meshes

cdb0y511, 2019-10-29:
@lramati 
Thanks

RealSenseCustomerSupport, 2019-11-06:
----------------------------------------------

Please let us know if further assistance is needed for this topic.
Thank you.

closed 2019-11-15 by RealSenseCustomerSupport