realsenseai/librealsense #5121: Crash calling librealsense::environment::get_instance()

mikepurvis, 2019-10-24:
Because of various reliability issues we've experienced with Realsense, we're using a relay to control power to the USB hub they are connected to, so that we can power cycle them when they get themselves into a bad state. Unfortunately, this has itself revealed a number of other problems which occur when a camera disappears from the system while the driver is running, for example:

```
#0  std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count (__r=..., this=<optimized out>) at /usr/include/c++/5/bits/shared_ptr_base.h:824
No locals.
#1  std::__shared_ptr<librealsense::stream_interface const, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<librealsense::stream_interface> (__r=..., this=<optimized out>)
    at /usr/include/c++/5/bits/shared_ptr_base.h:949
No locals.
#2  std::shared_ptr<librealsense::stream_interface const>::shared_ptr<librealsense::stream_interface> (__r=..., this=<optimized out>) at /usr/include/c++/5/bits/shared_ptr.h:251
No locals.
#3  std::enable_shared_from_this<librealsense::stream_interface>::shared_from_this (this=<optimized out>) at /usr/include/c++/5/bits/shared_ptr.h:577
No locals.
#4  librealsense::extrinsics_graph::find_stream_profile (this=this@entry=0x7fc6fbf4dec0 <librealsense::environment::get_instance()::env>, p=...) at ../../../src/librealsense2/src/environment.cpp:90
        sp = <optimized out>
        max = <optimized out>
#5  0x00007fc6fb23a892 in librealsense::extrinsics_graph::try_fetch_extrinsics (this=0x7fc6fbf4dec0 <librealsense::environment::get_instance()::env>, from=..., to=..., extr=extr@entry=0x7fc6facef790)
    at ../../../src/librealsense2/src/environment.cpp:108
        lock = {_M_device = @0x7fc6fbf4dec0}
        from_idx = 77
        to_idx = <optimized out>
        visited = std::set with 140492004130848 elements<error reading variable: Cannot access memory at address 0x18>
#6  0x00007fc6fb27bdd7 in rs2_get_extrinsics (from=0x7fc6d838d710, to=0x7fc6d833cf68, extrin=extrin@entry=0x7fc6facef790, error=error@entry=0x7fc6facef6c8) at ../../../src/librealsense2/src/rs.cpp:986
        __FUNCTION__ = "rs2_get_extrinsics"
#7  0x00007fc728ba4c20 in rs2::stream_profile::get_extrinsics_to (to=..., this=0x7fc6facef760) at /tmp/buildd/workspace/install/librealsense2/include/librealsense2/hpp/rs_frame.hpp:152
        e = 0x0
#8  realsense2_camera::BaseRealSenseNode::calcAndPublishStaticTransform (this=0x7fc6d81d64d0, stream=..., base_profile=...) at ../../../src/realsense2_camera/src/base_realsense_node.cpp:1832
        quaternion_optical = {<tf::QuadWord> = {m_floats = {-0.5, 0.49999999999999989, -0.5, 0.50000000000000011}}, <No data fields>}
        zero_trans = {x = 0, y = 0, z = 0}
        transform_ts_ = {<ros::TimeBase<ros::Time, ros::Duration>> = {sec = 1571892256, nsec = 747410567}, <No data fields>}
        ex = {rotation = {-5.37329835e+35, 4.58364728e-41, 3.32713073e+34, 4.58378741e-41, 2.91470081e-43, 0, 8.60957776e-42, 0, -5.3731906e+35}, translation = {4.58364728e-41, 1.63258587e-14, 4.58378741e-41}}
        __PRETTY_FUNCTION__ = "virtual void realsense2_camera::BaseRealSenseNode::calcAndPublishStaticTransform(const stream_index_pair&, const rs2::stream_profile&)"
        Q = {<tf::QuadWord> = {m_floats = {7.6678988234561464e-321, 1.1857575500189917e-322, 7.5888483201215469e-321, 6.9412561285613668e-310}}, <No data fields>}
        trans = {x = -7.96888232e+14, y = 4.58364728e-41, z = -9.84750102e+14}
```
Full backtrace in https://gist.github.com/mikepurvis/2bf217a1bc821e5c36dd9d3d2081b4ce

Have also seen similar crashes in `realsense2_camera::BaseRealSenseNode::getAProfile` 

```
#5  std::shared_ptr<rs2_stream_profile>::shared_ptr (this=0x7fb2cf6769b8) at /usr/include/c++/5/bits/shared_ptr.h:107
No locals.
#6  rs2::stream_profile::stream_profile (this=0x7fb2cf6769b0) at /home/administrator/realsense_ws/devel/include/librealsense2/hpp/rs_frame.hpp:21
No locals.
#7  realsense2_camera::BaseRealSenseNode::getAProfile (this=this@entry=0x7fb2a415f8b0, stream=...) at /home/administrator/realsense_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:2115
        profiles = std::vector of length 70, capacity 128 = {{_profile = 0x7fb2a4014d30, _clone = std::shared_ptr (empty) 0x0, _index = 1, _uid = 85, _framerate = 30, _format = RS2_FORMAT_Y8, _type = RS2_STREAM_INFRARED, _default = false}, .... {_profile = 0x7fb2a4084de0, _clone = std::shared_ptr (empty) 0x0, _index = 1, _uid = 85, _framerate = 15, _format = RS2_FORMAT_Y8, _type = RS2_STREAM_INFRARED, _default = false}, {_profile = 0x7fb2a4084ea0, _clone = std::shared_ptr (empty) 0x0, _index = 1, _uid = 85, _framerate = 6, _format = RS2_FORMAT_Y8, _type = RS2_STREAM_INFRARED, _default = false}, {_profile = 0x7fb2a44dd950, _clone = std::shared_ptr (empty) 0x0, _index = 2, _uid = 86, _framerate = 6, _format = RS2_FORMAT_Y8, _type = RS2_STREAM_INFRARED, _default = false}}
#8  0x00007fb3002de61b in realsense2_camera::BaseRealSenseNode::publishStaticTransforms (this=this@entry=0x7fb2a415f8b0) at /home/administrator/realsense_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:1884
        base_profile = {_profile = 0x20656d6172662065, _clone = <error reading variable: Cannot access memory at address 0x6f6320230a6c6580>, _index = 0, _uid = 0, _framerate = 0, _format = RS2_FORMAT_ANY, _type = RS2_STREAM_ANY, _default = false}```

ev-mp, 2019-10-29:
@mikepurvis ,afaik  the `realsense2_camera` should recover from device disconnect events automatically.
So it is definitely an issue that need to be fixed. Please re-submit this report to `realsense2_camera` repo for handling.

mikepurvis, 2019-10-30:
Sure, I can open a different issue there. I wasn't sure if this error should have been handled at that layer or this one.

closed 2019-10-30 by mikepurvis