realsenseai/librealsense #5122: C# Wrapper - samples crash on launch (librealsense-2.29.0 SDK)

KarlAbson, 2019-10-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (05.11.15.00) | 
| Operating System & Version |   Win (10)   | 
| Platform                                 | PC  |
| Language                            |  { C# }                          | 

### Issue Description

I believe that there is an issue with the new SDK release in regards to the C# Wrapper and examples...I just downloaded the new 2.29.0 and the samples will compile without issue but all will suffer from various run-time errors? When they get to the first line of code requiring the dlls a run-time error will occur.

Upon inspection the dlls created during compilation appear to be named wrong? They all have a "d" in the name which was not there before...for example realsensed.dll has become realsense2d.dll. If I rename these files to what they were originally, then the program will run.

KarlAbson, 2019-10-25:
An additional point that I just found out...

Renaming the realsense2d.dll to realsense2.dll appears to solve the problem in the x64 version. It does not however in the x86 version. I cannot seem to come up with a way of making the x86 version usable.

ogoshen, 2019-11-04:
I just ran into this issue myself for the first time, the "d" postfix for debug builds was added in #3611

We could add a preprocessor directive for .NET [here](https://github.com/IntelRealSense/librealsense/blob/81d469db173dd682d3bada9bd7c7570db0f7cf76/wrappers/csharp/Intel.RealSense/NativeMethods.cs#L19), the thing is that the two configurations aren't really tied-up together... 

Aside from some form of dynamic pinvoke calls (LoadLibrary & GetProcAddress \ dlopen & dlsym) which have overhead, there doesn't seem to be a real solution...

I consider this issue related to #5069

RealSenseCustomerSupport, 2019-12-19:
----------------------------------------------

Hi @KarlAbson,

As @ogoshen has noticed, the "d" postfix was added for debug builds. User that want to utilize RealSense SDK functionality and that doesn't intend to develop/contribute code for the SDK itself should install SDK by following installation guide instructions.
Please let us know if you need any further clarifications regarding this topic.

Thank you!

KarlAbson, 2019-12-19:
@IntelRealSense Yes this has been made clear. It would be good if this was
in the release notes. On last check I couldn't find anything.

RealSenseCustomerSupport, 2019-12-19:
----------------------------------------------

@KarlAbson,
Thank you for bringing this matter to our attention. The documentation will be updated accordingly by the team responsible for this issue. Do you need any additional help with this question?

closed 2019-12-26 by RealSenseCustomerSupport