realsenseai/librealsense #5123: RealSense Viewer Saving

josephdanielchang, 2019-10-24:
#4911 # Issue Description

I've built the realsense viewer and it works. I'm looking through the code though and it seems all gui related. Where would I modify the viewer so it saves frames live as png instead of saving to rosbag?

MartyG-RealSense, 2019-10-24:
The graphics in the central window of the Viewer use OpenGL for their rendering, and the interface uses a system called imgui.

https://github.com/IntelRealSense/librealsense/issues/2450

I can imagine that saving PNGs live, if it were possible, would provide very slow performance.  Bags are the most efficient way to record live, and the bag can be converted into PNGs or another format such as csv after the recording is completed (by using a conversion tool or post-processing).  The link below has a very long and interesting discussion on the subject

https://github.com/IntelRealSense/librealsense/issues/1485

Depth can be aligned with color in the Viewer's 3D mode.  It is harder to align in the Viewer's 2D mode though.

https://forums.intel.com/s/question/0D70P0000069EWaSAM

josephdanielchang, 2019-10-24:
I see depth/color can't be aligned in the default viewer, but there must be a way to add align_to_color.process to the code so it can align. My question was where would I find that file to change the code. Same with saving as png. Despite it being slower, where would I find the code to add those to the existing viewer code?

[image]

MartyG-RealSense, 2019-10-24:
Custom editing of the Viewer is outside of my knowledge unfortunately.  One of the Intel guys on this forum should be able to advise you well.  Good luck!

RealSenseCustomerSupport, 2019-11-02:
----------------------------------------------

The Realsense SDK contains great resource/examples that demonstrate image saving (png) as well as frame align.
Please take a look at the following two examples:
https://github.com/IntelRealSense/librealsense/tree/master/examples/save-to-disk
https://github.com/IntelRealSense/librealsense/tree/master/examples/align

Please let us know if you have any other question. Thank you.

closed 2019-11-14 by RealSenseCustomerSupport