realsenseai/librealsense #5124: Configuring recorder compression mode in the pyrealsense2 wrapper

bbrucee, 2019-10-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | pyrealsense2 2.29.0.1124 | 
| Operating System & Version |   Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Platform                                 | PC |
| SDK Version                            |  pyrealsense2 2.29.0.1124  | 
| Language                            |  Python                           | 

### Issue Description
Recent pull request https://github.com/IntelRealSense/librealsense/pull/2687 allows for toggling of compression in the C++ API, will this be eventually exposed for the Python wrapper?

lramati, 2019-10-28:
Yes. I will update here when it becomes available

lramati, 2019-11-07:
This is added by #5200

RealSenseCustomerSupport, 2019-11-22:
----------------------------------------------

@bbrucee #5200 now merged to development branch. Could you please have a try and feedback? Thanks!

RealSenseCustomerSupport, 2019-12-02:
----------------------------------------------

@bbrucee Did you get chance to try #5200? Cloud you please update? Thanks!

bbrucee, 2019-12-02:
Sorry I will need some time to get around to it but when I do I will update

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

@bbrucee Any update now? Thanks!

RealSenseCustomerSupport, 2020-01-15:
----------------------------------------------

@bbrucee Any good news? Thanks!

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

@bbrucee Sorry that we didn't get response from you for weeks. Will close this issue at this point. Please feel free to create new one if you still have questions or issues. Thanks!

closed 2020-02-03 by RealSenseCustomerSupport