realsenseai/librealsense #5125: When the bag file is played back through wait_for_frames(), there are many duplicates in the framesets and the timestamp is also repeated

nevynwong, 2019-10-25:
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Operating System & Version | Linux (Ubuntu 16 | 
| Platform                                 | PC |
| Language                            |   python | 

When I use wait_for_frame() to play back the recorded bag file, I get a much larger number of frames than I did with rosbag.Careful analysis shows that timestamps and frames repeat a lot. Why?Here's my program
import pyrealsense2 as rs
import numpy as np
import matplotlib.pyplot as plt
import cv2
import argparse
import os.path
import scipy.io as scio

pipe = rs.pipeline()
config = rs.config()
config.enable_device_from_file('/plant.bag',repeat_playback=False)
profile = pipe.start(config)
playback = profile.get_device().as_playback()
playback.set_real_time(False)

#pointcloud
pc = rs.pointcloud()
points = rs.points()

colorizer = rs.colorizer()

align_to = rs.stream.color
align = rs.align(align_to)
n = 0
m = 0
while (playback.current_status() == rs.playback_status.playing) :
n += 1
print n
frames = pipe.wait_for_frames()
****(** means other processing)

lramati, 2019-10-28:
Duplicate of #5107

closed 2019-10-30 by nevynwong

nitishlatkar, 2021-05-27:
I was getting duplicate frames with wait_for_frames(). Also the Timestamp and **frame.get_frame_number()**  will give you  repeating frame number. 

I solved this issue  by calling **frames.keep()**    after **frames = pipeline.wait_for_frames()**

If you see the documentation. It says "Keep the frame, otherwise if no refernce to the frame, the frame will be released."

Using frames.keep() solved issue of repeating frames.