realsenseai/librealsense #5128: d435i and slam in unity

TheBricktop, 2019-10-25:
Unity 2019.2.2 / android & win & linux / d435i  / VR

I know that there is official ROS example of d435i using SLAM (mentioned [here](https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i))
Does this mean that d435i would work with SLAM example in unity realsense package?

MartyG-RealSense, 2019-10-25:
The Unity SLAM example is meant to be used with the T265 Tracking Camera.

TheBricktop, 2019-10-25:
that im aware of, but since there is official ROS example of d435i using SLAM (mentioned [here](https://github.com/IntelRealSense/librealsense/issues/5128#issue-512511709)), i was thinking that the other imu enabled intel camera would actually be able to provide 6dof pose.

MartyG-RealSense, 2019-10-25:
T265 has support for pose built in but the D435i does not unfortunately.  It is possible to get pose on a D435 with a workaround, but that would likely not be compatible with the Unity example.

The link below has details of the D435 workarounds for pose if you are interested.

https://forums.intel.com/s/question/0D70P000006GEZUSA4

TheBricktop, 2019-10-25:
I am, thank You very much for such quick response.

closed 2019-11-01 by RealSenseCustomerSupport