realsenseai/librealsense #5129: t265 and d435 together on jetson TX2

jungladicitta, 2019-10-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { d435 and t265 together } | 
| Firmware Version                   | latest| 
| Operating System & Version |   16.04 | 
| Kernel Version (Linux Only)    |  4.4.38-tegra                                         | 
| Platform                                 | NVIDIA Jetson TX2 |
| SDK Version                            |  2.28.1                         | 
| Segment			|  Robot                      | 

### Issue Description
I try to run d435 and t265 together on Jetson TX2 via realsense-viewer. To do this I have connected both cameras to 4-ports USB 3.0 hub. (https://www.tp-link.com/us/home-networking/usb-hub/uh400/)
If I connect only one of this cameras - each of them works ok.
But if connect them both to USB-hub and try to start stream from t265, t265 works 2-10 seconds approximately and then become disconnected
[image]
and d435 works ok regardless to t265 streaming

I tried to run them on different PC via the same USB-hub and they worked ok.
What can I do to solve this problem? Is there any related issues? Is this supported to run d435 and t265 together on jetson tx2?

Thank you in advance!

dataplayer12, 2019-10-26:
I have not tried connecting both cameras to TX2, but I have done so with the Xavier, by connecting the cameras to two different USB ports on the board. Basically, your hub is not able to provide enough power to both cameras. The D435i needs ~500 mA current while T265 requires 300 mA. 

Since there is a micro-USB port on the TX2 kit and a OTG adapter is provided in the box, you could connect your T265 to the micro-USB port.

Also, the USB hub you linked to doesn't seem to be externally powered. It is recommended to use a powered USB hub for the D435i.

jungladicitta, 2019-10-27:
@dataplayer12 thanks for your response.

> Also, the USB hub you linked to doesn't seem to be externally powered. It is recommended to use a powered USB hub for the D435i
Yes, this hub is not externally powered, but I tried to use both cameras working together on a different PC (not jetson) via the same hub and it was working! Dont you think problem is not related to hub? Also, I have d435 not d435i.

> Since there is a micro-USB port on the TX2 kit and a OTG adapter is provided in the box, you could connect your T265 to the micro-USB port.
Yes, I'll try.

dataplayer12, 2019-10-27:
> I tried to use both cameras working together on a different PC (not jetson) via the same hub and it was working!

Yes, this happens because the cameras receive power from the PC. A desktop PC or even a laptop can provide up to 1 A current easily, but the jetson cannot.

jungladicitta, 2019-10-28:
@dataplayer12 
I did test with only t265 connected to that micro-USB port via OTG adapter provided in the box and working in realsense-viewer. T265 works ok for some time but then 2 or 1 events happens regardless to each other:

1. t265 starts moving extremely up with acceleration
[image]

2. notification about relocalization appears
[image]

So, I conclude that working with t265 via OTG adapter is not acceptable.

I have already ordered powered USB 3.0 hub and let you know here about success.

dataplayer12, 2019-10-28:
Thank you for the info. I have not experienced this with any Jetson board. Have you tried using a different cable, just to be sure? Also, try streaming just the fisheye frames (not position data) from the camera with a C++ or python script and see how long it remains stable. Also, don't block out the camera with something. The T265 is constantly using visual features in the image to localize itself, so sudden changes in the view of the camera will confuse it and cause it to relocalize. Good luck.

jungladicitta, 2019-10-30:
@dataplayer12 
I have bought Ginzzu gr-384uab powered USB hub and.. it didnt help. I am still have the same issue, t265 become disconnected.
@dorodnic can you help me?

bsbretly, 2019-11-19:
@jungladicitta - are you able to please provide the steps you followed to successfully run at least one of your realsense devices on your TX2?

Are you running L4T 28.1.0?

Much appreciated!

jungladicitta, 2019-12-09:
@bsbretly sorry for delay, I have installed librealsense via apt
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

I am running L4T 28.2.1

How is your success with jetson and realsense devices? which issues are you facing now?
you have mobile robot?

bsbretly, 2019-12-10:
@jungladicitta - thanks for following up. I was able to get the t265 running by following the standard instructions here: https://github.com/IntelRealSense/librealsense

However, I have yet to get the d435 running. I still have some troubleshooting to complete regarding the d345, but will update you on my status when I have results.

shreyas2311, 2020-01-16:
I am having the same issue. However when I disable streaming the fish-eye camera views, it seems to be fine.

RealSenseSupport, 2020-09-21:
Thank you for highlighting the issue using both T265 and D435 on TX2. We have moved our focus to our next generation of products and consequently, we will not be addressing this issue in the T265.

dataplayer12, 2020-09-22:
I just installed JetPack 4.4 and librealsense v2.38.1 on my TX2 and had no problems in using both D435i and T265 simultaneously without even a powered hub. So, the issue seems to have been resolved.

closed 2020-09-23 by RealSenseSupport