realsenseai/librealsense #5130: No camera detected on odroid xu4

TobiMiller, 2019-10-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 } | 
| Firmware Version                   | (not possible) | 
| Operating System & Version |   Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  (4.14)                                         | 
| Platform                                 | Odroid XU4  |
| SDK Version                            |  { newest with compile tutorial}                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {ROS }                          | 

### Issue Description

I am running the official Ubuntu 18.04 on my odroid xu4. compilation worked fine. but i cannot see any Realsense T265 Camera in my Viewer. 
I have got the camera running on my jetson tx2 with ubuntu 16.04 and under x86 ubuntu 18.04 on my desktop.
dmesg shows me, that it detects a movidius.
can you give me some support how to debug my problem?!

thanks a lot!!!

ev-mp, 2019-10-27:
@TobiMiller hello,
When T265 is recognized as Movidius  then the system is configured properly.

In order to make the camera available to the SDK two things need to be checked:
- Make sure udev-rules are installed. (You can run `sudo realsense-viewer` and if you'd see the device recognized this would mean that `udev-rules` **are not installed**)
- Make sure that the SDK is built with "-DBUILD_WITH_TM2=true" and that you have an internet access required to get  the T265 image. Having no/firewalled internet access is a common reason for T265 not being recognized by SDK, such as #4812 
>Configuration failed.
CMake Warning at D:\librealsense\wrappers\android\librealsense\..\..\..\CMakeLists.txt:10 (message):
  No internet connection, disabling BUILD_WITH_TM2

TobiMiller, 2019-10-29:
The build flag solved my problem. i dont remember that i have set it on my jetson or pc. 
now i had a drift problem with the ros wrapper. solved it by deactivating the fisheye camera topics

RealSenseCustomerSupport, 2019-11-20:
----------------------------------------------

Hi @TobiMiller

Is there anything further you need on this thread?

closed 2019-11-30 by TobiMiller