realsenseai/librealsense #5131: Heterogeneous timestamp domains with multiple cameras?

evercoast-smm, 2019-10-25:
Camera Model: D415
Firmware Version: 05.11.15.00
OS: Ubuntu 18.04 with kernel 4.18.20-041820-generic
Platform: PC
SDK Version: librealsense-2.29.0
Segment: Telecommunications

For this test we have 10x D415 cameras attached to our machine. We're highly confident of bus bandwidth and power.

We've recently added an assert to our frame grabber that I was surprised to see always trigger. In pseudo code:
```
if (rs2_video_frame.get_frame_timestamp_domain() != RS2_TIMESTAMP_DOMAIN_GLOBAL_TIME) {
    // assert here, unexpected behavior
}
```

Our software reports that:
```
Frame from sensor (833412060946) contains non-global timestamp domain (System Time)
```

So far, we've consistently seen this only for the 1st camera in the list that comes back from `query_sensors()`

Hardware resetting the camera doesn't help.

We are not using a hardware synchronization cable in this instance.

Questions:

1) Is this the expected behavior?

2) If not, should we explicitly set the desired time-stamp domain on each device/sensor?

Thank you!

dorodnic, 2019-10-27:
Hi @evercoast-smm 
No, this is not the expected behavior, in fact only one camera reporting system time on Linux is very strange. Could you please add `rs2::log_to_file(RS2_LOG_SEVERITY_DEBUG, "filename.log");` and share the logs? If you have direct support account with Intel, you can also share this way.

dorodnic, 2019-10-27:
Also, `dmesg` output... @ev-mp - additional suggestions?

ev-mp, 2019-10-27:
@evercoast-smm, please provide application/kernel logs for review as @dorodnic mentioned.
Additionally, please answer the following : 
- What do you mean by
>So far, we've consistently seen this only for the 1st camera in the list that comes back from query_sensors()

Is this reproduced only for the first device in a list, and for the rest the ts domain is `global_ts` ? 
Is also reproduced for other cameras as well ?

Is it a temporal or persistent issue, i.e  does the  `system_ts` get replaced by `global_ts` after some period or not ? 

Since kernel 4.16 the frame metadata, upon which the `global_ts` is based is handled differently at the kernel level, and it is not guaranteed anymore that all frames will receive it.
Though we haven't received user reports that this actually happened to them  so far.

evercoast-smm, 2019-10-27:
@dorodnic , attached please find the log that you requested as "librealsense2.log":
[librealsense2.log](https://github.com/IntelRealSense/librealsense/files/3776007/librealsense2.log)

@ev-mp , attached please find the dmesg log that you requested as "dmesg.log":
[dmesg.log](https://github.com/IntelRealSense/librealsense/files/3776009/dmesg.log)

^^ The above was a 20 second run with with 8x D415 cameras streaming RGB8 color at 1920x1080, and Z16 depth at 1280x720.

Camera status from our API:
firmware for sensor serial # (831512060547) is (05.11.15.00)
firmware for sensor serial # (836212060156) is (05.11.15.00)
firmware for sensor serial # (805212060282) is (05.11.15.00)
firmware for sensor serial # (836112060232) is (05.11.15.00)
firmware for sensor serial # (828612061994) is (05.11.15.00)
firmware for sensor serial # (828612061314) is (05.11.15.00)
firmware for sensor serial # (833412060946) is (05.11.15.00)
firmware for sensor serial # (836212060150) is (05.11.15.00)
interface for sensor serial # (831512060547) is (USB 3.2)
interface for sensor serial # (836212060156) is (USB 3.2)
interface for sensor serial # (805212060282) is (USB 3.2)
interface for sensor serial # (836112060232) is (USB 3.2)
interface for sensor serial # (828612061994) is (USB 3.2)
interface for sensor serial # (828612061314) is (USB 3.2)
interface for sensor serial # (833412060946) is (USB 3.2)
interface for sensor serial # (836212060150) is (USB 3.2)

@ev-mp _"Is this reproduced only for the first device in a list, and for the rest the ts domain is global_ts ? Is also reproduced for other cameras as well ?"_
Yes, what you said above is correct. Furthermore, we've tried removing and replacing the offending camera.  The issue will then transfer to the new 1st camera in the list of devices returned by `query_sensors()`.

@ev-mp Is it a temporal or persistent issue, i.e does the system_ts get replaced by global_ts after some period or not ?
Persistent. Across reboots, etc. We're developing a detailed series of unit tests for our product, and have consistently noted this since we started trapping it a week or so ago. I just did several back to back 60 second runs in response to your question about it being temporal. It does not appear to be, we see it on every frame. Only for one sensor. And that sensor is consistently the 1st sensor in the list from `query_sensors()`.

Thanks.

RealSenseCustomerSupport, 2020-02-07:
----------------------------------------------

Please kindly let us know if this issue still persists.
Thank you.

evercoast-smm, 2020-02-07:
We can confirm that this issue has been resolved in 2.31 with
FW 05.12.02.100. Note that we can also confirm that it will only work when
using an LTS kernel.

Thank you!

Evercoast <http://www.evercoast.com>  |  Sebastian Marino  |  Co-founder &
CTO

RealSenseCustomerSupport, 2020-02-11:
----------------------------------------------

Hi Sebastian,
Thank you very much for the update.
Closing this ticket now.

closed 2020-02-11 by RealSenseCustomerSupport