realsenseai/librealsense #5132: Timeouts and dropped frames with rs2::Syncer try_wait_for_frames()?

evercoast-smm, 2019-10-25:
Camera Model: D415
Firmware Version: 05.11.15.00
OS: Ubuntu 18.04 with kernel 4.18.20-041820-generic
Platform: PC
SDK Version: librealsense-2.29.0
Segment: Telecommunications

For this test we have 10x D415 cameras attached to our machine. We're highly confident of bus bandwidth and power.

We're experimenting with using the `rs2::Syner` class with multiple cameras. For our tests we're capturing with the color camera at 1920x1080 (RGB) and the stereo camera at 1280x720 (Z16) at 30 fps, our test duration is 1 minute of capture.

We note the following:

1) We frequently (~33% of all tests) result in timeouts (at the default of 1500 ms) from `try_wait_for_frames()` at the outset of or during the capture interval.

2) We also frequently see drops of ~10 consecutive frames at a time, during runs where we don't observe a timeout.

Per @dorodnic 's response to our previous question, we're using a queue size of 2. See:
https://github.com/IntelRealSense/librealsense/issues/5041#issuecomment-542241323

Questions:

1) Why do we observe the timeouts & drops from `rs2::Syncer::try_wait_for_frames()`? Is this behavior expected? Is `rs2::Syncer` built to handle 10+ devices? Are we using this for its intended purpose?

2) Are we correct in using a queue size of 2 while using 10+ devices?

Thank you!

RealSenseCustomerSupport, 2019-11-13:
----------------------------------------------

Is this still an issue or has been resolved?
The timeout might be due to the waiting of coherent set of frames from 10 cameras.

evercoast-smm, 2019-11-13:
I'd love to better understand your process for "waiting of coherent set of frames from 10 cameras"? Can you please elaborate on that, and/or refer us to a specific section of the RS code?

Thanks!

RealSenseCustomerSupport, 2019-11-14:
----------------------------------------------

Clarification, I meant to say the rs2::Syncer would try to wait and deliver synchronized frames (one RGB and one Depth stream) per camera. In this case, there are total of 10 cameras.

The frame syncer is documented here:
https://github.com/IntelRealSense/librealsense/blob/659860bb9a0eca56814c97d90a162b3cdfb2af82/doc/frame_lifetime.md#frame-syncer

RealSenseCustomerSupport, 2019-12-18:
----------------------------------------------

Please let us know if more help is needed.
Thank you.

closed 2020-01-08 by RealSenseCustomerSupport