realsenseai/librealsense #5134: how to get the intel SR305 intrinsic and extrinsic calibration information

davidbun, 2019-10-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR305                                                     | 
| Firmware Version                   | 3.27.3.0                                                   | 
| Operating System & Version | Win 10                                                    | 
| Kernel Version (Linux Only)    |                                                                | 
| Platform                                 | PC                                                           |
| SDK Version                           | 2.22                                                         | 
| Language                               |  C++                                                       | 
| Segment			         |  Inspection                                             | 

### Issue Description
ControlRW application works well for the D415 sensor and does not work for the SR305 sensor
How can I get all the intrinsic and extrinsic parameters from the SR305 sensor?

MartyG-RealSense, 2019-10-26:
You can get the extrinsics of the SR300 and SR305 with an SDK tool called **sensor-control**.

https://github.com/IntelRealSense/librealsense/tree/master/examples/sensor-control

If you have a Windows PC, a pre-made executable version of this tool can be found in the 'Intel RealSense SDK 2.0 > Tools' SDK folder.  The file is called **rs-sensor-control**

When the tool is run, it will display the index name of attached devices.  If no other camera is attached, it should assign index '0' to your camera.

[image]

Input '0' for the index number and press the Enter key.

It will then list the information of your camera and ask you to input '0' for the details of the depth sensor or '1' for details of the RGB sensor.  Input '0' again.

[image]

Next, enter '2' to display stream intrinsics, or '3' to display device extrinsics.

[image]

If, for example, you choose '3' for extrinsics, then you are asked again to choose between '0' for the depth sensor or '1' for the RGB.   

[image]

If you choose '0' for the depth sensor, you get something like this.

[image]

Input a profile number (e.g '23'), and again enter '0' for depth stream details or '1' for RGB details.

[image]

Enter the stream profile number again.   You will then get the extrinsics for that stream profile and be offered the chance to check another extrinsic.

[image]

RealSenseCustomerSupport, 2019-12-05:
----------------------------------------------

@davidbun,

If no further question, will close this.

closed 2019-12-10 by dorodnic