realsenseai/librealsense #5135: Getting 3D coordinates of pixels

GOBish, 2019-10-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   |  | 
| Operating System & Version |   / Linux (Ubuntu 14/16/17)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 |  NVIDIA Jetson / |
| SDK Version                            |  }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot /AI}                          | 

Hoping for some guidance--
I followed the example here
https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb 
for getting the distance of an area / pixel after using object detection. Very helpful, thanks!
@dorodnic -- could you provide an example of how I can get the x and y coordinates of a pixel or point in the image relative to the camera?

I saw this issue: https://github.com/IntelRealSense/librealsense/issues/3688 but couldn't totally follow it.
An example like the one for distance would be a huge help to me!
Thanks.

MartyG-RealSense, 2019-10-26:
I wonder if the link below would be of help to you in getting pixel coordinates with Python, as it contains Pyrealsense2 scripts for doing so.

https://github.com/IntelRealSense/librealsense/issues/1904

If you are interested in finding out more about object detection with Python, the list of links for such programs below may be of interest too:

https://github.com/topics/mobilenet-ssd?l=python

GOBish, 2019-10-27:
I will review these, thank you for the help!

RealSenseCustomerSupport, 2019-11-04:
----------------------------------------------

Hi @GOBish Do you need more help on this?

GOBish, 2019-11-04:
Still working on it!

RealSenseCustomerSupport, 2019-12-04:
----------------------------------------------

Hi @GOBish Any progress? Do you need more help on this?

GOBish, 2019-12-05:
I am still working on it, but I think we can close the topic as I am making some good progress. Thank you for the help!

closed 2019-12-05 by GOBish