realsenseai/librealsense #5136: MacOS Build Error

iPsych, 2019-10-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Mac |
| SDK Version                            |   l2.2.29.0 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
MacOS compile shows an error as below.
The homebrew libusb 1.0.22 is installed and properly linked.

```
[ 62%] Building CXX object CMakeFiles/realsense2.dir/third-party/easyloggingpp/src/easylogging++.cc.o
1 warning generated.
[ 62%] Linking CXX shared library librealsense2.dylib
Undefined symbols for architecture x86_64:
  "_libusb_get_port_number", referenced from:
      perc::Device::GetUsbConnectionDescriptor() in libtm.a(Device.cpp.o)
  "_libusb_get_port_numbers", referenced from:
      librealsense::platform::get_usb_descriptors(libusb_device*) in enumerator-libusb.cpp.o
      librealsense::platform::get_usb_descriptors(libusb_device*) in libuvc.cpp.o
      perc::UsbPlugListener::EnumerateDevices() in libtm.a(UsbPlugListener.cpp.o)
      perc::Device::GetUsbConnectionDescriptor() in libtm.a(Device.cpp.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [librealsense2.2.29.0.dylib] Error 1
make[1]: *** [CMakeFiles/realsense2.dir/all] Error 2
make: *** [all] Error 2
```

MartyG-RealSense, 2019-10-27:
Another user who had the same undefined symbol error on MacOS offered the advice in the link below:

https://github.com/IntelRealSense/librealsense/issues/2685#issuecomment-436498519

iPsych, 2019-10-27:
> Another user who had the same undefined symbol error on MacOS offered the advice in the link below:
> 
> [#2685 (comment)](https://github.com/IntelRealSense/librealsense/issues/2685#issuecomment-436498519)

It's compilation of one cpp, but the error happens in
`make -j2 
`after successful cmake
`cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -DBUILD_WITH_TM2=false`

MartyG-RealSense, 2019-10-27:
If it is tripping up over libusb, I suspect that **-DHWM_OVER_XU=false** might be the cause.  Are you able to build successfully if that flag is left out of the build command?

ElnazTaqizadeh, 2019-11-04:
@iPsych have you been able to resolve the issue?

RealSenseCustomerSupport, 2019-11-06:
----------------------------------------------

@iPsych Did you try the workaround below described in https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md ? Looking forward to your update. Thanks!

**Note:** On some Mac systems you might encounter `ld: library not found for -lusb-1.0` error (either in the terminal during make or in XCode) This can be worked-around by setting environment variable: `/bin/launchctl setenv LIBRARY_PATH /usr/local/lib`

iPsych, 2019-11-06:
@RealSenseCustomerSupport @ElnazTaqizadeh 
Yes, After re-install libusb using homebrew, and set the env variable using /bin/launchctl setenv LIBRARY_PATH /usr/local/lib, it resolves the issue.

neilyoung, 2019-11-06:
@iPsych Does this mean you are able to build the latest 2.30.0 SDK on macOS Catalina? And does it also provide full T265 support? Asking this because T265 support is still officially not supported on macOS.

iPsych, 2019-11-06:
@neilyoung No, I am staying High-Sierra, since Mojave and Catalina have a lot of stability problems in my projects. I am using D435.

neilyoung, 2019-11-06:
@iPsych OK, not a thing. I think I'm going to give it a try, nevertheless. Thanks.

seanngpack, 2019-11-06:
@neilyoung could you report back when you try the build on Catalina?

neilyoung, 2019-11-06:
@seanngpack Sure

RealSenseCustomerSupport, 2019-11-07:
----------------------------------------------

@iPsych Glad to hear the issue resolved. Close it accordingly. Thanks!

closed 2019-11-07 by RealSenseCustomerSupport

neilyoung, 2019-11-07:
@seanngpack Sorry, was not even able to generate a Makefile...:( Since it is not my priority I'm not pursuing this any further. At least the gist here is nonsense from first to last on Catalina :)

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md

neilyoung, 2019-11-18:
@seanngpack OK, meanwhile I could build on macOS Catalina. https://github.com/IntelRealSense/librealsense/issues/5258