realsenseai/librealsense #5137: "Colorizing" the right IR imager stream with color camera (d415 unity)

TheBricktop, 2019-10-27:
Hey, 
I've got not so usual problem, 
In unity i've managed to get left and rightmost image streams : right IR and color, 
in attempt to use it with stereo camera setup (XR) as a stereo ar background, and I want to colorize the ir output using colour reference. Currently im thinking about custom shader that would stylize and mix both streams to simulate disparity, but maybe there is a way to do it using existing methods.

dorodnic, 2019-11-11:
Hi @TheBricktop 
The problem makes sense, sorry for delay in response. There is unfortunately no built-in way to map one texture stream to another texture stream without one of them being depth. So probably using shader with extrinsics and extrinsics passed as uniforms. The following code may be useful as reference - [gl/pointcloud-gl.cpp:24-103](https://github.com/IntelRealSense/librealsense/blob/master/src/gl/pointcloud-gl.cpp#L24-L103)

RealSenseCustomerSupport, 2019-11-22:
----------------------------------------------

@TheBricktop Any other questions with the reply from dorodnic? Looking forward to your update. Thanks!

TheBricktop, 2019-11-22:
There's probably a way to do this using opencvHomography abeit it could be cpu taxing.

RealSenseCustomerSupport, 2019-12-06:
----------------------------------------------

@TheBricktop Thanks for the update! Did you get it through now?

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

@TheBricktop  Any other questions? Thanks!

RealSenseCustomerSupport, 2019-12-23:
----------------------------------------------

@TheBricktop Will close this if no other questions. Thanks!

closed 2019-12-23 by RealSenseCustomerSupport