realsenseai/librealsense #5138: T265: pipe.start(cfg) fails if "eyes are wide shut" :)

neilyoung, 2019-10-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.1.0.279
| Operating System & Version |  Ubuntu 16.04
| Kernel Version (Linux Only)    |   5.3.7   | 
| Platform                                 |  Banana PI M2 Zero |
| SDK Version                            |   2.2.9.0                           | 
| Language                            |                     | 
| Segment			|  others                           | 

### Issue Description
An attempt to start tracking with camera lenses are covered, environment is fully dark or cam is just laying looking downwards on the desk this is the reproducible exception generated by `pipe.start(cfg)`:

```
21:36:33.675 [3180] [E] Device-81B0: MessageID 0x1006 (SLAM_6DOF_CONTROL) failed with status 0x8
21:36:33.675 [3165] [E] Device-81B0: Error Transferring SLAM_6DOF_CONTROL
21:36:33.675 [3165] [E] Device-81B0: Error: Failed setting 6dof Control (0x6)

```

RealSenseCustomerSupport, 2019-12-06:
----------------------------------------------

Hi neilyoung,

This is expected behavior. The T265 using visual queues to predict pose, so occluding the camera lenses or operating in a low light environment reduces the efficiency of the SLAM prediction until it can no longer return values with any meaningful confidence. For maximum tracking results both cameras should be unobstructed.

Thanks

radfordi, 2019-12-06:
Hi @neilyoung, this happens because we are not sending poses *at all* when we are in 3dof mode (e.g. when the cameras are covered), and so there is no reasonable timeout to use (besides infinity, but that won't catch when the device has crashed and stopped sending poses).  An easy fix would be to just send poses when we are in 3dof mode.

neilyoung, 2019-12-07:
Thanks. Not sure I understand. I understand the problem to switch to 6dof while lenses are not working. I was just not sure if this situation has to end up in an exception or if it could simply be solved by sending 3dof with lower confidence or so

RealSenseCustomerSupport, 2019-12-18:
----------------------------------------------

That's a great suggestion Neil and I've passed this recommendation to our dev team as a potential future change. In the meantime feel free to handle the exception however works best for your purposes.

Thanks

closed 2020-01-22 by RealSenseCustomerSupport