realsenseai/librealsense #5139: D435i stopped working after firmware update

Kylemaycry, 2019-10-27:
| | |
|---|---|
|**librealsense**|2.29.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435I|
|**Serial Number**|838412070598|
|**Firmware Version**|05.10.13.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|903223052509|
|**Firmware Update Id**|903223052509|

Please provide a description of the problem
This issue appears since last time I upgraded the firmware, the upgrading process stopped at the middle for half an hour without any progressing, so I disconnected and powered off the device while firmware upgrading was still processing. 

After that, my camera d435i can show up in Device Manager of Windows, but Intel RealSense Viewer will crash once d435i is connected.  If I kept trying to open the RealSense Viewer program, it can be launched by chance. But I still cannot access the cameras or update firmware in the Viewer as shown below. 
[image]
[image]

I also tried using Command Prompt, which shows that the firmware has been updated. But it's still not working.
[image]

And I have installed  the Intel RealSense Viewer on Ubuntu, the camera cannot be detected by the Viewer on Ubuntu.

ev-mp, 2019-10-28:
@Kylemaycry , can you share the outputs of `rs-enumerate-devices`, `lsusb`  and  `dmesg` after plugging in the device?

Kylemaycry, 2019-10-28:
C:\Program Files (x86)\Intel RealSense SDK 2.0\tools>rs-enumerate-devices
Device info:
    Name                          :     Intel RealSense D435I
    Serial Number                 :     838412070598
    Firmware Version              :     05.10.13.00
    Recommended Firmware Version  :     05.11.15.00
    Physical Port                 :     \\?\usb#vid_8086&pid_0b3a&mi_00#6&cd8de86&0&0000#{e5323777-f976-4f5b-9b55-b94699c46e44}\global
    Debug Op Code                 :     15
    Advanced Mode                 :     YES
    Product Id                    :     0B3A
    Camera Locked                 :     YES
    Usb Type Descriptor           :     3.2
    Product Line                  :     D400
    Asic Serial Number            :     903223052509
    Firmware Update Id            :     903223052509

Stream Profiles supported by Stereo Module
 Supported modes:
    stream       resolution      fps       format
    Infrared 1    1280x800      @ 30Hz     Y8
    Infrared 2    1280x800      @ 30Hz     Y8
    Infrared 1    1280x800      @ 25Hz     Y16
    Infrared 2    1280x800      @ 25Hz     Y16
    Infrared 1    1280x800      @ 15Hz     Y16
    Infrared 2    1280x800      @ 15Hz     Y16
    Infrared 1    1280x800      @ 15Hz     Y8
    Infrared 2    1280x800      @ 15Hz     Y8
    Infrared 2    1280x720      @ 30Hz     Y8
    Infrared 1    1280x720      @ 30Hz     Y8
    Infrared 2    1280x720      @ 15Hz     Y8
    Infrared 1    1280x720      @ 15Hz     Y8
    Infrared 2    1280x720      @ 6Hz      Y8
    Infrared 1    1280x720      @ 6Hz      Y8
    Infrared 2    848x480       @ 90Hz     Y8
    Infrared 1    848x480       @ 90Hz     Y8
    Infrared 2    848x480       @ 60Hz     Y8
    Infrared 1    848x480       @ 60Hz     Y8
    Infrared 1    848x480       @ 30Hz     Y8
    Infrared 2    848x480       @ 30Hz     Y8
    Infrared 2    848x480       @ 15Hz     Y8
    Infrared 1    848x480       @ 15Hz     Y8
    Infrared 2    848x480       @ 6Hz      Y8
    Infrared 1    848x480       @ 6Hz      Y8
    Infrared 1    640x480       @ 90Hz     Y8
    Infrared 2    640x480       @ 90Hz     Y8
    Infrared 1    640x480       @ 60Hz     Y8
    Infrared 2    640x480       @ 60Hz     Y8
    Infrared 2    640x480       @ 30Hz     Y8
    Infrared 1    640x480       @ 30Hz     Y8
    Infrared 1    640x480       @ 15Hz     Y8
    Infrared 2    640x480       @ 15Hz     Y8
    Infrared 2    640x480       @ 6Hz      Y8
    Infrared 1    640x480       @ 6Hz      Y8
    Infrared 1    640x400       @ 25Hz     Y16
    Infrared 2    640x400       @ 25Hz     Y16
    Infrared 2    640x400       @ 15Hz     Y16
    Infrared 1    640x400       @ 15Hz     Y16
    Infrared 2    640x360       @ 90Hz     Y8
    Infrared 1    640x360       @ 90Hz     Y8
    Infrared 1    640x360       @ 60Hz     Y8
    Infrared 2    640x360       @ 60Hz     Y8
    Infrared 2    640x360       @ 30Hz     Y8
    Infrared 1    640x360       @ 30Hz     Y8
    Infrared 1    640x360       @ 15Hz     Y8
    Infrared 2    640x360       @ 15Hz     Y8
    Infrared 1    640x360       @ 6Hz      Y8
    Infrared 2    640x360       @ 6Hz      Y8
    Infrared 1    480x270       @ 90Hz     Y8
    Infrared 2    480x270       @ 90Hz     Y8
    Infrared 1    480x270       @ 60Hz     Y8
[... 227 lines cut ...]
    Color         320x180       @ 60Hz     RGBA8
    Color         320x180       @ 60Hz     BGR8
    Color         320x180       @ 60Hz     YUYV
    Color         320x180       @ 30Hz     RGB8
    Color         320x180       @ 30Hz     Y16
    Color         320x180       @ 30Hz     BGRA8
    Color         320x180       @ 30Hz     RGBA8
    Color         320x180       @ 30Hz     BGR8
    Color         320x180       @ 30Hz     YUYV
    Color         320x180       @ 6Hz      RGB8
    Color         320x180       @ 6Hz      Y16
    Color         320x180       @ 6Hz      BGRA8
    Color         320x180       @ 6Hz      RGBA8
    Color         320x180       @ 6Hz      BGR8
    Color         320x180       @ 6Hz      YUYV

Stream Profiles supported by Motion Module
 Supported modes:
    stream       resolution      fps       format
    Gyro         N/A            @ 200Hz    MOTION_XYZ32F
    Gyro         N/A            @ 400Hz    MOTION_XYZ32F
    Accel        N/A            @ 63Hz     MOTION_XYZ32F
    Accel        N/A            @ 250Hz    MOTION_XYZ32F

I didn't get any outputs of lsusb and dmesg from Command Prompt on Windows.

ev-mp, 2019-10-28:
Thanks for the quick response. According to the second snapshot you're trying to recover rather than update the firmware. Did you try
>rs-fw-update -f <file_name> 

and what were the results ? In case it failed - please provide application/sdk logs.
Additionally I would try to update the firmware with a different USB port.

Also - what is the Windows version in case further assistance will be required (Win+R -> winver -> Enter)  ?

Kylemaycry, 2019-10-28:
Just tried `rs-fw-update -f <file_name>` , the results are that the firmware has been updated to version: 05.11.15.00, but the Viewer is still crashing when camera is connected.

[image]

I also tried to update with different USB port, the results are same.

I am using Windows 10.

Thanks for your kind help.

ev-mp, 2019-10-29:
Hi, It seem that the FW update was done successfully, so these are good news.
As for the error - a  similar issue was previously reported in #4571 and resolved by updating Windows Bulid to 18963.

Did you manage to check the current version ?
>what is the Windows version in case further assistance will be required (Win+R -> winver -> Enter) ?

Kylemaycry, 2019-10-29:
[image]

Do I need to joint the Windows Insider Program to update Windows Bulid to 18963?

Kylemaycry, 2019-10-30:
Also, the camera cannot be detected by the Viewer on Ubuntu.

ev-mp, 2019-10-30:
@Kylemaycry, can you provide kernel/app logs for review?
- kernel (`dmesg`, `lsusb`)
- application level (`rs-enumerate-devices`)

RealSenseCustomerSupport, 2019-11-11:
----------------------------------------------

Hi @Kylemaycry,

Do you have any update to provide more debug info?

Kylemaycry, 2019-11-11:
Still working on it

Kylemaycry, 2019-11-11:
From` rs-fw-logger.exe`

Waiting for RealSense device to connect...
RealSense device was connected...
Device Name: Intel RealSense D435I
Device Location: \\?\usb#vid_8086&pid_0b3a&mi_00#6&cd8de86&0&0000#{e5323777-f976-4f5b-9b55-b94699c46e44}\global

2019-11-10 22:47:23  FW_Log_Data:0F 00 00 00 A0 02 05 10 50 00 83 F0 C4 00 00 00 00 00 00 00 BD 63 88 1C A0 01 1D 10 26 00 63 52 01 00 14 00 8A 3B FC 3F 7E 37 F1 11 A0 A1 1D 10 97 00 3E 64 03 00 01 00 2C 00 00 00 D2 F9 EE 12 A0 A1 1D 10 97 00 3E 74 01 00 14 00 8A 3B FC 3F 6E 33 06 13 A0 01 26 10 07 00 EF 80 00 00 7F 07 00 00 00 00 D0 40 06 13 A0 01 1D 10 26 00 63 92 01 00 14 00 8A 3B FC 3F 66 43 06 13 A0 A1 1D 10 97 00 3E A4 03 00 01 00 6A 00 00 00 6A 23 13 14 A0 01 0D 10 5B 00 1C B2 03 00 2C 00 00 00 00 00 3A 41 13 14 A0 01 0D 10 6A 01 21 C2 00 00 00 00 00 00 00 00 E4 42 13 14 A0 A1 1D 10 97 00 3E D4 01 00 14 00 8A 3B FC 3F D5 03 2C 14 A0 01 26 10 07 00 EF E0 00 00 7F 07 00 00 00 00 66 11 2C 14 A0 01 1D 10 26 00 63 F2 01 00 14 00 8A 3B FC 3F 2D 14 2C 14 A0 A1 1D 10 97 00 3E 04 01 00 14 00 8A 3B FC 3F C1 D4 9D 15 A0 01 26 10 07 00 EF 10 00 00 7F 07 00 00 00 00 52 E2 9D 15 A0 01 1D 10 26 00 63 22 01 00 14 00 8A 3B FC 3F 19 E5 9D 15 A0 A1 1D 20 80 02 D6 33 06 00 04 00 A6 00 00 00 D5 9F F8 16 A0 A1 1D 20 80 02 D6 43 06 00 02 00 08 00 00 00 6D 75 F9 16 A0 A2 1D 20 A9 00 E6 53 06 00 14 00 06 00 00 00 8F AD 03 17 A0 A1 1D 20 80 02 D6 63 00 00 02 00 06 00 00 00 2E B7 04 17 A0 A1 1D 20 80 02 D6 73 07 00 01 00 00 00 00 00 C2 EA 9A 17 A0 A2 1D 20 A9 00 E6 83 06 00 14 00 06 00 00 00 1E 13 A5 17 A0 A1 1D 20 80 02 D6 93 00 00 02 00 06 00 00 00 18 29 A6 17 A0 A1 1D 20 80 02 D6 A3 07 00 01 00 00 00 00 00 2E A5 36 18 A0 A2 1D 20 A9 00 E6 B3 06 00 14 00 06 00 00 00 69 01 45 18 A0 A1 1D 20 80 02 D6 C3 00 00 02 00 06 00 00 00 07 0C 46 18

2019-11-10 22:47:24  FW_Log_Data:0F 00 00 00 A0 A1 1D 20 80 02 D6 D3 07 00 02 00 00 00 00 00 3F 69 D6 18 A0 A2 1D 20 A9 00 E6 E3 06 00 14 00 06 00 00 00 F1 E0 E6 18 A0 A1 1D 20 80 02 D6 F3 00 00 02 00 06 00 00 00 A8 D7 E7 18 A0 A1 1D 20 80 02 D6 03 07 00 02 00 00 00 00 00 CB E5 86 19 A0 A2 1D 20 A9 00 E6 13 06 00 14 00 06 00 00 00 F5 D3 91 19 A0 A1 1D 20 80 02 D6 23 00 00 02 00 06 00 00 00 CC C0 92 19 A0 A1 1D 20 80 02 D6 33 07 00 03 00 32 00 00 00 92 B3 3E 1A A0 A2 1D 20 A9 00 E6 43 06 00 14 00 06 00 00 00 D4 B7 47 1A A0 A1 1D 20 80 02 D6 53 00 00 02 00 06 00 00 00 A2 D8 48 1A A0 A1 1D 20 80 02 D6 63 07 00 03 00 32 00 00 00 E8 A5 D7 1A A0 A2 1D 20 A9 00 E6 73 06 00 14 00 06 00 00 00 C7 2C E2 1A A0 A1 1D 20 80 02 D6 83 00 00 02 00 06 00 00 00 8A 63 E3 1A A0 A1 1D 20 80 02 D6 93 06 00 04 00 A6 00 00 00 FC C1 75 1B A0 A1 1D 20 80 02 D6 A3 06 00 02 00 08 00 00 00 06 42 76 1B A0 A2 1D 20 A9 00 E6 B3 06 00 14 00 06 00 00 00 FC FA 80 1B A0 A1 1D 20 80 02 D6 C3 00 00 02 00 06 00 00 00 69 09 82 1B A0 A1 1D 20 80 02 D6 D3 06 00 04 00 A6 00 00 00 57 E8 0E 1C A0 A1 1D 20 80 02 D6 E3 06 00 02 00 08 00 00 00 3C B9 0F 1C A0 A2 1D 20 A9 00 E6 F3 06 00 14 00 06 00 00 00 4F 1E 1D 1C A0 A1 1D 20 80 02 D6 03 00 00 02 00 06 00 00 00 1D 26 1E 1C A0 A1 1D 20 80 02 D6 13 07 00 04 00 40 00 00 00 90 77 B7 1C A0 A2 1D 20 A9 00 E6 23 06 00 14 00 06 00 00 00 2E A4 CA 1C A0 A1 1D 20 80 02 D6 33 00 00 02 00 06 00 00 00 DE 85 CB 1C A0 A1 1D 20 80 02 D6 43 07 00 04 00 40 00 00 00 EC BF 6A 1D A0 A2 1D 20 A9 00 E6 53 06 00 14 00 06 00 00 00 28 0E 75 1D

2019-11-10 22:47:24  FW_Log_Data:0F 00 00 00 A0 A1 1D 20 80 02 D6 63 00 00 02 00 06 00 00 00 D9 49 76 1D A0 A2 1D 20 A9 00 E6 73 06 00 14 00 06 00 00 00 1A 85 0F 23 A0 A1 1D 20 80 02 D6 83 00 00 02 00 06 00 00 00 5F 70 10 23 A0 01 0A 10 8B 02 74 92 1D 00 1B 00 00 00 00 00 9C 40 BF 30 A0 02 19 10 01 02 D0 A0 00 00 00 00 00 00 00 00 CA DB 91 36 A0 01 18 10 29 00 34 B0 04 00 00 00 00 00 00 00 6D 61 92 36 A0 01 3A 10 19 00 6A C0 07 00 0A 41 00 00 00 00 3B 02 97 36 A0 01 3A 10 19 00 6A D0 0A 00 0A 00 00 00 00 00 0F 05 0D 37 A0 01 23 10 4B 02 54 E2 01 00 01 00 1E 00 00 00 34 B8 0E 37 A0 01 3A 10 19 00 6A F0 07 00 0A 41 00 00 00 00 AB BA 0E 37 A0 01 18 10 2A 00 34 00 02 00 01 00 00 00 00 00 AC D0 0E 37 A0 01 18 10 2A 00 34 10 03 00 01 00 00 00 00 00 8C D2 0E 37 A0 02 19 10 02 02 0E 21 00 00 00 00 00 00 00 00 4C DB 0E 37 A0 A1 1D 10 97 00 3E 34 01 00 14 00 8A 3B FC 3F 35 F8 B3 37 A0 01 26 10 07 00 EF 40 00 00 7F 07 00 00 00 00 C2 05 B4 37 A0 01 1D 10 26 00 63 52 01 00 14 00 8A 3B FC 3F 89 08 B4 37 A0 81 18 10 2A 00 34 60 04 00 00 00 00 00 00 00 E2 26 32 3C A0 02 19 10 01 02 D0 70 00 00 00 00 00 00 00 00 0E 6A 65 4A A0 01 18 10 29 00 34 80 04 00 01 00 00 00 00 00 B1 EF 65 4A A0 01 3A 10 19 00 6A 90 07 00 0A 41 00 00 00 00 7F 90 6A 4A A0 01 3A 10 19 00 6A A0 0A 00 0A 00 00 00 00 00 63 91 E0 4A A0 01 23 10 4B 02 54 B2 01 00 01 00 1E 00 00 00 68 44 E2 4A A0 01 3A 10 19 00 6A C0 07 00 0A 41 00 00 00 00 DF 46 E2 4A A0 01 18 10 2A 00 34 D0 02 00 01 00 00 00 00 00 CC 5C E2 4A A0 01 18 10 2A 00 34 E0 03 00 01 00 00 00 00 00 AC 5E E2 4A

2019-11-10 22:47:24  FW_Log_Data:0F 00 00 00 A0 02 19 10 02 02 0E F1 00 00 00 00 00 00 00 00 E8 67 E2 4A A0 A1 1D 10 97 00 3E 04 03 00 01 00 30 00 00 00 B5 FF 26 4B A0 01 0D 10 30 01 21 12 00 00 00 00 00 00 00 00 A5 20 27 4B A0 A1 1D 10 97 00 3E 24 01 00 14 00 8A 3B FC 3F FD D4 36 4B A0 01 26 10 07 00 EF 30 00 00 7F 07 00 00 00 00 8E E2 36 4B A0 01 1D 10 26 00 63 42 01 00 14 00 8A 3B FC 3F 55 E5 36 4B A0 81 18 10 2A 00 34 50 04 00 00 00 00 00 00 00 22 B2 05 50 A0 02 19 10 01 02 D0 60 00 00 00 00 00 00 00 00 62 70 3E 5E A0 01 18 10 29 00 34 70 04 00 01 00 00 00 00 00 05 F6 3E 5E A0 01 3A 10 19 00 6A 80 07  [... cut ...] E 63 A0 01 0A 10 8B 02 74 32 1D 00 1B 00 00 00 00 00 68 C8 4A 6C A0 02 19 10 01 02 D0 40 00 00 00 00 00 00 00 00 1A EC 06 72 A0 01 18 10 29 00 34 50 04 00 01 00 00 00 00 00 BD 71 07 72 A0 01 3A 10 19 00 6A 60 07 00 0A 41 00 00 00 00 6F 15 0C 72 A0 01 31 18 1A 02 9A 70 01 00 03 00 E8 03 00 00 28 17 10 72
RealSense error calling rs2_send_and_receive_raw_data(device:000002133EDB4E30):
    MFCreateDeviceSource(_device_attrs, &_source) returned: HResult 0x800706be: "The remote procedure call failed."

Kylemaycry, 2019-11-11:
From `rs-fw-logger.exe`

Waiting for RealSense device to connect...
RealSense device was connected...
Device Name: Intel RealSense D435I
Device Location: \\?\usb#vid_8086&pid_0b3a&mi_00#6&cd8de86&0&0000#{e5323777-f976-4f5b-9b55-b94699c46e44}\global

2019-11-10 22:53:03  FW_Log_Data:0F 00 00 00 A0 A1 1D 10 97 00 3E 84 01 00 14 00 8A 3B FC 3F B3 36 16 6C A0 01 26 10 07 00 EF 90 00 00 7F 07 00 00 00 00 15 44 16 6C A0 01 1D 10 26 00 63 A2 01 00 14 00 8A 3B FC 3F AB 46 16 6C A0 01 0A 10 8B 02 74 B2 1B 00 1A 00 00 00 00 00 F3 DC 4F 6C A0 A1 1D 10 97 00 3E C4 03 00 01 00 10 00 00 00 9F DE 8E 7C A0 01 0D 10 5B 00 1C D2 03 00 0F 00 00 00 00 00 65 FD 8E 7C A0 01 0D 10 10 01 21 E2 00 00 00 00 00 00 00 00 0F FF 8E 7C A0 A1 1D 10 97 00 3E F4 01 00 14 00 8A 3B FC 3F 47 FC AC 7C A0 01 26 10 07 00 EF 00 00 00 7F 07 00 00 00 00 A9 09 AD 7C A0 01 1D 10 26 00 63 12 01 00 14 00 8A 3B FC 3F 3F 0C AD 7C A0 A1 1D 10 97 00 3E 24 03 00 01 00 10 00 00 00 63 50 FB 7D A0 A1 1D 10 97 00 3E 34 01 00 14 00 8A 3B FC 3F 3F 0E 0E 7E A0 01 26 10 07 00 EF 40 00 00 7F 07 00 00 00 00 A1 1B 0E 7E A0 01 1D 10 26 00 63 52 01 00 14 00 8A 3B FC 3F 37 1E 0E 7E A0 A1 1D 10 97 00 3E 64 03 00 01 00 10 00 00 00 23 B3 E1 7E A0 A1 1D 10 97 00 3E 74 01 00 14 00 8A 3B FC 3F 5F F4 F2 7E A0 01 26 10 07 00 EF 80 00 00 7F 07 00 00 00 00 C1 01 F3 7E A0 01 1D 10 26 00 63 92 01 00 14 00 8A 3B FC 3F 57 04 F3 7E A0 A1 1D 10 97 00 3E A4 03 00 01 00 30 00 00 00 0F EB C3 7F A0 01 0D 10 5B 00 1C B2 03 00 10 00 00 00 00 00 21 0C C4 7F A0 01 0D 10 30 01 21 C2 00 00 00 00 00 00 00 00 CB 0D C4 7F A0 A1 1D 10 97 00 3E D4 01 00 14 00 8A 3B FC 3F 7F 8B D7 7F A0 01 26 10 07 00 EF E0 00 00 7F 07 00 00 00 00 E1 98 D7 7F A0 01 1D 10 26 00 63 F2 01 00 14 00 8A 3B FC 3F 77 9B D7 7F A0 A1 1D 10 97 00 3E 04 03 00 01 00 10 00 00 00 AB 05 8D 80

2019-11-10 22:53:04  FW_Log_Data:0F 00 00 00 A0 01 0D 10 10 01 21 12 00 00 00 00 00 00 00 00 D4 27 8D 80 A0 A1 1D 10 97 00 3E 24 01 00 14 00 8A 3B FC 3F FF 40 A1 80 A0 01 26 10 07 00 EF 30 00 00 7F 07 00 00 00 00 61 4E A1 80 A0 01 1D 10 26 00 63 42 01 00 14 00 8A 3B FC 3F F7 50 A1 80 A0 A1 1D 10 97 00 3E 54 03 00 01 00 4F 00 00 00 8B AF 0E 89 A0 04 13 10 7A 00 FF 65 00 00 FC FF 09 00 00 00 BA CD 0E 89 A0 01 0D 10 4F 01 21 72 15 00 00 00 08 02 00 00 4B 61 0F 89 A0 A1 1D 10 97 00 3E 84 01 00 14 00 8A 3B FC 3F D3 49 24 89 A0 01 26 10 07 00 EF 90 00 00 7F 07 00 00 00 00 35 57 24 89 A0 01 1D 10 26 00 63 A2 01 00 14 00 8A 3B FC 3F CB 59 24 89 A0 A1 1D 10 97 00 3E B4 03 00 01 00 15 00 00 00 F7 44 F1 89 A0 01 0D 10 15 01 21 C2 00 00 20 00 00 00 00 00 EF 66 F1 89 A0 A1 1D 10 97 00 3E D4 01 00 14 00 8A 3B FC 3F 8F 1D 05 8A A0 01 26 10 07 00 EF E0 00 00 7F 07 00 00 00 00 F1 2A 05 8A A0 01 1D 10 26 00 63 F2 01 00 14 00 8A 3B FC 3F 87 2D 05 8A A0 01 0D 10 0F 01 21 02 00 00 F4 01 00 00 00 00 C9 8E 52 8D A0 A1 1D 10 97 00 3E 14 01 00 14 00 8A 3B FC 3F 4F 72 65 8D A0 01 26 10 07 00 EF 20 00 00 7F 07 00 00 00 00 B1 7F 65 8D A0 01 1D 10 26 00 63 32 01 00 14 00 8A 3B FC 3F 47 82 65 8D

2019-11-10 22:53:04  FW_Log_Data:0F 00 00 00 A0 A1 1D 10 97 00 3E 44 01 00 14 00 8A 3B FC 3F 13 FB 7F 94 A0 01 26 10 07 00 EF 50 00 00 7F 07 00 00 00 00 75 08 80 94 A0 01 1D 10 26 00 63 62 01 00 14 00 8A 3B FC 3F 0B 0B 80 94
RealSense error calling rs2_send_and_receive_raw_data(device:000002243DFF9A50):
    MFCreateDeviceSource(_device_attrs, &_source) returned: HResult 0x800706be: "The remote procedure call failed."

Kylemaycry, 2019-11-11:
From `rs-enumerate-devices`
Device info:
    Name                          :     Intel RealSense D435I
    Serial Number                 :     838412070598
    Firmware Version              :     05.10.13.00
    Recommended Firmware Version  :     05.11.15.00
    Physical Port                 :     \\?\usb#vid_8086&pid_0b3a&mi_00#6&cd8de86&0&0000#{e5323777-f976-4f5b-9b55-b94699c46e44}\global
    Debug Op Code                 :     15
    Advanced Mode                 :     YES
    Product Id                    :     0B3A
    Camera Locked                 :     YES
    Usb Type Descriptor           :     3.2
    Product Line                  :     D400
    Asic Serial Number            :     903223052509
    Firmware Update Id            :     903223052509

Stream Profiles supported by Stereo Module
 Supported modes:
    stream       resolution      fps       format
    Infrared 1    1280x800      @ 30Hz     Y8
    Infrared 2    1280x800      @ 30Hz     Y8
    Infrared 1    1280x800      @ 25Hz     Y16
    Infrared 2    1280x800      @ 25Hz     Y16
    Infrared 1    1280x800      @ 15Hz     Y16
    Infrared 2    1280x800      @ 15Hz     Y16
    Infrared 1    1280x800      @ 15Hz     Y8
    Infrared 2    1280x800      @ 15Hz     Y8
    Infrared 2    1280x720      @ 30Hz     Y8
    Infrared 1    1280x720      @ 30Hz     Y8
    Infrared 2    1280x720      @ 15Hz     Y8
    Infrared 1    1280x720      @ 15Hz     Y8
    Infrared 2    1280x720      @ 6Hz      Y8
    Infrared 1    1280x720      @ 6Hz      Y8
    Infrared 2    848x480       @ 90Hz     Y8
    Infrared 1    848x480       @ 90Hz     Y8
    Infrared 2    848x480       @ 60Hz     Y8
    Infrared 1    848x480       @ 60Hz     Y8
    Infrared 1    848x480       @ 30Hz     Y8
    Infrared 2    848x480       @ 30Hz     Y8
    Infrared 2    848x480       @ 15Hz     Y8
    Infrared 1    848x480       @ 15Hz     Y8
    Infrared 2    848x480       @ 6Hz      Y8
    Infrared 1    848x480       @ 6Hz      Y8
    Infrared 1    640x480       @ 90Hz     Y8
    Infrared 2    640x480       @ 90Hz     Y8
    Infrared 1    640x480       @ 60Hz     Y8
    Infrared 2    640x480       @ 60Hz     Y8
    Infrared 2    640x480       @ 30Hz     Y8
    Infrared 1    640x480       @ 30Hz     Y8
    Infrared 1    640x480       @ 15Hz     Y8
    Infrared 2    640x480       @ 15Hz     Y8
    Infrared 2    640x480       @ 6Hz      Y8
    Infrared 1    640x480       @ 6Hz      Y8
    Infrared 1    640x400       @ 25Hz     Y16
    Infrared 2    640x400       @ 25Hz     Y16
    Infrared 2    640x400       @ 15Hz     Y16
    Infrared 1    640x400       @ 15Hz     Y16
    Infrared 2    640x360       @ 90Hz     Y8
    Infrared 1    640x360       @ 90Hz     Y8
    Infrared 1    640x360       @ 60Hz     Y8
    Infrared 2    640x360       @ 60Hz     Y8
    Infrared 2    640x360       @ 30Hz     Y8
    Infrared 1    640x360       @ 30Hz     Y8
    Infrared 1    640x360       @ 15Hz     Y8
    Infrared 2    640x360       @ 15Hz     Y8
    Infrared 1    640x360       @ 6Hz      Y8
    Infrared 2    640x360       @ 6Hz      Y8
    Infrared 1    480x270       @ 90Hz     Y8
    Infrared 2    480x270       @ 90Hz     Y8
    Infrared 1    480x270       @ 60Hz     Y8
    Infrared 2    480x270       @ 60Hz     Y8
    Infrared 1    480x270       @ 30Hz     Y8
    Infrared 2    480x270       @ 30Hz     Y8
    Infrared 1    480x270       @ 15Hz     Y8
    Infrared 2    480x270       @ 15Hz     Y8
    Infrared 2    480x270       @ 6Hz      Y8
    Infrared 1    480x270       @ 6Hz      Y8
    Infrared 2    424x240       @ 90Hz     Y8
    Infrared 1    424x240       @ 90Hz     Y8
    Infrared 1    424x240       @ 60Hz     Y8
    Infrared 2    424x240       @ 60Hz     Y8
    Infrared 2    424x240       @ 30Hz     Y8
    Infrared 1    424x240       @ 30Hz     Y8
    Infrared 2    424x240       @ 15Hz     Y8
    Infrared 1    424x240       @ 15Hz     Y8
    Infrared 2    424x240       @ 6Hz      Y8
    Infrared 1    424x240       @ 6Hz      Y8
    Depth         1280x720      @ 30Hz     Z16
    Depth         1280x720      @ 15Hz     Z16
    Depth         1280x720      @ 6Hz      Z16
    Depth         848x480       @ 90Hz     Z16
    Depth         848x480       @ 60Hz     Z16
    Depth         848x480       @ 30Hz     Z16
    Depth         848x480       @ 15Hz     Z16
    Depth         848x480       @ 6Hz      Z16
    Depth         640x480       @ 90Hz     Z16
    Depth         640x480       @ 60Hz     Z16
    Depth         640x480       @ 30Hz     Z16
    Depth         640x480       @ 15Hz     Z16
    Depth         640x480       @ 6Hz      Z16
    Depth         640x360       @ 90Hz     Z16
    Depth         640x360       @ 60Hz     Z16
    Depth         640x360       @ 30Hz     Z16
    Depth         640x360       @ 15Hz     Z16
    Depth         640x360       @ 6Hz      Z16
    Depth         480x270       @ 90Hz     Z16
    Depth         480x270       @ 60Hz     Z16
    Depth         480x270       @ 30Hz     Z16
    Depth         480x270       @ 15Hz     Z16
    Depth         480x270       @ 6Hz      Z16
    Depth         424x240       @ 90Hz     Z16
    Depth         424x240       @ 60Hz     Z16
    Depth         424x240       @ 30Hz     Z16
    Depth         424x240       @ 15Hz     Z16
    Depth         424x240       @ 6Hz      Z16

Stream Profiles supported by RGB Camera
 Supported modes:
    stream       resolution      fps       format
RealSense error calling rs2_get_stream_profiles(sensor:00000213C9C8EB90):
    MFCreateDeviceSource(_device_attrs, &_source) returned: HResult 0x800706be: "The remote procedure call failed."

RealSenseCustomerSupport, 2019-11-12:
----------------------------------------------

Hi @Kylemaycry,

1. It looks some data corrupted. Please also run "rs-enumerate-devices -c".
2. Your D435i firmware version still show 5.10.13.0. It's not upgraded to 5.11.15.0 successfully.

Kylemaycry, 2019-11-12:
Because I tried to downgrade it to recover it. It can be upgraded to the
latest version, but still doesn't work.

Kylemaycry, 2019-11-12:
[rs-enumerate-devices -c.txt](https://github.com/IntelRealSense/librealsense/files/3837372/rs-enumerate-devices.-c.txt)

RealSenseCustomerSupport, 2019-11-15:
----------------------------------------------

Hi @Kylemaycry,

Not see the related issue in the calibration data. Since the fw downgrade/upgrade can't resolve the issue, you can check your reseller about RMA replacement.

RealSenseCustomerSupport, 2019-12-26:
----------------------------------------------

Hi @Kylemaycry,

Has there been an update to this situation?

Kylemaycry, 2019-12-30:
Hi,

Thanks for your asking, I was not able to make the RAM replacement, because
it is an academic purchase, the camera was purchased by the
department secretary for research purposes.

There is not any update.

RealSenseCustomerSupport, 2019-12-30:
----------------------------------------------

No problem...have you tested out on a different D435i camera from your department?

Kylemaycry, 2019-12-31:
Still not able to purchase a new one. I will try. Thanks.

RealSenseCustomerSupport, 2019-12-31:
----------------------------------------------

I think we should close this thread out until you're able to get a new device and can test it out at that time.

closed 2020-01-23 by RealSenseCustomerSupport