realsenseai/librealsense #5140: ‘RS2_OPTION_ENABLE_POSE_JUMPING’  ‘RS2_OPTION_ENABLE_RELOCALIZATION error when building with make

tmichals, 2019-10-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T235 | 
| Firmware Version                   | RealSense ROS | 
| Operating System & Version |  Ubuntu 18.04 LTS | 
| Kernel Version (Linux Only)    |  Linux 4.9.140-tegra aarch64                                    | 
| Platform                                 | NVIDIA Jetson|
| SDK Version                            |    JetPack 3.3.1           | 

I am trying to build trabmap https://github.com/introlab/rtabmap_ros on Step 9. >>>>
When I try building rtabmap.... I get the following RealSense errors....

###  error: 
++++++++++++++++++++++++++++++++++++++++++++++++
tom@tx2:~/rtabmap/build$ make
[  6%] Built target rtabmap_utilite
[  6%] Built target res_tool
[  6%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraRealSense2.cpp.o
/home/lcs/rtabmap/corelib/src/camera/CameraRealSense2.cpp: In member function ‘virtual bool rtabmap::CameraRealSense2::init(const string&, const string&)’:
/home/lcs/rtabmap/corelib/src/camera/CameraRealSense2.cpp:547:42: error: ‘RS2_OPTION_ENABLE_POSE_JUMPING’ is not a member of ‘rs2_option’
    sensors.back().set_option(rs2_option::RS2_OPTION_ENABLE_POSE_JUMPING, 0);
                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/lcs/rtabmap/corelib/src/camera/CameraRealSense2.cpp:548:42: error: ‘RS2_OPTION_ENABLE_RELOCALIZATION’ is not a member of ‘rs2_option’
    sensors.back().set_option(rs2_option::RS2_OPTION_ENABLE_RELOCALIZATION, 0);
                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:547: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraRealSense2.cpp.o' failed
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/camera/CameraRealSense2.cpp.o] Error 1
CMakeFiles/Makefile2:267: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

Any ideas where I need to make changes?  I know that these https://github.com/IntelRealSense/librealsense/wiki/API-Changes where made for the T265 but they seem to create a problem with the rtabmap build.

dorodnic, 2019-11-11:
Seems like a mismatch between ROS wrapper version and installed librealsense version (installed version seem to be older than the one required by the wrapper)
I would uninstall any version of librealsense and the wrapper, clone/install both according to instructions on the ROS wrapper repo and try again.

RealSenseCustomerSupport, 2019-12-06:
----------------------------------------------

Hi,
Have you tried dorodnic's suggestion. Assuming no news is good news?

Thanks

tmichals, 2019-12-07:
Yes, this is from my memory. I platformed my TX2 to a clean install for the Jetpack flash.   Which installed Ubuntu 18.04.  I then "tried to" installed ROS Kenetic, which for the most part installed, but it had issues.  Then I put a full sucessfull install ROS Melodic.  Then I was able to install rtabmap.  All the specifics are in the attached file.  
My goal is to link the RealSense topics on a Nano ---> to rtabmap on a TX2.  I may have to install the RealSense wrapper on the TX2(like the Nano) and connect the cameras direct.  Issue is that tx2 only has 1 usb port so then I have to have a usb hub...rather use a Nano because it has 4 usb ports only uses 5v 3amps.  Curently rtabmap does not seem to have away to attach to ROS topics...but I am not sure right now.

 
 

  
-- Performing Test HAVE_CXX_WFORMAT
-- Performing Test HAVE_CXX_WFORMAT - Success
-- Performing Test HAVE_C_WFORMAT
-- Performing Test HAVE_C_WFORMAT - Success
-- Performing Test HAVE_CXX_WERROR_FORMAT_SECURITY
-- Performing Test HAVE_CXX_WERROR_FORMAT_SECURITY - Success
-- Performing Test HAVE_C_WERROR_FORMAT_SECURITY
-- Performing Test HAVE_C_WERROR_FORMAT_SECURITY - Success
-- Performing Test HAVE_CXX_WMISSING_DECLARATIONS
-- Performing Test HAVE_CXX_WMISSING_DECLARATIONS - Success
-- Performing Test HAVE_C_WMISSING_DECLARATIONS
-- Performing Test HAVE_C_WMISSING_DECLARATIONS - Success
-- Performing Test HAVE_CXX_WMISSING_PROTOTYPES
-- Performing Test HAVE_CXX_WMISSING_PROTOTYPES - Failed
-- Performing Test HAVE_C_WMISSING_PROTOTYPES
-- Performing Test HAVE_C_WMISSING_PROTOTYPES - Success
-- Performing Test HAVE_CXX_WSTRICT_PROTOTYPES
-- Performing Test HAVE_CXX_WSTRICT_PROTOTYPES - Failed
-- Performing Test HAVE_C_WSTRICT_PROTOTYPES
-- Performing Test HAVE_C_WSTRICT_PROTOTYPES - Success
-- Performing Test HAVE_CXX_WUNDEF
-- Performing Test HAVE_CXX_WUNDEF - Success
-- Performing Test HAVE_C_WUNDEF
-- Performing Test HAVE_C_WUNDEF - Success
-- Performing Test HAVE_CXX_WINIT_SELF
-- Performing Test HAVE_CXX_WINIT_SELF - Success
-- Performing Test HAVE_C_WINIT_SELF
-- Performing Test HAVE_C_WINIT_SELF - Success
-- Performing Test HAVE_CXX_WPOINTER_ARITH
-- Performing Test HAVE_CXX_WPOINTER_ARITH - Success
-- Performing Test HAVE_C_WPOINTER_ARITH
-- Performing Test HAVE_C_WPOINTER_ARITH - Success
-- Performing Test HAVE_CXX_WSHADOW
-- Performing Test HAVE_CXX_WSHADOW - Success
-- Performing Test HAVE_C_WSHADOW
-- Performing Test HAVE_C_WSHADOW - Success
-- Performing Test HAVE_CXX_WSIGN_PROMO
-- Performing Test HAVE_CXX_WSIGN_PROMO - Success
-- Performing Test HAVE_C_WSIGN_PROMO
-- Performing Test HAVE_C_WSIGN_PROMO - Failed
-- Performing Test HAVE_CXX_WUNINITIALIZED
-- Performing Test HAVE_CXX_WUNINITIALIZED - Success
-- Performing Test HAVE_C_WUNINITIALIZED
-- Performing Test HAVE_C_WUNINITIALIZED - Success
-- Performing Test HAVE_CXX_WNO_DELETE_NON_VIRTUAL_DTOR
-- Performing Test HAVE_CXX_WNO_DELETE_NON_VIRTUAL_DTOR - Success
-- Performing Test HAVE_C_WNO_DELETE_NON_VIRTUAL_DTOR
-- Performing Test HAVE_C_WNO_DELETE_NON_VIRTUAL_DTOR - Failed
-- Performing Test HAVE_CXX_WNO_UNNAMED_TYPE_TEMPLATE_ARGS
-- Performing Test HAVE_CXX_WNO_UNNAMED_TYPE_TEMPLATE_ARGS - Failed
-- Performing Test HAVE_C_WNO_UNNAMED_TYPE_TEMPLATE_ARGS
-- Performing Test HAVE_C_WNO_UNNAMED_TYPE_TEMPLATE_ARGS - Failed
-- Performing Test HAVE_CXX_WNO_COMMENT
-- Performing Test HAVE_CXX_WNO_COMMENT - Success
-- Performing Test HAVE_C_WNO_COMMENT
-- Performing Test HAVE_C_WNO_COMMENT - Success
-- Performing Test HAVE_CXX_WIMPLICIT_FALLTHROUGH_3
-- Performing Test HAVE_CXX_WIMPLICIT_FALLTHROUGH_3 - Success
-- Performing Test HAVE_C_WIMPLICIT_FALLTHROUGH_3
-- Performing Test HAVE_C_WIMPLICIT_FALLTHROUGH_3 - Success
-- Performing Test HAVE_CXX_WNO_STRICT_OVERFLOW
-- Performing Test HAVE_CXX_WNO_STRICT_OVERFLOW - Success
-- Performing Test HAVE_C_WNO_STRICT_OVERFLOW
[... 764 lines cut ...]
[ 21%] Linking CXX static library ../../lib/libopencv_freetype_pch_dephelp.a
[ 21%] Built target opencv_freetype_pch_dephelp
Scanning dependencies of target opencv_hfs_pch_dephelp
[ 21%] Building CXX object modules/hfs/CMakeFiles/opencv_hfs_pch_dephelp.dir/opencv_hfs_pch_dephelp.cxx.o
[ 21%] Linking CXX static library ../../lib/libopencv_features2d_pch_dephelp.a
[ 21%] Built target opencv_features2d_pch_dephelp
Scanning dependencies of target opencv_img_hash_pch_dephelp
[ 21%] Building CXX object modules/img_hash/CMakeFiles/opencv_img_hash_pch_dephelp.dir/opencv_img_hash_pch_dephelp.cxx.o
[ 21%] Linking CXX static library ../../lib/libopencv_fuzzy_pch_dephelp.a
[ 21%] Built target opencv_fuzzy_pch_dephelp
Scanning dependencies of target opencv_imgcodecs_pch_dephelp
[ 21%] Building CXX object modules/imgcodecs/CMakeFiles/opencv_imgcodecs_pch_dephelp.dir/opencv_imgcodecs_pch_dephelp.cxx.o
[ 21%] Linking CXX static library ../../lib/libopencv_gapi_pch_dephelp.a
[ 21%] Built target opencv_gapi_pch_dephelp
Scanning dependencies of target opencv_line_descriptor_pch_dephelp
[ 21%] Building CXX object modules/line_descriptor/CMakeFiles/opencv_line_descriptor_pch_dephelp.dir/opencv_line_descriptor_pch_dephelp.cxx.o
[ 22%] Linking CXX static library ../../lib/libopencv_hfs_pch_dephelp.a
[ 22%] Built target opencv_hfs_pch_dephelp
Scanning dependencies of target opencv_photo_pch_dephelp
[ 22%] Building CXX object modules/photo/CMakeFiles/opencv_photo_pch_dephelp.dir/opencv_photo_pch_dephelp.cxx.o
[ 22%] Linking CXX static library ../../lib/libopencv_img_hash_pch_dephelp.a
[ 22%] Built target opencv_img_hash_pch_dephelp
Scanning dependencies of target opencv_saliency_pch_dephelp
[ 22%] Building CXX object modules/saliency/CMakeFiles/opencv_saliency_pch_dephelp.dir/opencv_saliency_pch_dephelp.cxx.o
[ 22%] Linking CXX static library ../../lib/libopencv_imgcodecs_pch_deph

tmichals, 2019-12-26:
I used the latest JetsonHacks Nano installs on my TX2 and everything appears to be working now.  I flashed the TX2 with JetPack 4.3 L4T 32.3.1, then used JetsonHacks scripts for RealSense, ROS, ROS RealSense Wrapper... everything seems to be working well.   Installing is very easy now.

closed 2019-12-26 by tmichals