realsenseai/librealsense #5142: Could not initialize offscreen context!

xingguang95, 2019-10-28:
| Camera Model                       | D435  | 
| Operating System & Version | Ubuntu 16 |                                   
| Platform                                 | Raspberry Pi |

Hi,
I have an Intel RealSense depth camera D435 connecting to the raspberry pi. I could run RealSense viewer and get image from the camera. And I am using VNC(Virtual Network Computing) to remote control the raspberry pi. But when I run the RealSense viewer in the VNC viewer, I get the error: Could not initialize offscreen context!
Thanks in advance for the help.
Xingguang

closed 2019-10-29 by xingguang95

Wats0ns, 2019-11-04:
Hey @xingguang95, if you cold post how you solved this, this would be of great help !
Thanks a lot !

pojashad, 2020-05-18:
Hi i have the same issue, what did you do to solve this?

wdfalfred, 2020-07-21:
I have the same issue. I am using Raspberry Pi 4 with D435i. 

I tried to access the Raspberry Pi remotely with the method in the following link.

> 1. https://electricnoodlebox.com/diy-pages/remote-view-your-raspberry-pi-on-mac-using-vnc/
> 2..https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435
> 

In the first link, "tightvncserver" is used for the remote control. And the instruction says that "sudo service lightdm stop", which I think will stop the GUI. Maybe, that is why the error mentions "offscreen".

When I tried the method in the second link, using "VNC Viewer by RealVNC",  the issue disappeared. I think this may be the solution.

I hope this could solve the issue.