realsenseai/librealsense #5143: Wrong timestamp using frame callback

ahuizxc, 2019-10-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Firmware Version                   | 05.11.11.100 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |4.15.0-66-generic | 
| Platform                                 | PC|
| SDK Version                            |2.28| 
| Language                            |  {C/python}                          | 
| Segment			| | 

### Issue Description
Hi all, when i was using realsense camera, I found that the timestamp from the realsense was incorrect. Sometimes the timestamp from two frame are same. Firstly I assume that the camera just given two same frame when the timestamp are same, but it not like this.

here is my code:
```python

from __future__ import print_function

import pyrealsense2 as rs
import cv2
import numpy as np
from math import tan, pi

from threading import Lock
frame_mutex = Lock()
frame_data = {"depth"  : None,
              "color" : None,
              "timestamp_ms" : None
              }

"""
This callback is called on a separate thread, so we must use a mutex
to ensure that data is synchronized properly. We should also be
careful not to do much work on this thread to avoid data backing up in the
callback queue.
"""

time_begin = 0
frame_num_begin = 0
color_img_before = np.empty(shape=None)
def callback(frame):
    global frame_data
    global time_begin
    global frame_num_begin
    global color_img_before
    if frame.is_frameset():
        frameset = frame.as_frameset()
        f1 = frameset.get_depth_frame()
        f2 = frameset.get_color_frame()
        depth_data = np.asanyarray(f1.get_data())
        color_data = np.asanyarray(f2.get_data())

        ts = frameset.get_timestamp()
        frame_mutex.acquire()
        frame_data["depth"] = depth_data
        frame_data["color"] = color_data
        frame_data["timestamp_ms"] = ts
        span = ts-time_begin
        if span == 0:
            diff = cv2.subtract(color_img_before, color_data)
            if not np.any(diff):
                print("the two imgs are same.")
            print(np.sum(diff))
            print("frame now: "+str(frameset.get_frame_number()))
            print("frame last: "+str(frame_num_begin))
        time_begin = ts
        color_img_before = color_data
        frame_num_begin = frameset.get_frame_number()
        frame_mutex.release()

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
frame_mutex = Lock()
pipeline.start(config, callback)

try:
    WINDOW_TITLE = 'Realsense'
    cv2.namedWindow(WINDOW_TITLE, cv2.WINDOW_NORMAL)
    while True:
        frame_mutex.acquire()
        valid = frame_data["depth"] is not None
        frame_mutex.release()

        if valid:

            frame_mutex.acquire()
            frame_copy = {"depth"  : frame_data["depth"].copy(),
                          "color" : frame_data["color"].copy()}
            frame_mutex.release()

            depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(frame_copy['depth'], alpha=0.03), cv2.COLORMAP_JET)
            cv2.imshow(WINDOW_TITLE, np.hstack((depth_colormap, frame_copy['color'])))

        key = cv2.waitKey(1)
        if key == ord('q') or cv2.getWindowProperty(WINDOW_TITLE, cv2.WND_PROP_VISIBLE) < 1:
            break
finally:
    pipeline.stop()
```
here is the log:
```bash
hui@hui-desktop:~/librealsense/wrappers/python/examples$ python test_callback_d435.py 
6823991
frame now: 1315
frame last: 1315
3421861
frame now: 2110
frame last: 2110
1534174
frame now: 2265
frame last: 2265
1241201
frame now: 3868
frame last: 3868
^CTraceback (most recent call last):
  File "test_callback_d435.py", line 83, in <module>
    key = cv2.waitKey(1)
KeyboardInterrupt
```
Thanks :)

ev-mp, 2019-10-28:
@ahuizxc , the pipeline callback generates a synchronized bundle of video frames (as opposed to "raw" sensor callback API). Hence certain frames will be repeated by design in order to provide coherent framesets. 
See [#3817]( https://github.com/IntelRealSense/librealsense/issues/3817#issuecomment-485663496)

ahuizxc, 2019-10-29:
> @ahuizxc , the pipeline callback generates a synchronized bundle of video frames (as opposed to "raw" sensor callback API). Hence certain frames will be repeated by design in order to provide [...]

But the frames was not repeated because the timestamp was same but the data was different, it may cause some errors for the future usage.

ev-mp, 2019-10-29:
@ahuizxc , to clarify - in the above sample it wasn't the case for the timestamps being repeated, it were the frames. 
As highlighted above, the same frames may appear in more than one `wait_for_frames` call by design.
In case a different behavior is envisaged you can use alternative APIs #5107 .

ahuizxc, 2019-10-30:
> call by

I am still confused because the fact I get two color frame at the same  time but the images are different, so I think there must be an wrong image. Otherwise, if I use the camera for motion estimation, which image should I use?

ev-mp, 2019-10-30:
@ahuizxc , that is an implementation issue - In the above snippet you compare the RGB frames:
>diff = cv2.subtract(color_img_before, color_data)
            if not np.any(diff):

while tracking and trying to match those with the timestamps of the Depth frames:
>ts = frameset.get_timestamp()
...

The `frameset` is a container for multiple frames in many ways resembling `std::vector`. When you query its timestamp the container implicitly extracts this attribute from the first frame embedded, which is Depth by design.

This inconsistency can be rectified by extracting the timestamp from the actual RGB frame rather than `frameset` :
>f1 = frameset.get_depth_frame()
        f2 = frameset.get_color_frame()
...
        ts = ~~frameset~~**f2**.get_timestamp()
```

RealSenseCustomerSupport, 2019-11-11:
----------------------------------------------

@ahuizxc Did you get it through with the explanation from ev-mp? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-11-18:
----------------------------------------------

@ahuizxc Could you please update? Thanks!

ahuizxc, 2019-11-18:
> @ahuizxc Could you please update? Thanks!

Got it, thanks for your help:)

closed 2019-11-18 by ahuizxc

tispratik, 2020-06-01:
We had faced the issue of same timestamps for consecutive frames upto 8 frames. We internally referred to it as "stale frames" issue, mainly because not just the timestamp was same, the entire frame was duplicate and the camera was doling out the same frame again and again. Upgrading the firmware helped fix that issue.