realsenseai/librealsense #5145: dmesg Failed to query (GET_CUR) errors when runnning realsense-viewer

jruizaranguren, 2019-10-28:
I have found and read a couple of issues related to "Failed to query" problem, but most of them are closed and none of them is reported with latests versions of Kernel patches.

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | 05.11.15.00 | 
| Operating System & Version |   Ubuntu 18.04   | 
| Kernel Version (Linux Only)    |   4.15.0-66-generic     | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.29 }                          | 
| Language                            |  python }                          | 
| Segment			|  others                          | 

### Issue Description

I have followed the [instructions provided here](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md). I have compiled the library myself because I'm planning to use a Jetson TX2i, and it is required to enable alignment acceleration with CUDA:

`cmake ../ -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=Release -DBUILD_WITH_CUDA=true -DBUILD_PYTHON_DOCS=true`

After full install, and executing realsense-viewer, everything seems to work fine, but I'm getting about 200+ lines of errors in `dmesg` each time I start capturing from the cameras. Something like (I have omitted repeated messages):

```
[ 8062.195867] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 8919.994163] uvcvideo: Failed to query (SET_CUR) UVC control 4 on unit 1: -32 (exp. 4).
[ 8925.055347] uvcvideo: Failed to query (SET_CUR) UVC control 10 on unit 3: -32 (exp. 2).
```

Can this bug be solved?

RealSenseCustomerSupport, 2020-02-11:
----------------------------------------------

Hi @jruizaranguren,

Please invoke "-DFORCE_LIBUVC=true" flag when you are building the library from sources. This flag provides a user-space implementation version of kernel patches.
Please let us know if you still experience the same problem or if you need any further assistance with this matter.

Thank you!

closed 2020-02-18 by RealSenseCustomerSupport

lucbettaieb, 2020-10-21:
@RealSenseCustomerSupport 
This occurs when the kernel patch is installed through apt in the dkms package, too.

doisyg, 2020-11-14:
Same on kernel 5.4 Ubuntu with patches applied