realsenseai/librealsense #5146: Align code source

95ImadEL, 2019-10-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D435 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 18  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            | C++                         | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hello everyone,

I'd like to apply alignment function on a png dataset, I'm wondering if it is this function
 ```
void align_images(const rs2_intrinsics& depth_intrin, const rs2_extrinsics& depth_to_other,
        const rs2_intrinsics& other_intrin, GET_DEPTH get_depth, TRANSFER_PIXEL transfer_pixel)

```
who perform alignment between RGB and depth frames. Thank you in  advance.

ev-mp, 2019-10-29:
@95ImadEL , the function you referred to is used to aligns the images internally. It is a templated function invoked by external code, so I'm not sure that it is advisable to call this function directly. More probably you'd need to "zoom out" into a higher resolution API, like looking into "rs2::align" class can APIs.

For me the fundamental question is that whether you have all the artifacts required for alignment - i.e. the content of PNG files allows to reproduce the original depth and that you also have the corresponding camera's  intrinsic/extrinsic data stored.
Follow a similar #2231 for details

95ImadEL, 2019-10-29:
Thank you @ev-mp for your reply. In fact, I have depth images 16-bits PNG format, and camera's  intrinsic/extrinsic. I 'd like to get the depth value for an RGB pixel. I found also this function.

```
static void rs2_project_color_pixel_to_depth_pixel(float to_pixel[2],
    const uint16_t* data, float depth_scale,
    float depth_min, float depth_max,
    const struct rs2_intrinsics* depth_intrin,
    const struct rs2_intrinsics* color_intrin,
    const struct rs2_extrinsics* color_to_depth,
    const struct rs2_extrinsics* depth_to_color,
    const float from_pixel[2])
{

```
 I was wondering if I can use it to align the whole image. Thank you

closed 2019-11-19 by dorodnic