realsenseai/librealsense #5147: Kernel patches will need UVC_INFO_QUIRK

dankamongmen, 2019-10-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 5.11.15.0 | 
| Operating System & Version |   Debian Unstable  | 
| Kernel Version (Linux Only)    |  5.3.7         | 
| Platform                                 | ATX x86 |
| SDK Version                            |  2.29.0                         | 
| Language                            |  C                     | 
| Segment			|  All                | 

### Issue Description
I've been keeping a copy of the Realsense kernel patches up-to-date against mainline. Recently, I started getting an OOPs on device attach. I traced this down to the USB device definitions, where `UVC_QUIRK_APPEND_UVC_HEADER` is being directly stashed into `.driver_info`. The 5.x series uvc model requires that this field be a pointer to a uvc device definition. The `UVC_INFO_QUIRK` macro is provided to facilitate this. By changing all `.driver_info = UVC_QUIRK_APPEND_UVC_HEADER` to `.driver_info = UVC_INFO_QUIRK(UVC_QUIRK_APPEND_UVC_HEADER)`, the problem is resolved.

My current diff can be found at
[https://www.dsscaw.com/resources/realsense-5.3.7.diff]https://www.dsscaw.com/resources/realsense-5.3.7.diff).

If you'd like to take this into the project, I can set up a proper PR.

seebq, 2019-10-28:
As beneficiary of this, I won't pile on or +1, as we need others to validate, but can confirm we're using this.

ev-mp, 2019-10-29:
@dankamongmen , a similar adjustment has already been incorporated into LTS kernel v5.0 patches supported by SDK - 
https://github.com/IntelRealSense/librealsense/blob/master/scripts/realsense-metadata-bionic-hwe.patch
>.driver_info		= UVC_INFO_META(V4L2_META_FMT_D4XX) },

Invoke the patches script `./scripts/patch-realsense-ubuntu-lts.sh` with kernel 5.0 to pickup this implementation.

The implementation that you provide seem to be more generic, so can you confirm that you read all the metadata attributes and not only the header correctly ?

dankamongmen, 2019-10-29:
I may well be missing some -- I definitely get some dmesg output about uninitialized entity types (Processing 2 and 7, Camera 1 and 6, and Extension 8). I furthermore see some "failed to query UVC control 1 on unit 3", so it's quite possible. realsense-viewer seems to be functioning. How can I verify correct metadata attribute processing? Thank you for your time.

ev-mp, 2019-11-10:
@dankamongmen , you can check the metadata attributes in `realsense-viewer`:
[image]

The warning `failed to query UVC control 1 on unit 3` can be disregarded as SDK handles it internally.

dankamongmen, 2019-11-10:
Thanks, I should be able to look into this tomorrow (Monday 2019-11-11).

RealSenseCustomerSupport, 2019-12-05:
----------------------------------------------

Hi @dankamongmen,

Any update? Is the issue clarified and resolved?

closed 2019-12-26 by RealSenseCustomerSupport