realsenseai/librealsense #5149: I want calibration method of multiple realsense

devLupin, 2019-10-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | latest | 
| Operating System & Version |   Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  latest                           | 
| Language                            |  C++ or C#                          | 
| Segment			|  others                          | 

### Issue Description
<Describe your question>
I'm planning a system that measures values ​​using multiple realsense. This task requires calibration of multiple realsense. please recommend the most accurate way possible

I would be grateful if you could introduce a method or API for this.
Thank you

MartyG-RealSense, 2019-10-29:
Intel offer a calibration system capable of multiple camera calibration in the RealSense store, though it costs $1500 USD due to the cost of manufacturing the large physical chessboard.  It has a special OEM version of the Dynamic Calibrator software that can calibrate both intrinsics and extrinsics (the standard free public version only calibrates extrinsics).

https://store.intelrealsense.com/buy-intel-realsense-d400-cameras-calibration-target.html

Alternatively, multiple 400 Series cameras can be calibrated using a program called Vicalib, which Intel themselves have used.  It uses a board that you show to each of the cameras in turn., and it establishes overlapping regions to then minimize the pose of each of those together

https://github.com/arpg/vicalib

devLupin, 2019-10-29:
Thank you @MartyG-RealSense 

What is the difference between the two accuracy?
Must be accurate because the company will develop it.

But $ 1500 is so expensive, please let me know how the accuracy is different.

MartyG-RealSense, 2019-10-29:
I do not have any data of the results of a Dynamic Calibrator calibration compared to a Vicalib one.I assume that Vicalib is at least good enough for it to have been recommended by Intel RealSense staff for the purposes of multi-cam calibration (during their 2018 online webinar session about multiple cameras).

The Conix Research Center at Carnegie Mellon developed a multicam point cloud stitching system that could use 10-20 400 Series cameras.  Their documentation says that the calibration method that they used was to "use a Theodolite, a survey precision instrument, to obtain real world coordinates of each camera".  Unfortunately, the cheapest laser theodolite I could see on Amazon was $500, with the average price around $1000-1200.

https://github.com/conix-center/pointcloud_stitching

In conclusion, unless you are doing a project that involves stitching multiple point clouds together into a single mega cloud, it may be easiest and cheapest to just calibrate each camera individually with the standard free Dynamic Calibrator tool to help ensure that you are getting excellent results from them.

RealSenseCustomerSupport, 2019-11-19:
----------------------------------------------

Hi devLupin,

The Dynamic Calibration only optimizes the extrinsic parameters where the OEM Calibration does both extrinsic and intrinsic parameters.
I believe currently the Vicalib is not supported on D400 camera anymore, it is for T265, the tracking camera.

Thanks

**Please be informed that we will close this ticket in seven (7) days if we have not heard back from you.

devLupin, 2019-11-19:
@RealSenseCustomerSupport 
Thank you.

Does the free tool Vicalib no longer support my 435 camera?
Do I have to buy the dynamic calibrator mentioned above?

RealSenseCustomerSupport, 2019-11-19:
----------------------------------------------

Hi devLupin

Our Calibration Tool development engineer says we don't support Vicalib on D4xx. The "Dynamic Calibration" tool is FREE to users. https://downloadcenter.intel.com/download/28517/Intel-RealSense-D400-Series-Calibration-Tools-and-API

Thanks

**Please be informed that we will close this ticket in seven (7) days if we have not heard back from you.

devLupin, 2019-11-19:
@RealSenseCustomerSupport 
Thanks

closed 2019-11-24 by RealSenseCustomerSupport

monajalal, 2025-12-15:
> Intel offer a calibration system capable of multiple camera calibration in the RealSense store, though it costs $1500 USD due to the cost of manufacturing the large physical chessboard. It has a [...]

Hi @MartyG-RealSense  the link doesn't work anymore. How can we calibrate two Intel RealSense cameras with each other so we can receive posed images? I need to use these posed images for 3D scene reconstruction.