realsenseai/librealsense #5150: D435i on odroid get bus error

AlmostNoMemory, 2019-10-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435I| 
| Firmware Version                   | 05.11.15.00| 
| Operating System & Version |   {Ubuntu mate for odroid 18.04 | 
| Kernel Version (Linux Only)    |  4.14.141-169                                 | 
| Platform                                 | Odroid xu4 |

### Issue Description
<Describe your issue / question / feature request / etc..>

Hi , developer. I was using D435i on odroid xu4 board, when i tried some example such as rs-capture or rs-hello-realsense then i get error like this:
 
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to open exynos (search paths /usr/lib/arm-linux-gnueabihf/dri:\$${ORIGIN}/dri:/usr/lib/dri)
libGL error: failed to load driver: exynos
Bus error

but when i use T265 there is no Bus error and i  got it work very well, so i think the main issue here is the Bus error and i was wondering if you guys can help me solve this problem, i will be really appreciated.
I have run the scripts patch-realsense-ubuntu-odroid-xu4-4.14.sh and succeed which means my driver seems ok. My cmake version is 3.8.0-rc2 and gcc version is 5.5.0
For more information here is what i get when i try

lsusb -t 
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Vendor Specific Class, Driver=r8152, 5000M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/2p, 5000M
        |__ Port 2: Dev 6, If 0, Class=Video, Driver=uvcvideo, 5000M
        |__ Port 2: Dev 6, If 1, Class=Video, Driver=uvcvideo, 5000M
        |__ Port 2: Dev 6, If 2, Class=Video, Driver=uvcvideo, 5000M
        |__ Port 2: Dev 6, If 3, Class=Video, Driver=, 5000M
        |__ Port 2: Dev 6, If 4, Class=Video, Driver=, 5000M
        |__ Port 2: Dev 6, If 5, Class=Human Interface Device, Driver=, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/2p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=exynos-ohci/3p, 12M
    |__ Port 1: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 1: Dev 4, If 1, Class=Human Interface Device, Driver=usbhid, 12M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=exynos-ehci/3p, 480M

dmesg
[ 6117.099699] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[ 6117.100755] uvcvideo: Unable to create debugfs 4-6 directory.
[ 6117.101156] uvcvideo 4-1.2:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[ 6117.101166] uvcvideo 4-1.2:1.0: Entity type for entity Processing 2 was not initialized!
[ 6117.101172] uvcvideo 4-1.2:1.0: Entity type for entity Camera 1 was not initialized!
[ 6117.101556] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/soc/soc:usb3-0/12000000.dwc3/xhci-hcd.3.auto/usb4/4-1/4-1.2/4-1.2:1.0/input/input16
[ 6118.155680] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[ 6118.156567] uvcvideo: Unable to create debugfs 4-6 directory.
[ 6118.156928] uvcvideo 4-1.2:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[ 6118.156942] uvcvideo 4-1.2:1.0: Entity type for entity Processing 2 was not initialized!
[ 6118.156951] uvcvideo 4-1.2:1.0: Entity type for entity Camera 1 was not initialized!
[ 6118.157287] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/soc/soc:usb3-0/12000000.dwc3/xhci-hcd.3.auto/usb4/4-1/4-1.2/4-1.2:1.0/input/input17

rs-enumerate-devices -s 
Device Name                   Serial Number       Firmware Version    
Intel RealSense D435I         912112073985        05.11.15.00         
Device info: 
    Name                          : 	Intel RealSense D435I
    Serial Number                 : 	912112073985
    Firmware Version              : 	05.11.15.00
    Recommended Firmware Version  : 	05.11.15.00
    Physical Port                 : 	
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B3A
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	919423051288
    Firmware Update Id            : 	919423051288

RealSenseCustomerSupport, 2019-11-26:
----------------------------------------------

@AlmostNoMemory Did you install librealsense on odroid by DKMS or built from source code? Welcome to your contribution. Thanks!

ev-mp, 2019-12-01:
@AlmostNoMemory hello, the error that you mention:
>libGL error: failed to load driver: exynos
Bus error

Seem to be related to the openGl driver.
To verify whether the SDK is properly configured please build the SDK with 
```cmake .. ... -DBUILD_EXAMPLES=true -DBUILD_GRAPHIC_EXAMPLES=false```,
then run `rs-enumerate-devices`.

RealSenseCustomerSupport, 2020-01-07:
----------------------------------------------

@AlmostNoMemory Will close the issue accordingly if no other questions. Thanks!

closed 2020-01-07 by RealSenseCustomerSupport