realsenseai/librealsense #5151: No laptop, desktop available

devLupin, 2019-10-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | latest | 
| Operating System & Version |   Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Laptop |
| SDK Version                            |  latest                           | 
| Language                            |  C++ or C#                          | 
| Segment			|  others                          | 

### Issue Description
<My laptop Info>
CPU : Intel 8th i5-8250U
RAM : 16GB(DDR4)
Graphic : Intel UHD graphics 620
USB 3.0 port :  OK
+ 45W AC Power adapter

Using the realsense viewer gives you 100% share of the graphics card and doesn't get over the loading screen.

But desktops with CPUs with much less specs will run well.
I installed it the same way, but the laptop did not work and the desktop worked.

Is there a graphics minimum? Please let me know
In fact, I plan to connect it to a mini PC. No other features are required, the Realsense viewer works fine and you need to run the code in Visual Studio 2019.

Please provide the minimum hardware specifications for realsense. (CPU, Graphic card, RAM, etc)

MartyG-RealSense, 2019-10-29:
As far as I know, there isn't an official minimum specification.   The cameras can work with very low spec hardware.  To quote my favorite saying about the subject, "they could run on a potato" as long as it had a USB port and any Intel or ARM processor (there have been a couple of cases where the cameras have worked on AMD Threadripper processors, but Intel and ARM are the recommended processor types).

However, there is a difference between 'working' and 'working well'.  Whilst the camera may work with a low spec when using a single stream, if more than one stream is active at the same time or there is more than one camera on the same computing device, a stronger spec is recommended.  Having said that, your laptop's spec is very strong, apart from the built-in GPU.

There are methods by which you can offload some of the processor's graphics workload onto the computer's GPU instead of the CPU having to do the bulk of the processing. 

https://github.com/IntelRealSense/librealsense/issues/4905#issuecomment-533854888

RealSenseCustomerSupport, 2019-12-06:
----------------------------------------------

@MartyG-RealSense Thanks very much for your good share!
@devLupin For the issue on your laptop, please check if the windows version updated. Thanks!

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

@devLupin Did you still have issue on your laptop after windows version updated? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-12-23:
----------------------------------------------

@devLupin Any update from your side? Thanks!

devLupin, 2019-12-23:
@RealSenseCustomerSupport 
No problem! Thanks!

RealSenseCustomerSupport, 2019-12-27:
----------------------------------------------

@devLupin Glad to hear that. Will close this accordingly.

closed 2019-12-27 by RealSenseCustomerSupport