realsenseai/librealsense #5152: How to get 6dof pose prediction from a 2D (x,y) point?!

lucianoclugli, 2019-10-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (05.11.06.250) | 
| Operating System & Version |   {Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  (4.16.0-66)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { 2.24.0 }                          | 
| Language                            |  {opencv/python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description

Gentlemen,

I'm extracting a 2D (x,y) point from a vine's scene in RGB (to plot a potential cutting point on a specific branch). Using both D415/D435 cameras, I have the constraint that in some spur regions, each scene is acquired from a maximum distance (camera to vine) of 0.10m. This means that initially I'm not able to use the depth information, since the minimum distance for an accurate depth data, according to Intel website is: D415: 0.16m and D435: 0.11m (https://software.intel.com/en-us/realsense/d400).

Although, in practice this minimum distance (for both cameras) is greater than in theory (about 0.30m ~ 0.40m).

In order to use the depth, I need to move the camera away from the spur region, what is not suitable, because the spur region is constrained on a minimum distance of 0.10m (to avoid background issues and other branches interferences). 

So, I ask for help in two possible solutions:

01. How to use depth as a second information and the 2D point as a keypoint to match a correspondence coordinate, then computing the 6dof from this information? RANSAC maybe?!

02. Ignoring the depth, just using the previous 2D point information, how to compute the 6dof pose of that point?! DLT? Homography?!

The aim is to provide a reliable 6dof pose prediction of this 2D point, representing then the potential cutting point, so a robot manipulator shall approach to the vine's spur and perform the cut in that point. 

I appreciate any help...

If more information is needed, let me know...

agrunnet, 2019-10-29:
I will only answer one aspect: The minimum distance depends on the resolution. So for lowest resolution you get smallest minimum distance.

lucianoclugli, 2019-11-03:
Hi @agrunnet... the resolution must be reliable, considering details on the branches... I've computed some changes with lower resolution, but it also fails to segment relevant information from natural scenes... anyways, I appreciate, thanks...

RealSenseCustomerSupport, 2019-12-04:
----------------------------------------------

@lucianoclugli

I wonder if you have any progress about this use case?

closed 2020-02-05 by RealSenseCustomerSupport