realsenseai/librealsense #5153: T265 Firmware Crashing/Disconnecting

jfaulring, 2019-10-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  T265  | 
| Firmware Version                   | 0.1.0.279 | 
| Operating System & Version |  Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |  4.15.0-65                                         | 
| Platform                                 | PC |
| SDK Version                            |2.29.0-0 Realsense.1580                          | 
| Language                            |  C++                           | 
| Segment			| Robot                          | 

### Issue Description
I'm having an issue with the T265 where the device appears to be randomly crashing and rebooting. Using just the realsense-viewer I start streaming from the device, then after running for for anywhere from 1 to 5 minutes I see the following set of errors reported:

`11:36:11.556 [6704] [E] Device-69B0: FW crashed - got error in stream endpoint thread function: status = -1 (LIBUSB_ERROR_IO), actual = 0`

`11:36:11.584 [6703] [E] Device-69B0: FW crashed - got error in interrupt endpoint thread function: status = -1 (LIBUSB_ERROR_IO), actual = 0`

`11:36:11.756 [6682] [E] Device-69B0: State [ACTIVE_STATE] got event [ON_ERROR] ==> [ERROR_STATE]
11:36:11.756 [6682] [E] Device-69B0: Entered state [ERROR_STATE]`

I see the same/similar error if I try this under Windows as well. I have test this across multiple Linux host machines and various releases of the SDK & kernel driver with the same results. Streaming from the device can be restarted after the error occurs. It feels like the embedded side goes out to lunch and reboots such that the USB connection is renegotiated. I haven't been able to find mention of this behavior anywhere else - do I have a unit with a hardware issue?

boewing, 2019-10-30:
I have the same problem! I will buy a shorter cable and try again. Maybe there is a brownout.

jfaulring, 2019-10-30:
I went down that same path as well - it's failing with the cable that Intel included with the camera kit along with the high-quality Newnex cable that I bought for it. I also threw a powered USB 3.0 hub in the mix just in case the host adapter was insufficient to no avail as well (I've been running all of the tests on an Intel NUC that the camera has supposedly been qualified to work with).

anbello, 2019-10-30:
Same problems here. I am using T265 with Raspberry 3 and Ubuntu 16.04. I bought another short cable but I haven't the time to test it, even if from what I red it will be useless.

boewing, 2019-10-30:
Using a NVIDIA Jetson Nano

I solved the problem by disconnecting mouse and keyboard. Apparently it was a power deficit at the camera.

yputra, 2019-11-03:
I also have the same problem, I am using the Raspberry Pi 4, with Ubuntu 18.04, sometimes I can run like more than 20 minutes without problem, sometimes only few minutes (~5 minutes). 
Have been trying to find the solution, but some of the intel github software guys are still working on it for the next update or so

bfulkers-i, 2019-12-03:
This sounds like it is most likely a cable issue, but it's possible #5213 will help. You may with to retry with this PR or with a release once the PR is merged.

anbello, 2019-12-03:
Tested new cable, shortest then the Intel one, and now It Is Ok.

jfaulring, 2019-12-04:
I ended up RMAing the T265 camera and replacing it with another unit - I haven't had any issues since using the same computer and cables. It looks like it truly was faulty hardware.

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

Hi @jfaulring

I assume no further issues so this thread can be closed now?

closed 2020-02-10 by jfaulring