realsenseai/librealsense #5154: RealSense D435i - Incomplete frame on 5.11.15.0 when using USB2.1

theodop, 2019-10-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i | 
| Firmware Version                   | 5.11.15.0 | 
| Operating System & Version |   Ubuntu 18.10 | 
| Kernel Version (Linux Only)    |  4.18.0-25-generic                                        | 
| Platform                                 | PC + Rasberry Pi |
| SDK Version                            |  2                         | 
| Language                            |  C++                        | 
| Segment			|                           | 

### Issue Description
Since updating the firmware on my D435i to 5.11.15.0 to overcome the auto-exposure issues experienced in versions up to and including 5.11.11.100, I am experiencing issues with frames missing data when using USB 2.1. These issues do not occur when using USB3.2. The issues are more common when the camera is moving. The attached shows one frame captured which should be a clear shot of my head.

I have reproduced the issue using realsense-viewer on different ports on my laptop, with different cables, and also when using realsense-viewer on the Raspberry Pi 3 B+.

As a workaround, is there a firmware version which does not exhibit this issue but resolves the auto-exposure issue?

[image]

theodop, 2019-10-30:
I should note, sorry, that this issue is frequent enough for the device to be unusable until a workaround is found.

agrunnet, 2019-10-30:
What resolution and frame rate are you running?
Both color and depth are streaming?

theodop, 2019-10-30:
Heya, I'm running the default pipeline settings. Just depth is streaming.

In realsense-viewer, the issue only seems to occur for 640x480, 15fps, USB2.1. If I use the same settings in USB3.2 there is no issue. If I raise the framerate to 30 or lower to 6, or lower the resolution, the issue seems to go away.

In code, I am using:

p = new rs2::pipeline();
auto frame = p->wait_for_frames();

I will attempt to use a pipeline configuration in code.

theodop, 2019-10-30:
Further investigation shows that when running at any framerate, even though the issue of a completely blank frame may not occur, frames are not properly filled with depth information over USB2.1. Reverting to the earlier framework immediately resolves both issues.

theodop, 2019-10-30:
Earlier and later firmware with settings visible

5.11.11.100:
[image]

5.11.15.0:
[image]

agrunnet, 2019-10-30:
This actually kind of looks like the laser projector lost sync. When you say “incomplete frame” you are referring to the fact that depth is missing in the background, correct?  Not that frames are being dropped?

theodop, 2019-10-30:
As far as I can tell, no frames are being dropped. At best I get depth missing in the background, at worst I get almost a completely empty frame (as in the original post).

RealSenseCustomerSupport, 2019-11-11:
----------------------------------------------

Hi theodop
Thank you very much for the report.
Yes, the issue can be reproduced. This is firmware bug after issue (https://github.com/IntelRealSense/librealsense/issues/4022) was fixed. I can fix the issue by manually changing some register settings. So anyway, will have our firmware team fix it in next release or after.

Thanks

btravers1, 2019-11-15:
I ran into this same issue and did a bunch of testing. Here's what I found:

When connected over USB 2, some bug in the image processing firmware is severely degrading the dynamic range of the raw IR images. Consequently, disparity (depth) cannot be calculated in many types of scenes. This issue has similar results to the old "autoexposure lockup" issue. From what I can tell, there appears to be two bugs:
1. Firmware is not able to correctly set exposure and/or gain (it appears that it's getting set, but the setting isn't making it all the way to image sensor control)
2. It appears that a frame for HDR isn't being processed correctly in only USB2 mode. The impressive ability for the camera to pick up textures on USB3 seems to indicate that Intel is likely taking two frames to increase the dynamic range of the sensor. Essentially, this allows the camera to see things like texture on a concrete surface in full sunlight, for example.

Below is an example of this bug in sunlight. The same issue is very apparent in indoor lighting conditions, too. (theodop's original post is a good example of that) Both images below are taken with identical settings. (640x480 @ 30FPS, IR projector disabled)
Captured via USB2:
[image]

Captured via USB3:
[image]

Since the USB2 version is mostly washed out, most of the depth data is wrong or mostly missing. (a.k.a. not filled)

I did a bit more digging and found that a bug is also preventing you from being able to manually setting "Gain" or "Exposure" values - seems very similar to the old "autoexposure" bug except now the software-based autoexposure function I wrote to get around that bug doesn't work :-)

It should be noted that this issue has nothing to do with the laser projector. The D420 (OEM module without IR projector) exhibits identical behavior.

RealSenseCustomerSupport, 2019-11-16:
----------------------------------------------

Hi btravers1

Thanks a lot for reporting the issue and spending time on testing. Yes, the issue does impact both D430 and D420 (but not D415 or D410), and it will be fixed in FW5.12.1.0 as planned.
Please stay tuned....
(btw, currently there is no HDR feature in either the VPU chip, nor in the image sensor. This issue that only happens under USB2 is due to some registers were set incorrectly.)

Thanks

btravers1, 2019-11-18:
Thank you for the information!
That would make sense - I tested a D410 module and found it worked much better. (unfortunately, we can't use it in our device due to FoV/global shutter requirements) I'm excited for the update!

Are there any rough estimates for the release date? (helpful so I can develop a rough field test schedule)

RealSenseCustomerSupport, 2019-11-18:
----------------------------------------------

Hi btravers1,

Thanks for asking, I don't see when FW5.12.1.0 is going to release to public since currently the FW5.12.0.0 is about to release. I would expect it two weeks later after FW5.12.0.0 being released. Please check out here (https://dev.intelrealsense.com/docs/firmware-releases?_ga=2.7935409.2059913868.1574099817-666642886.1566408450)

Thanks

**Please be informed that we will close this ticket in seven (7) days if there is no activities on this ticket.

btravers1, 2019-11-18:
Great, thank you!

closed 2019-11-26 by RealSenseCustomerSupport

benhanginaround, 2020-02-12:
Hello,
This issue apparently didn't make it into FW5.12.1.0 (or 5.12.2.1). Do you have an update for when this bug should be resolved?
Thank you