realsenseai/librealsense #5158: T265 - Raspberry Pi Zero, Device Not Found?

tropicdragon, 2019-10-30:
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265| 
| Operating System & Version |  Raspbian ( Linux) | 
| Platform                                 | Raspberry Pi Zero W |

### Issue Description
I have a Raspberry Pi Zero currently, and every time I try to connect the T265 camera it seems to not be detected. When I run rs-enumerate-devices it prints the following to the terminal:

_No device detected. Is it plugged in?_

 I also have a D415 camera, and when I plug that it in, it seems to detect it with no issues. It seems to only have a problem with the T265. Additionally, I created a python script that does more or less the same thing as rs-enumerate-devices with the same exact results as stated above. 

I am using a USB 2.0 4 Port HUB to connect the T265 to the Raspberry Pi Zero. Does anyone know how to fix this issue of the camera not being detected? 

I have actually tried this same exact solution with Raspberry PI 4 with success, just wondering what specifically is going wrong with the Raspberry Pi Zero that is causing this.

(@snowymo)

dorodnic, 2019-11-02:
When building librealsense on the Pi, the build process requires internet connection (to download and embed T265 firmware). When connection cannot be established it is stated in the CMake log, and librealsense is built with T265 support disabled. 
If I had to guess, this is the simplest cause for "No device detected". Of course there can be some other problem at the USB level. Please also share `dmesg` log when connecting the device, as well as the output of `lsusb`

tropicdragon, 2019-11-04:
Sure, here is the dmesg log:
[  26.584912] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  26.584928] usb 1-1.1: Product: Movidius MA2X5X
[  26.584941] usb 1-1.1: Manufacturer: Movidius Ltd.
[  26.584956] usb 1-1.1: SerialNumber: 03e72150

And here is the lsusb log:

Bus 001 Device 006: ID 413c:2113 Dell Computer Corp.
Bus 001 Device 005: ID 1bcf:08a0 Sunplus Innovation Technology Inc.
Bus 001 Device 007: ID 03e7:2150 Intel Myriad VPU [Movidius Neural Compute Stick]
Bus 001 Device 003: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

And I made sure that there was internet connection during the build process, so I do not believe it could be that. I ran cmake, make, and make install all while having internet connection. Let me know if anything looks off.

dorodnic, 2019-11-11:
The device is detected at OS level. Please share CMake output

tropicdragon, 2019-11-12:
Yes I have attached it below:

cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release -DFORCE_LIBUVC=true
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.28.0
-- Setting Unix configurations
-- Using libuvc (by force)
-- using RS2_USE_LIBUVC_BACKEND
-- pybind11 v2.2.1
CMake Warning (dev) at /usr/share/cmake-3.13/Modules/FindOpenGL.cmake:270 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/arm-linux-gnueabihf/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/arm-linux-gnueabihf/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/arm-linux-gnueabihf/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  CMake/opengl_config.cmake:1 (find_package)
  src/gl/CMakeLists.txt:36 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/share/cmake-3.13/Modules/FindOpenGL.cmake:270 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/arm-linux-gnueabihf/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/arm-linux-gnueabihf/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/arm-linux-gnueabihf/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  CMake/opengl_config.cmake:1 (find_package)
  examples/CMakeLists.txt:14 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Could NOT find apriltag (missing: APRILTAG_INC APRILTAG_LIB) 
-- Unable to find apriltag library, skipping pose-apriltag example
CMake Warning (dev) at /usr/share/cmake-3.13/Modules/FindOpenGL.cmake:270 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/arm-linux-gnueabihf/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/arm-linux-gnueabihf/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/arm-linux-gnueabihf/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  CMake/opengl_config.cmake:1 (find_package)
  tools/CMakeLists.txt:22 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Building with TM2
-- Download TM2 firmware 0;"returning early; file already exists with expected SHA1 hash" for target-0.1.0.279.mvcmd
-- Download TM2 firmware 0;"returning early; file already exists with expected SHA1 hash" for central_app-2.0.19.271.bin
-- Download TM2 firmware 0;"returning early; file already exists with expected SHA1 hash" for central_bl-1.0.1.112.bin
-- Fetching recommended firmwares:
-- D4XX_RC_VERSION: 5.11.14.0
-- D4XX_FW_VERSION: 5.11.11.100
-- SR3XX_FW_VERSION: 3.26.1.0
-- Download firmware 0;"returning early; file already exists with expected SHA1 hash" for D4XX_FW_Image-5.11.11.100.bin
-- Download firmware 0;"returning early; file already exists with expected SHA1 hash" for D4XX_RC_Image-5.11.14.0.bin
-- Download firmware 0;"returning early; file already exists with expected SHA1 hash" for SR3XX_FW_Image-3.26.1.0.bin
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Desktop/librealsense/build

dorodnic, 2019-11-13:
Thank you @tropicdragon 
You are right, the build process went Ok as well. 
Did you try to run any of the examples under sudo (I'm trying to eliminate permissions issue) 
Also, could you check using `which` command that you are running the version you compiled and not some older version installed on the device? (not very likely, I know, but I'm trying to eliminate common reasons for this issue first)

tropicdragon, 2019-11-13:
Yes, so right now I upgraded to 2.30, originally I was at 2.28, I upgraded to see if it would fix my issues and the problem continues to persist. I uninstalled 2.28, and I checked in usr/local/lib and it is no longer there anymore. So right now I am using 2.30. 

I had a python file that worked on my Raspberry Pi 4, I would run the command
` python3 t265-pose.py `
I am not sure in this case where I would use the keyword which?
This originally worked on my Raspberry Pi 4, but now I tried doing this on my Pi 4 and it doesn't work anymore, in the same way that it doesn't work with the Zero. (It says the camera is not connected). I rebuilt the library recently, I am wondering if that is why?

As before, the D415 camera is detected, but not the T265.

Also, I noticed that when I try running realsense-viewer I get a segmentation fault. I don't know if that indicates anything.

I tried running some of the examples with sudo, and it still seems to not detect T265, but detected D415 as stated above.

tropicdragon, 2019-12-04:
So just an update, I found that specifically version 2.29 is the only one that seems to work.

RealSenseCustomerSupport, 2020-01-27:
----------------------------------------------

Hi @tropicdragon
Are you still seeing this issue with the latest 2.32.1 librealsense?  Do you still need assistance on this issue?

RealSenseCustomerSupport, 2020-02-07:
----------------------------------------------

Hi @tropicdragon
Are you still seeing this issue with the latest 2.32.1 librealsense?  Do you still need assistance on this issue?

***If we don’t hear from you in 7 days, this issue will be closed.**

closed 2020-02-24 by RealSenseCustomerSupport