realsenseai/librealsense #5159: GigE for Realsense

tispratik, 2019-10-30:
Is it possible to have a stable GigE connector for Realsense? A company out of Germany seems to have done it ( https://www.framos.com/en/framos-depth-camera-d435e-camera-only-22806 ). I wonder if there is some option available to do it locally in the US.

MartyG-RealSense, 2019-10-31:
As I am not an Intel RealSense employee, the statement below should be taken as my personal opinion rather than an official Intel position ...

RealSense cameras tend to aim for a certain maximum price range at the lower end (e.g 200 dollars instead of 400).  As there are costs associated with GigE - a licensing fee as well as the components - adding GigE may push the camera's price point towards the higher end.  This is just my speculation though.

It is already possible to connect the 400 Series cameras to ethernet, though in a less cable-tidy way than the D435e.  Example projects are EtherSense, and the multi-cam point cloud stitching system of the CONIX Research Center at Carnegie Mellon.

https://github.com/krejov100/EtherSense

https://github.com/conix-center/pointcloud_stitching

closed 2019-11-14 by RealSenseCustomerSupport

tispratik, 2019-11-14:
Thanks Marty. You specifically mentioned D435e... do you mean to say that D435e has some feature related to GigE that D435 does not? I presume D435e is more data intensive on the network as compared to D435.

MartyG-RealSense, 2019-11-14:
Extensive information about the feature-set of the FRAMOS D435e and the unique features it adds that the Intel D435 lacks can be found in the link below.

https://github.com/IntelRealSense/librealsense/issues/4769

The D435e's GigE support will enable the camera to be controlled with software that is compliant with the GigE standard, according to FRAMOS's press information.  that information also adds that FRAMOS increase transmission reliability by using packet resends.  

I assume that the process is managed in hardware with FRAMOS's custom GigE processing board inside the camera, and their information states that the camera is very reliable and works over long distance (a cable length of up to 100 meters).