realsenseai/librealsense #5161: python build visual studio 2015 error file

scizors, 2019-10-31:
I have used cmake and generated the for 64 bit and try to build python wrappers in vs 2015

the output which I got after builing in vs2015 
-realsense2d.dll
-pyrealsense2d.pyd

what is d means in this case can anyone explain

ev-mp, 2019-10-31:
The suffix 'd' applied to compiled libraries in a debug build. this is Windows convention.

scizors, 2019-10-31:
Hello,
 Thanks for the reply 
-It is mentioned in python wrappers that
Open the output folder of the project (e.g build\x64-Release\Release\) and copy pyrealsense2.pyd into your python's site-packages (e.g C:\Python27\Lib\site-packages)
-should I need to copy with 'd' or edit the file and copy

ev-mp, 2019-10-31:
The instructions refer to release build as indicated -
> (e.g build\x64-**Release\Release**)

Just drop the trailing `d` from pyrealsense2**~~d~~**.pyd and run as usual :
```python
import pyrealsense2 as rs
rs.     .....
```

Note - do not rename realsense2d.dll

scizors, 2019-10-31:
thanks for the reply