realsenseai/librealsense #5162: Way to open pointcloud AND align-advanced viewers?

aalarcon96, 2019-10-31:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Operating System & Version |  MacOS  | 
| Platform                                 | PC  |
| Language                            |  {C/C#/                          | 

### Issue Description
Hi, 

Is there a way to get two viewers while running one sample? I am doing some work with align-advanced and I've been trying to add a second viewer to see the pointcloud as well but I am just getting this error: 
RealSense error calling rs2_pipeline_start(pipe:0x7fd6a7c55ba0):
    Could not open device. Last Error: Operation timed out
I tried adding another window app and starting a new pipeline using different variable names so it wouldn't get confused but I have not had much luck. Any suggestions?

neilyoung, 2019-10-31:
The only way I found in order to achieve this is to make a recording and playback this in two instances of the viewer.

aalarcon96, 2019-10-31:
So, just to clarify, you have not found a way to open two instances of the viewer while live streaming?

neilyoung, 2019-10-31:
Exactly

dorodnic, 2019-11-02:
Most operating systems enforce strict ownership on video streaming. 
While this can be circumvented using a custom driver or system-wide service, we decided not to go against common practice and keep it simple. Hence only one process can stream from a specific RealSense device (to be precise, specific sensor object within device class)

closed 2019-11-11 by dorodnic