realsenseai/librealsense #5164: How to get a part of the 3D points? For example, the 3D points of an object in the view.

JimXu1989, 2019-11-01:
I have a  D435i sensor.
Use this camera, I can align color image with depth image, and the depth image can be converted to 3D points.
Now I want segment an object in the color image, and get the 3D points corrsponding to the pixels of the object.
The numpy formated image can be get in the following way:
frames = pipeline.wait_for_frames()
depth_frame = frames.get_depth_frame()
color_frame = frames.get_color_frame()
depth_image = np.asanyarray(depth_frame.get_data())
 color_image = np.asanyarray(color_frame.get_data())

While, when I segment the object with numpy formatted image,  I set all the pixels not belong to the object to 0, and I want reset it to the frame, but I falled.
The 3D points could only be get with frame:
points = pc.calculate(depth_frame)

Could you give me some advise?
Thanks a lot!!

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) }  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  python                    | 
 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2019-11-01:
I hope the discussion linked to below, about focusing on an area of coordinates within a 'bounding box', will be of help to you.

https://github.com/IntelRealSense/librealsense/issues/2769

JimXu1989, 2019-11-04:
@MartyG-RealSense  Thanks a lot for the advise!! while, could you tell me the difference between vertix and 3D points? cause in issue #2769 they use vertix (return by get_vertices()) instead of 3D points.

MartyG-RealSense, 2019-11-04:
Goshen has just replied to your question.

https://github.com/IntelRealSense/librealsense/issues/2769#issuecomment-549241643

JimXu1989, 2019-11-05:
@MartyG-RealSense  Thanks

RealSenseCustomerSupport, 2019-11-11:
----------------------------------------------

Hi @JimXu Do you need further help on this?

JimXu1989, 2019-11-13:
> Hi @jimxu Do you need further help on this?

This issue could be closed

closed 2019-11-13 by JimXu1989