realsenseai/librealsense #5165: add "software_sensor" to realsense device

Jiloc, 2019-11-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR300 / D400} | 
| Language                            |  {C++ }                          | 

I am using an external camera in conjunction with a SR300.
I already calibrated the system, so I have the instrinsics of the external camera and the extrinsics in relationship with the realsense.

I would like to create a `rs2::device` able to generate RGB frames from the external camera and Depth frames from the realsense.

I know that there is a `rs2::software_device` available, the the relative example. But in that case, from my understanding, I can only add "software_sensor" to the `rs2::software_device`. 

In my case i need to add "software_sensor" to the RealSense sensor which generates frames using the external camera. Or vice versa, I could create a "rs2::software_device" and add to it the depth sensor from the realsense and a "software_sensor"

MartyG-RealSense, 2019-11-01:
I recalled that there was a past case where somebody wanted to align the depth of a RealSense D410 camera with the RGB from an external non-RealSense MIPI camera.  It was recommended that software-device was used to generate software sensor, like you suggested.

I managed to find the link to the discussion, if you have not already seen it.

https://github.com/IntelRealSense/librealsense/issues/4202

RealSenseCustomerSupport, 2019-12-04:
----------------------------------------------

@Jiloc
Any update? If no other question, will close this.

closed 2019-12-26 by RealSenseCustomerSupport