realsenseai/librealsense #5166: after run realsense-viewer ,show this question

Grandzxw, 2019-11-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
/home/grandz/图片/D435i.png

MartyG-RealSense, 2019-11-01:
The link to your png picture did not work.  Could you try again please?

Grandzxw, 2019-11-01:
ok, I write it:
01/11 19:22:56,228 WARNING [140708642838272] (types.cpp:48) hwmon command 0x4f failed. Error type: No data to return (-21).
 01/11 19:22:56,494 WARNING [140708642838272] (backend-v4l2.cpp:1148) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 01/11 19:23:01,562 WARNING [140708642838272] (types.cpp:48) set_pu(id=3) failed! Last Error: Input/output error
 01/11 19:23:01,563 WARNING [140708642838272] (sensor.cpp:766) Exception was thrown when inspecting RGB Camera property Controls exposure time of color camera. Setting any value will disable auto exposure
 01/11 19:23:06,608 WARNING [140708642838272] (types.cpp:48) set_pu(id=9) failed! Last Error: Input/output error
 01/11 19:23:06,608 WARNING [140708642838272] (sensor.cpp:766) Exception was thrown when inspecting RGB Camera property Controls white balance of color image. Setting any value will disable auto white balance
 01/11 19:23:13,324 WARNING [140708507784960] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
 01/11 19:23:13,897 WARNING [140708351567616] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
 01/11 19:23:14,249 WARNING [140707817842432] (sensor.cpp:1152) HID timestamp not found, switching to Host timestamps.

Grandzxw, 2019-11-01:
and the RGB image and the stereo image show normally

Grandzxw, 2019-11-01:
**Environments**
Ubuntu kernel: 4.15.0-66-generic
Camera: D435i
Firmware: 05.11.15.00 (latest development release version)
librealsense version: SDK2.29.0

ROS realsense version: 
ROS version: kinetic

Grandzxw, 2019-11-01:
ROS realsense version:2.2.9

Grandzxw, 2019-11-01:
and when I run: roslaunch realsense2_camera rs_camera.launch
it shows: "Frame metadata isn't available
UVC metadata payloads not available.
what I should do?

MartyG-RealSense, 2019-11-01:
Did it work before you used the new 2.2.9 ROS wrapper, please?

Grandzxw, 2019-11-01:
no. before I used the the new 2.2.9 ROS wrapper,it show above after I run realsense-viewer.

MartyG-RealSense, 2019-11-01:
Are you using the motion tracking module in the RealSense Viewer?   If you are, do the errors still happen if you have accel and gyro turned off?

Grandzxw, 2019-11-01:
Yes,the error above is when I turn on the motion tracking module.
and when I turn off the motion tracking module ,it happens:
 01/11 19:56:12,567 WARNING [139852830111488] (types.cpp:48) hwmon command 0x4f failed. Error type: No data to return (-21).
 01/11 19:56:12,735 WARNING [139852830111488] (backend-v4l2.cpp:1148) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 01/11 19:56:17,799 WARNING [139852830111488] (types.cpp:48) set_pu(id=3) failed! Last Error: Input/output error
 01/11 19:56:17,799 WARNING [139852830111488] (sensor.cpp:766) Exception was thrown when inspecting RGB Camera property Controls exposure time of color camera. Setting any value will disable auto exposure
 01/11 19:56:22,838 WARNING [139852830111488] (types.cpp:48) set_pu(id=9) failed! Last Error: Input/output error
 01/11 19:56:22,839 WARNING [139852830111488] (sensor.cpp:766) Exception was thrown when inspecting RGB Camera property Controls white balance of color image. Setting any value will disable auto white balance
 01/11 19:56:52,860 WARNING [139852695717632] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
 01/11 19:57:07,098 WARNING [139852673361664] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.

MartyG-RealSense, 2019-11-01:
The 'No data to return (-21).' is a known problem with the D435i when the motion tracking module is activated, and other users have reported it.  They found that when the accel and gyro were turned off, the camera worked normally and this error did not occur.

What computer hardware are you using, please?   Is it a Raspberry Pi 4?

Grandzxw, 2019-11-01:
NO, I used  PC: DELL

MartyG-RealSense, 2019-11-01:
Have you calibrated the IMU component of your D435i please?   This component is not calibrated in the factory and has to be done by the user after they purchase the camera.

Grandzxw, 2019-11-01:
I don' t  calibrated the IMU component.so what do you mean is that it shows normal when I  calibrated it?
and how do I calibrate it? Thank you!

MartyG-RealSense, 2019-11-01:
The IMU of the D435i is calibrated using a different tool from the one that is used to calibrate the depth sensor.  Details of how to calibrate the D435i IMU are in this document:

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_Depth_D435i_IMU_Calib.pdf

Before you do the calibration though, it is worth reading the long discussion in the link below, which describes other users who have been having these errors and the advice Intel gave to them,

https://github.com/IntelRealSense/realsense-ros/issues/727

Grandzxw, 2019-11-01:
OK，Thank you!

Grandzxw, 2019-11-01:
I calibrated the IMU module,when I run realsense-viewer(I turn off the motion tracking module),it happens:
 01/11 21:34:38,591 WARNING [140011488782080] (backend-v4l2.cpp:1148) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 01/11 21:34:43,656 WARNING [140011488782080] (types.cpp:48) set_pu(id=3) failed! Last Error: Input/output error
 01/11 21:34:43,657 WARNING [140011488782080] (sensor.cpp:766) Exception was thrown when inspecting RGB Camera property Controls exposure time of color camera. Setting any value will disable auto exposure
 01/11 21:34:48,693 WARNING [140011488782080] (types.cpp:48) set_pu(id=9) failed! Last Error: Input/output error
 01/11 21:34:48,693 WARNING [140011488782080] (sensor.cpp:766) Exception was thrown when inspecting RGB Camera property Controls white balance of color image. Setting any value will disable auto white balance
 01/11 21:34:50,570 WARNING [140011353663232] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
 01/11 21:34:51,482 WARNING [140011331897088] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.

Grandzxw, 2019-11-01:
and ,when I run realsense-viewer(I turn on the motion tracking module),it happens:
 01/11 21:34:38,591 WARNING [140011488782080] (backend-v4l2.cpp:1148) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 01/11 21:34:43,656 WARNING [140011488782080] (types.cpp:48) set_pu(id=3) failed! Last Error: Input/output error
 01/11 21:34:43,657 WARNING [140011488782080] (sensor.cpp:766) Exception was thrown when inspecting RGB Camera property Controls exposure time of color camera. Setting any value will disable auto exposure
 01/11 21:34:48,693 WARNING [140011488782080] (types.cpp:48) set_pu(id=9) failed! Last Error: Input/output error
 01/11 21:34:48,693 WARNING [140011488782080] (sensor.cpp:766) Exception was thrown when inspecting RGB Camera property Controls white balance of color image. Setting any value will disable auto white balance
 01/11 21:34:50,570 WARNING [140011353663232] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
 01/11 21:34:51,482 WARNING [140011331897088] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
 01/11 21:37:54,240 WARNING [140010867111680] (sensor.cpp:1152) HID timestamp not found, switching to Host timestamps.

Grandzxw, 2019-11-01:
I also saw another article,but I still don't know how to solve it .

MartyG-RealSense, 2019-11-01:
At least the -21 error seems to have gone after the calibration has been done.

We will try to deal with the metadata errors next.  Windows 10 supports metadata automatically but Linux has to have support for metadata patched in to the kernel.

To add metadata support in Linux, can you try rebuilding Librealsense using the **uvc-backend** installation method please?  

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

Grandzxw, 2019-11-01:
So I should uninstall the librealsense that  I have installed .and if necessrary ,how I do it?

MartyG-RealSense, 2019-11-01:
It is not something I have done myself (I am a Windows user).  It may be a good idea to delete the contents of your Librealsense 'build' folder before doing it, as that is what is done when doing a new build with the conventional CMake installation process.

Grandzxw, 2019-11-02:
when I run:wget https://github.com/IntelRealSense/librealsense/raw/master/scripts/libuvc_installation.sh
it happens:--2019-11-02 10:05:32--  (尝试次数： 2)  https://raw.githubusercontent.com/IntelRealSense/librealsense/master/scripts/libuvc_installation.sh
正在连接 raw.githubusercontent.com (raw.githubusercontent.com)|31.13.85.16|:443... 失败：连接超时。
重试中。
I try the solution on the internet,but it don't work.

Grandzxw, 2019-11-02:
or if you have this file,can you send me:https://raw.githubusercontent.com/IntelRealSense/librealsense/master/scripts/libuvc_installation.sh

Grandzxw, 2019-11-02:
my email:  g20188592@xs.ustb.edu.cn

Grandzxw, 2019-11-02:
after I try rebuilding Librealsense using the uvc-backend installation method. it still happens :
 02/11 14:43:37,990 WARNING [140477764278016] (backend-v4l2.cpp:1148) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 02/11 14:43:43,052 WARNING [140477764278016] (types.cpp:48) set_pu(id=3) failed! Last Error: Input/output error
 02/11 14:43:43,052 WARNING [140477764278016] (sensor.cpp:766) Exception was thrown when inspecting RGB Camera property Controls exposure time of color camera. Setting any value will disable auto exposure
 02/11 14:43:48,089 WARNING [140477764278016] (types.cpp:48) set_pu(id=9) failed! Last Error: Input/output error
 02/11 14:43:48,090 WARNING [140477764278016] (sensor.cpp:766) Exception was thrown when inspecting RGB Camera property Controls white balance of color image. Setting any value will disable auto white balance
 02/11 14:43:49,242 WARNING [140477628147456] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
 02/11 14:43:49,880 WARNING [140477607167744] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.

MartyG-RealSense, 2019-11-02:
Do the errors in the Real sense Viewer still continue if you go to the More' icon at the top of the options side panel (the icon that looks like a pile of plates) and select the Reset Hardware menu option?

Grandzxw, 2019-11-02:
YES，the error still continue.
: 02/11 15:21:45,006 WARNING [140018306037504] (backend-v4l2.cpp:1148) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 02/11 15:21:50,068 WARNING [140018306037504] (types.cpp:48) set_pu(id=3) failed! Last Error: Input/output error
 02/11 15:21:50,069 WARNING [140018306037504] (sensor.cpp:766) Exception was thrown when inspecting RGB Camera property Controls exposure time of color camera. Setting any value will disable auto exposure
 02/11 15:21:55,105 WARNING [140018306037504] (types.cpp:48) set_pu(id=9) failed! Last Error: Input/output error
 02/11 15:21:55,105 WARNING [140018306037504] (sensor.cpp:766) Exception was thrown when inspecting RGB Camera property Controls white balance of color image. Setting any value will disable auto white balance
 02/11 15:21:57,975 WARNING [140018171488000] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
 02/11 15:21:58,553 WARNING [140017880004352] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
 02/11 15:22:01,499 ERROR [140018207155968] (types.h:304) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 02/11 15:22:01,499 ERROR [140018207155968] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 02/11 15:22:01,553 ERROR [140018297644800] (types.h:304) Cannot open '/dev/video1 Last Error: No such file or directory
 02/11 15:22:01,554 ERROR [140018297644800] (global_timestamp_reader.cpp:165) Error during time_diff_keeper polling: Cannot open '/dev/video1 Last Error: No such file or directory
 02/11 15:22:01,563 ERROR [140018171488000] (types.h:304) xioctl(VIDIOC_DQBUF) failed for fd: 20 Last Error: No such device
 02/11 15:22:01,563 ERROR [140018171488000] (backend-v4l2.cpp:1235) xioctl(VIDIOC_DQBUF) failed for fd: 20 Last Error: No such device
 02/11 15:22:01,564 ERROR [140017880004352] (types.h:304) xioctl(VIDIOC_DQBUF) failed for fd: 28 Last Error: No such device
 02/11 15:22:01,564 ERROR [140017880004352] (backend-v4l2.cpp:1235) xioctl(VIDIOC_DQBUF) failed for fd: 28 Last Error: No such device
 02/11 15:22:02,500 ERROR [140018207155968] (types.h:304) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 02/11 15:22:02,500 ERROR [140018207155968] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 02/11 15:22:02,554 ERROR [140018297644800] (types.h:304) Cannot open '/dev/video1 Last Error: No such file or directory
 02/11 15:22:02,555 ERROR [140018297644800] (global_timestamp_reader.cpp:165) Error during time_diff_keeper polling: Cannot open '/dev/video1 Last Error: No such file or directory
 02/11 15:22:03,500 ERROR [140018207155968] (types.h:304) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 02/11 15:22:03,500 ERROR [140018207155968] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 02/11 15:22:03,555 ERROR [140018297644800] (types.h:304) Cannot open '/dev/video1 Last Error: No such file or directory
 02/11 15:22:03,555 ERROR [140018297644800] (global_timestamp_reader.cpp:165) Error during time_diff_keeper polling: Cannot open '/dev/video1 Last Error: No such file or directory
 02/11 15:22:04,501 ERROR [140018207155968] (types.h:304) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 02/11 15:22:04,501 ERROR [140018207155968] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 02/11 15:22:04,556 ERROR [140018297644800] (types.h:304) Cannot open '/dev/video1 Last Error: No such file or directory
 02/11 15:22:04,556 ERROR [140018297644800] (global_timestamp_reader.cpp:165) Error during time_diff_keeper polling: Cannot open '/dev/video1 Last Error: No such file or directory
 02/11 15:22:05,283 WARNING [140018485397312] (backend-v4l2.cpp:1148) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 02/11 15:22:05,325 ERROR [140018485397312] (types.h:304) xioctl(VIDIOC_STREAMOFF) failed for buf_type=1 Last Error: No such device
 02/11 15:22:13,368 WARNING [140018190370560] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
 02/11 15:22:15,016 WARNING [140017880004352] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.

MartyG-RealSense, 2019-11-02:
I found some further Intel advice about the errors you are encountering.

https://github.com/IntelRealSense/librealsense/issues/4881#issuecomment-532530091

Grandzxw, 2019-11-02:
I try it,but it also doesn't work .and I felt so hard to solve it .

MartyG-RealSense, 2019-11-02:
I understand your frustration very much.   Thank you for your patience.

These problems, when reported, seem to often involve the 4.15 kernel branch.

https://github.com/IntelRealSense/librealsense/issues/4794#issuecomment-532558814

Grandzxw, 2019-11-02:
when I  Build and apply patched kernel modules，./scripts/patch-realsense-ubuntu-lts.sh it happens:
:Replacing videodev  -
	Applying the patched module ... modprobe: ERROR: could not insert 'videodev': Required key not available
Failed to insert the patched module. Operation is aborted, the original module is restored
Verify that the current kernel version is aligned to the patched module version
The original  videodev  module was reloaded

MartyG-RealSense, 2019-11-02:
I do not know what else to suggest unfortunately after having gone through all the other possibilities above.  One of the Intel staff on this forum may be able to suggest something else to try (I do not work for Intel).  I apologize and wish you good luck.

Grandzxw, 2019-11-02:
Thank you for your patience! and how I contact with him?

MartyG-RealSense, 2019-11-02:
You will need to wait for one of the Intel staff (there are several regular ones) to post a reply here.  I will link in some of them to help draw this message to their attention.

@dorodnic @ev-mp @RealSense-Customer-Engineering @RealSenseCustomerSupport

ev-mp, 2019-11-03:
@Grandzxw , you haven't stated the platform/OS used, but based on the follow errors
>02/11 15:21:55,105 WARNING [140018306037504] (types.cpp:48) set_pu(id=9) failed! Last Error: Input/output error
02/11 15:21:55,105 WARNING [140018306037504] (sensor.cpp:766) Exception was thrown when inspecting RGB Camera property Controls white balance of color image. Setting any value will disable auto white balance
...
02/11 15:22:01,499 ERROR [140018207155968] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
02/11 15:22:01,553 ERROR [140018297644800] (types.h:304) Cannot open '/dev/video1 Last Error: No such file or directory
02/11 15:22:01,563 ERROR [140018171488000] (types.h:304) xioctl(VIDIOC_DQBUF) failed for fd: 20 Last

the host has difficulties to query the device and also the camera is being forcibly disconnected during streaming. Those symptoms can indicate underpower and other host-related issues. I would recommend to check the cabling, switch USB port and connect the device via powered USB3 hub.

The metadata warnings are raised as the kernel patches not being applied
>02/11 15:21:57,975 WARNING [140018171488000] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.

Follow the [installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) to patch kernel using one of the methods provided:
- [DKMS path](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages): 
```sudo apt install librealsense2-dkms
```
- [Running script](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites)
```./scripts/patch-realsense-ubuntu-lts.sh
```

MartyG-RealSense, 2019-11-03:
@ev-mp  Thanks so much for helping with this case!  @Grandzxw added his OS details a few comments in, and added further on that he is using a Dell PC.

https://github.com/IntelRealSense/librealsense/issues/5166#issuecomment-548756618

ev-mp, 2019-11-04:
@MartyG-RealSense thanks for the correction.
@Grandzxw , to patch kernel version 4.15.0-66-generic you have to remove DKMS package (if present) and run the manual script from the previous post. Note that it is important to follow the installation guide and perform all the steps required before running the script ,including
>sudo apt update && sudo apt dist-upgrade

Grandzxw, 2019-11-04:
>  to remove DKMS p

I don't know how to remove DKMS package，and  when I run   ./scripts/patch-realsense-ubuntu-lts.sh  ,it happens this error:

Replacing videodev  -
	Applying the patched module ... modprobe: ERROR: could not insert 'videodev': Required key not available
Failed to insert the patched module. Operation is aborted, the original module is restored
Verify that the current kernel version is aligned to the patched module version
The original  videodev  module was reloaded

ev-mp, 2019-11-04:
To remove DKMS package run ```sudo apt-get remove librealsense2-dkms```

The error
> modprobe: ERROR: could not insert 'videodev

indicates that there is inconsistency between the kernel version you use and the kernel sources used to generate the patched modules.
You need to follow the referred [installation guide/prerequisites section](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites) to resolve it

Grandzxw, 2019-11-06:
> To remove DKMS package run `sudo apt-get remove librealsense2-dkms`
>
> The error
>
>
> indicates that there is inconsistency between the kernel version you use and the kernel sources used to generate the [...]

I remove the  DKMS package,and I follow the installation guide but when I run  ./scripts/patch-realsense-ubuntu-lts.sh ,it still happens:
Replacing videodev  -
	Applying the patched module ... modprobe: ERROR: could not insert 'videodev': Required key not available
Failed to insert the patched module. Operation is aborted, the original module is restored
Verify that the current kernel version is aligned to the patched module version
The original  videodev  module was reloaded.
**and my kernel version is : 4.15.0-66-generic 16.04.1-Ubuntu.** ,I don't kown why?

Grandzxw, 2019-11-07:
> To remove DKMS package run `sudo apt-get remove librealsense2-dkms`
>
> The error
>
>
> indicates that there is inconsistency between the kernel version you use and the kernel sources used to generate the [...]

can you help me?

MartyG-RealSense, 2019-11-07:
@ev-mp can help with this question better than I can.  I do apologize.

ev-mp, 2019-11-07:
@Grandzxw, we'll try to troubleshoot the situation.
I presume that the work-space is already in some unstable state as it seems that you've tried but couldn't install according to the guide above , thus we'll make less assumptions and perform more "assertive" actions.
1. Uninstall DKMS 
```sh
sudo apt-get purge librealsense2-dkms
```
2. Update the system to avoid/resolve broken packages:
```sh
sudo apt-get update --fix-missing && sudo apt-get install -f && sudo apt-get dist-upgrade -y 
```
3.  In case you recognize that a newer kernel was installed in step 2 - reboot and select that kernel (should be from the 4.15... branch)
4. Reinstall the original kernel modules that match your current (uname -r) kernel:
```sh
sudo apt-get install linux-modules-extra-$(uname -r)
```
5 Rename/move the folders `ubuntu-xenial-hwe` and `ubuntu-xenial-hwe-edge` from librealsense root directory if present (the script should be executed from librealsense's root dir as well):
```sh
sudo mv ubuntu-xenial-hwe/ ubuntu-xenial-hwe-prev/
sudo mv ubuntu-xenial-hwe-edge ubuntu-xenial-hwe-edge-prev/
```
6. Rerun the patches script:
```sh ./scripts/patch-realsense-ubuntu-lts.sh```

In case the flow is interrupted with error- please provide the terminal/kernel logs collected to that point.

@MartyG-RealSense  - many thanks for the assistance

Grandzxw, 2019-11-08:
> @Grandzxw, we'll try to troubleshoot the situation.
> I presume that the work-space is already in some unstable state as it seems that you've tried but couldn't install according to the guide above , thus we'll make less assumptions and perform more "assertive" actions.
> 
>     1. Uninstall DKMS
> 
> 
> ```shell
> sudo apt-get purge librealsense2-dkms
> ```
> 
>     1. Update the system to avoid/resolve broken packages:
> 
> 
> ```shell
> sudo apt-get update --fix-missing && sudo apt-get install -f && sudo apt-get dist-upgrade -y 
> ```
> 
>     1. In case you recognize that a newer kernel was installed in step 2 - reboot and select that kernel (should be from the 4.15... branch)
> 
>     2. Reinstall the original kernel modules that match your current (uname -r) kernel:
> 
> 
> ```shell
> sudo apt-get install --reinstall linux-modules-extra-$(uname -r)
> ```
> 
> 5 Rename/move the folders `ubuntu-xenial-hwe` and `ubuntu-xenial-hwe-edge` from librealsense root directory if present (the script should be executed from librealsense's root dir as well):
> 
> ```shell
> sudo mv ubuntu-xenial-hwe/ ubuntu-xenial-hwe-prev/
> sudo mv ubuntu-xenial-hwe-edge ubuntu-xenial-hwe-edge-prev/
> ```
> 
>     1. Rerun the patches script:
>        `sh ./scripts/patch-realsense-ubuntu-lts.sh`
> 
> 
> In case the flow is interrupted with error- please provide the terminal/kernel logs collected to that point.
> 
> @MartyG-RealSense - many thanks for the assistance
when I follow your instructions , there happens two questions .the first is:
when I runs  sudo mv ubuntu-xenial-hwe-edge ubuntu-xenial-hwe-edge-prev/，it happens  MV: unable to get file state (STAT) for 'Ubuntu xenial HWE edge': no file or directory.
.then when I run ./scripts/patch-realsense-ubuntu-lts.sh  next. It still happens:
Replacing videodev  -
Module is resident, unloading ...  succeeded. 
	Applying the patched module ... modprobe: ERROR: could not insert 'videodev': Required key not available
Failed to insert the patched module. Operation is aborted, the original module is restored
Verify that the current kernel version is aligned to the patched module version
The original  videodev  module was reloaded

ev-mp, 2019-11-08:
@Grandzxw , the warning `required key not available` often indicates that the Secure Boot is turned on. Please provide the kernel log for confirmation, but at the same time you can follow [this comment](https://github.com/IntelRealSense/librealsense/issues/3354#issuecomment-469006598) to verify that it is turned off and then rerun the script. #2815, #3351

I've tried to follow this thread but still not sure - besides the missing metadata do you observe any issues in streaming the camera with Viewer ?

Grandzxw, 2019-11-08:
> @Grandzxw , the warning `required key not available` often indicates that the Secure Boot is turned on. Please provide the kernel log for confirmation, but at the same time you can follow [this [...]

First thank you so much! Follow you instruction,I solve that question:required key not available.but after  I compile successfully,when I run realsense-viewer,it still happens:
 08/11 17:43:27,011 WARNING [140119205996288] (backend-v4l2.cpp:1148) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 08/11 17:43:32,062 WARNING [140119205996288] (types.cpp:48) set_pu(id=3) failed! Last Error: Input/output error
 08/11 17:43:32,063 WARNING [140119205996288] (sensor.cpp:766) Exception was thrown when inspecting RGB Camera property Controls exposure time of color camera. Setting any value will disable auto exposure
 08/11 17:43:37,100 WARNING [140119205996288] (types.cpp:48) set_pu(id=9) failed! Last Error: Input/output error
 08/11 17:43:37,100 WARNING [140119205996288] (sensor.cpp:766) Exception was thrown when inspecting RGB Camera property Controls white balance of color image. Setting any value will disable auto white balance
 08/11 17:43:38,848 WARNING [140119071778560] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
 08/11 17:43:39,532 WARNING [140119050798848] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
 08/11 17:43:39,899 WARNING [140118660732672] (sensor.cpp:1152) HID timestamp not found, switching to Host timestamps.
and when I run **roslaunch realsense2_camera rs_camera.launch**, it happens:
 08/11 17:48:54,948 ERROR [140161502897920] (types.h:304) iio hid device is busy or not found! Last Error: Device or resource busy
 08/11 17:48:54,948 WARNING [140161077589760] (backend-hid.cpp:681) HID set_power 1 failed for /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.5/0003:8086:0B3A.0009/HID-SENSOR-200073.3.auto/iio:device0/buffer/enable
[ERROR] [1573206534.948377423]: An exception has been thrown: iio hid device is busy or not found! Last Error: Device or resource busy
terminate called after throwing an instance of 'rs2::backend_error'
  what():  iio hid device is busy or not found! Last Error: Device or resource busy
[camera/realsense2_camera_manager-2] process has died [pid 13762, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/grandz/.ros/log/f96946e8-020c-11ea-8c20-7c2a3140a853/camera-realsense2_camera_manager-2.log].
log file: /home/grandz/.ros/log/f96946e8-020c-11ea-8c20-7c2a3140a853/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has finished cleanly
log file: /home/grandz/.ros/log/f96946e8-020c-11ea-8c20-7c2a3140a853/camera-realsense2_camera-3*.log

Grandzxw, 2019-11-08:
> @Grandzxw , the warning `required key not available` often indicates that the Secure Boot is turned on. Please provide the kernel log for confirmation, but at the same time you can follow [this [...]

**and when I run: sudo dmesg | tail -n 50 ，it happens:**

[   24.264400] Bluetooth: BNEP filters: protocol multicast
[   24.264402] Bluetooth: BNEP socket layer initialized
[   25.857530] PKCS#7 signature not signed with a trusted key
[   25.858001] bbswitch: version 0.8
[   25.858005] bbswitch: Found integrated VGA device 0000:00:02.0: \_SB_.PCI0.GFX0
[   25.858011] bbswitch: Found discrete VGA device 0000:01:00.0: \_SB_.PCI0.PEG0.PEGP
[   25.858139] bbswitch: detected an Optimus _DSM function
[   25.858232] bbswitch: Succesfully loaded. Discrete card 0000:01:00.0 is on
[   26.602575] IPv6: ADDRCONF(NETDEV_UP): wlo1: link is not ready
[   26.776321] iwlwifi 0000:00:14.3: Conflict between TLV & NVM regarding enabling LAR (TLV = enabled NVM =disabled)
[   26.778223] iwlwifi 0000:00:14.3: BIOS contains WGDS but no WRDS
[   26.778611] IPv6: ADDRCONF(NETDEV_UP): wlo1: link is not ready
[   26.781197] IPv6: ADDRCONF(NETDEV_UP): enp2s0: link is not ready
[   26.918854] r8169 0000:02:00.0 enp2s0: link down
[   26.918943] IPv6: ADDRCONF(NETDEV_UP): enp2s0: link is not ready
[   27.130972] IPv6: ADDRCONF(NETDEV_UP): wlo1: link is not ready
[   28.443268] nvidia-modeset: Allocated GPU:0 (GPU-dde9f962-00cd-09dd-1cda-6ee58cc05aa5) @ PCI:0000:01:00.0
[   36.855134] Bluetooth: RFCOMM TTY layer initialized
[   36.855139] Bluetooth: RFCOMM socket layer initialized
[   36.855142] Bluetooth: RFCOMM ver 1.11
[   47.821895] wlo1: authenticate with 88:25:93:55:9a:56
[   47.824986] wlo1: send auth to 88:25:93:55:9a:56 (try 1/3)
[   47.848548] wlo1: authenticated
[   47.852067] wlo1: associate with 88:25:93:55:9a:56 (try 1/3)
[   47.860683] wlo1: RX AssocResp from 88:25:93:55:9a:56 (capab=0x431 status=0 aid=2)
[   47.868319] wlo1: associated
[   47.934287] iwlwifi 0000:00:14.3: Unhandled alg: 0x707
[   47.988117] IPv6: ADDRCONF(NETDEV_CHANGE): wlo1: link becomes ready
[  802.978966] wlo1: AP 88:25:93:55:9a:56 changed bandwidth, new config is 2437 MHz, width 1 (2437/0 MHz)
[  804.000819] wlo1: AP 88:25:93:55:9a:56 changed bandwidth, new config is 2437 MHz, width 2 (2447/0 MHz)
[ 1173.665467] usb 2-3: new SuperSpeed USB device number 2 using xhci_hcd
[ 1173.686330] usb 2-3: New USB device found, idVendor=8086, idProduct=0b3a
[ 1173.686335] usb 2-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1173.686339] usb 2-3: Product: Intel(R) RealSense(TM) Depth Camera 435i
[ 1173.686341] usb 2-3: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435i
[ 1173.686344] usb 2-3: SerialNumber: 915323051918
[ 1173.687989] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 1173.687997] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[ 1173.688085] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[ 1173.690287] uvcvideo: Unable to create debugfs 2-2 directory.
[ 1173.690461] uvcvideo 2-3:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[ 1173.690465] uvcvideo 2-3:1.0: Entity type for entity Processing 2 was not initialized!
[ 1173.690468] uvcvideo 2-3:1.0: Entity type for entity Camera 1 was not initialized!
[ 1173.690710] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/input/input28
[ 1173.691531] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[ 1173.691538] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[ 1173.692544] uvcvideo: Unable to create debugfs 2-2 directory.
[ 1173.692694] uvcvideo 2-3:1.3: Entity type for entity Processing 7 was not initialized!
[ 1173.692697] uvcvideo 2-3:1.3: Entity type for entity Extension 8 was not initialized!
[ 1173.692700] uvcvideo 2-3:1.3: Entity type for entity Camera 6 was not initialized!
so can you give me some suggestion?

Grandzxw, 2019-11-09:
> @Grandzxw , the warning `required key not available` often indicates that the Secure Boot is turned on. Please provide the kernel log for confirmation, but at the same time you can follow [this [...]

**I can't find the same question with me.  can you give me some suggestion?**

ev-mp, 2019-11-10:
@Grandzxw hello, what did you mean by
>I can't find the same question with me. can you give me some suggestion?

Did you manage to follow the reference link to disable secure boot in BIOS settings?
This is a key step to allow for kernel patching on the PC you're running.

I reviewed the attached kernel log and it reprots that the patches are is still not patches as needed:
>[ 1173.687989] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 1173.687997] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71

You need provide the **full log** of `./scripts/patch-realsense-ubuntu-lts.sh` and the corresponding `dmesg -wT` to identify the root cause. Additionally please attach the results of:
- `uname -r`
- `lsb_release -a`

The rest of the warnings, such as `[ 1173.692694] uvcvideo 2-3:1.3: Entity type for entity Processing 7 was not initialized!` can be safely disregarded.

Grandzxw, 2019-11-11:
Thank you so much! I have solve it .

closed 2019-11-17 by RealSenseCustomerSupport