realsenseai/librealsense #5167: T265: A question to the experts

neilyoung, 2019-11-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.1.0.279
| Operating System & Version |  Ubuntu 16.04
| Kernel Version (Linux Only)    |      | 
| Platform                                 |   |
| SDK Version                            |   2.2.9.0                           | 
| Language                            |                     | 
| Segment			|  others                           | 

As subject says: Not an issue, but a question. Given I would have a Linux box with a T265 plugged and a little script which maps and detects AprilTags. Given too, that I would - where ever I detect an AprilTag - store a `set_static_node` with some information in the map. Would I be able on a later run using the same map but maybe on another device to `get_static_node` in order to retrieve position and information while mapping/walking?

dorodnic, 2019-11-02:
Hi @neilyoung 
In theory - yes. In practice - the new firmware in #5169 is promising to improve that - making maps more device agnostic. I have personally not tested it yet, but @bfulkers-i and @radfordi may comment on this features maturity.

neilyoung, 2019-11-02:
Hi Sergey, thanks. I will give it a try.

neilyoung, 2019-11-03:
@dorodnic Trying to figure out, how to update the firmware. I suppose I have to use rs-fw-update for it, but I can't find the firmware image. Could you please give me a pointer? TIA

neilyoung, 2019-11-04:
Oh, I see, I have to clone the 'development` branch

neilyoung, 2019-11-04:
Hmm. Having now latest development branch on my laptop, but still no trace of the new firmware. Could somebody please explain the process of firmware update on T265?

bfulkers-i, 2019-11-04:
@neilyoung if you build the development branch with an internet connection, it will download the latest firmware (`0.2.0.857` on #5169 ). You may have to unplug and replug your device once to make sure it gets booted into the new firmware, but on a fresh startup librealsense will always load the latest T265 firmware onto the device automatically.

neilyoung, 2019-11-04:
@bfulkers-i Well, I did that. On my Linux box. Internet on (of course). Build and setup. No new firmware on the device.

neilyoung, 2019-11-04:
Repeating it right now. Could be that I have forgotten to uninstall the previous build before. Will post

neilyoung, 2019-11-04:
Confirmed. After doing it right, it works :) Firmware updated, SDK 2.30.0 running

neilyoung, 2019-11-04:
BTW: Is this issue tracker also supposed to discuss/open issues with the development branch?

Let me know please, I have one :)

radfordi, 2019-11-05:
> Would I be able on a later run using the same map but maybe on another device

Yes, this is a feature of the `2.30.0` release.  What you describe should work.

Also, this is the correct place for issues with the development branch.

neilyoung, 2019-11-05:
@radfordi Thanks. I had some minor problem with python and the dev branch (just some "invalid stream 0xc" traces), but I cannot reproduce now.

closed 2019-12-30 by neilyoung