realsenseai/librealsense #5168: Sync external Device with D435

ArniBarni, 2019-11-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   | 05.10.13  | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.25.0                          | 
| Language                            |  python                         | 
| Segment			|  others                         | 

### Issue Description

Hello,

i have a passive hw sync between two D435 cameras. Now i need to sync a external device to the D435 cameras, but the external device require a 5v pulse. So my question is, how can i shift the 1.8v  level pulse from the hw sync between the two D435 to the required 5V TTL pulse?

MartyG-RealSense, 2019-11-01:
Another user did this 1.8 to 5v conversion with a 'voltage shifter'.

https://github.com/IntelRealSense/librealsense/issues/4574

RealSenseCustomerSupport, 2019-11-06:
----------------------------------------------

Hi ArniBarni

There is a lot of breakout boards in the market, i.e. https://www.digikey.com/product-detail/en/sparkfun-electronics/BOB-11771/1568-1208-ND/5673794?WT.srch=1&gclid=EAIaIQobChMIyZbYi8TW5QIVleNkCh3eOArMEAkYBCABEgLcAvD_BwE

If you can do some simple HW job, you should be able to DIY.

Thanks

**Please be informed that we will close this ticket in seven (7) days if we have not heard back from you.

closed 2019-11-14 by RealSenseCustomerSupport