realsenseai/librealsense #5172: Cannot load presets with D415 using Pyrealsense2 Error: HResult 0x8007001f

hobbitsyfeet, 2019-11-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   |5.11.15.00| 
| Operating System & Version |   Win 10 (10.0.17763 Build 17763) | 
| Platform                                 | PC  |
| SDK Version                            |  2.29.0.1124                         | 
| Language                            |  Python  3.68                      | 
|USB version| 3.2 |

### Issue Description
I've been trying to load the preset options both by using:
    1. `depth_sensor.set_option(rs.option.visual_preset , 3)`
    2. `advanced_mode.load_json("./HighResHighAccuracyPreset.json")`

and both result in this error:
     `hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."`

I've encountered this problem with my code resembling the solution in https://github.com/IntelRealSense/librealsense/issues/2577 and the example code from https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-rs400-advanced-mode-example.py .
When I use the Realsense Viewer V2.29.0, the High Accuracy preset works fine.

The only success I've been able to find is using `depth_sensor.set_option(rs.option.visual_preset , 0)`

I don't know if this has anything to do with it, but if I try setting depth_sensor options that are not the presets, such as `depth_sensor.set_option(rs.option.max_distance,10)` I get object doesn't support option # 34. But rs.option.exposure works?

Is it possible to manually set the options I see in the JSON? 

I'll continue to try with a different computer and perhaps using Ubuntu instead of Windows, but ideally I would like to use Windows for this.

MartyG-RealSense, 2019-11-04:
Is the Python script in the link below for applying the High Accuracy preset any help to you please?

https://github.com/IntelRealSense/librealsense/issues/2577#issuecomment-432137634

ev-mp, 2019-11-04:
@hobbitsyfeet hello,

1.  `depth_sensor.set_option(rs.option.visual_preset , 3)` should work with D415. you can post the code fore review here. Also - please share the librealsense logs when this error is fired.

2.  >`advanced_mode.load_json("./HighResHighAccuracyPreset.json")` 

The syntax is not correct - the string object that you pass along should include the **content** of the json file, not its name. #3696, #4046.

3.  `depth_sensor.set_option(rs.option.max_distance,10)` -  is used in processing filters. It is not applicable to `depth_sensor`. 
It is recommended to use `sensor.supports(rs.option.<tbd>)` query before setting/getting option to improve code robustness.

4. >Is it possible to manually set the options I see in the JSON?

Correct, follow the python link you've posted and expand the demo by adding `advanced_mode.set_<depth_control/get_rau_thresholds_control/slo_penalty_control/tbd_control>(...)` calls
See [documentation](https://intelrealsense.github.io/librealsense/python_docs/search.html?q=pyrealsense2.rs400_advanced_mode&check_keywords=yes&area=default) for the full list of supported APIs

hobbitsyfeet, 2019-11-04:
Hi, and sorry, that syntax mistake is from trying to give context to to the input rather than a variable from the JSON content. Can sure tell this is my first issue, eh? I'm trying my best, just try to keep with me...

Here is what is going on when I execute the rs400-advanced-mode.

```
Found device that supports advanced mode: Intel RealSense D415
Advanced mode is enabled
Depth Control:
 minusDecrement: 10, deepSeaMedianThreshold: 500, scoreThreshA: 1, scoreThreshB: 2047, textureDifferenceThreshold: 0, textureCountThreshold: 0, deepSeaSecondPeakThreshold: 325, deepSeaNeighborThreshold: 7, lrAgreeThreshold: 24
RSM:
 rsmBypass: 0, diffThresh: 4, sloRauDiffThresh: 1, removeThresh: 63
RAU Support Vector Control:
 minWest: 1, minEast: 1, minWEsum: 3, minNorth: 1, minSouth: 1, minNSsum: 3, uShrink: 3, vShrink: 1
Color Control:
 disableSADColor: 0, disableRAUColor: 0, disableSLORightColor: 0, disableSLOLeftColor: 0, disableSADNormalize: 0
RAU Thresholds Control:
 rauDiffThresholdRed: 51, rauDiffThresholdGreen: 51, rauDiffThresholdBlue: 51
SLO Color Thresholds Control:
 diffThresholdRed: 72, diffThresholdGreen: 72, diffThresholdBlue: 72
SLO Penalty Control:
 sloK1Penalty: 60, sloK2Penalty: 342, sloK1PenaltyMod1: 105, sloK2PenaltyMod1: 190, sloK1PenaltyMod2: 70, sloK2PenaltyMod2: 130
HDAD:
 lambdaCensus: 26, lambdaAD: 800, ignoreSAD: 0
Color Correction:
 colorCorrection1: 0.461914, colorCorrection2: 0.540039, colorCorrection3: 0.540039, colorCorrection4: 0.208008, colorCorrection5: -0.332031, colorCorrection6: -0.212891, colorCorrection7: -0.212891, colorCorrection8: 0.68457, colorCorrection9: 0.930664, colorCorrection10: -0.553711, colorCorrection11: -0.553711, colorCorrection12: 0.0458984
Depth Table:
 depthUnits: 1000, depthClampMin: 0, depthClampMax: 65536, disparityMode: 0, disparityShift: 0
Auto Exposure Control:
 Mean Intensity Set Point: 400
Census:
 uDiameter: 9, vDiameter: 9
Depth Control Min Values:
  minusDecrement: 0, deepSeaMedianThreshold: 0, scoreThreshA: 0, scoreThreshB: 0, textureDifferenceThreshold: 0, textureCountThreshold: 0, deepSeaSecondPeakThreshold: 0, deepSeaNeighborThreshold: 0, lrAgreeThreshold: 0
Depth Control Max Values:
  minusDecrement: 255, deepSeaMedianThreshold: 1023, scoreThreshA: 1023, scoreThreshB: 4095, textureDifferenceThreshold: 4095, textureCountThreshold: 1023, deepSeaSecondPeakThreshold: 1023, deepSeaNeighborThreshold: 1023, lrAgreeThreshold: 2047
After Setting new value, Depth Control:
 minusDecrement: 10, deepSeaMedianThreshold: 500, scoreThreshA: 511, scoreThreshB: 2047, textureDifferenceThreshold: 0, textureCountThreshold: 0, deepSeaSecondPeakThreshold: 325, deepSeaNeighborThreshold: 7, lrAgreeThreshold: 24
Controls as JSON:
 {
    "aux-param-autoexposure-setpoint": "400",
    "aux-param-colorcorrection1": "0.461914",
    "aux-param-colorcorrection10": "-0.553711",
    "aux-param-colorcorrection11": "-0.553711",
    "aux-param-colorcorrection12": "0.0458984",
    "aux-param-colorcorrection2": "0.540039",
    "aux-param-colorcorrection3": "0.540039",
    "aux-param-colorcorrection4": "0.208008",
    "aux-param-colorcorrection5": "-0.332031",
    "aux-param-colorcorrection6": "-0.212891",
    "aux-param-colorcorrection7": "-0.212891",
    "aux-param-colorcorrection8": "0.68457",
    "aux-param-colorcorrection9": "0.930664",
    "aux-param-depthclampmax": "65536",
    "aux-param-depthclampmin": "0",
    "aux-param-disparityshift": "0",
    "controls-autoexposure-auto": "True",
    "controls-autoexposure-manual": "33000",
    "controls-color-autoexposure-auto": "True",
    "controls-color-autoexposure-manual": "156",
    "controls-color-backlight-compensation": "0",
    "controls-color-brightness": "0",
    "controls-color-contrast": "50",
    "controls-color-gain": "64",
    "controls-color-gamma": "300",
    "controls-color-hue": "0",
    "controls-color-power-line-frequency": "3",
    "controls-color-saturation": "64",
    "controls-color-sharpness": "50",
    "controls-color-white-balance-auto": "True",
    "controls-color-white-balance-manual": "4600",
[... 147 lines cut ...]
 "param-scanlinep2onediscon": 507,
 "param-scanlinep2twodiscon": 493,
 "param-secondpeakdelta": 647,
 "param-texturecountthresh": 0,
 "param-texturedifferencethresh": 1722,
 "param-usersm": 1
}
hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
```
Notice how there's quotes around the input in the first set but not the second? I tried adding quotes to the JSON and it failed, so why would there be in the serialized_json()? I tried to look up the definition of the function in https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.rs400_advanced_mode.html?highlight=advanced#pyrealsense2.rs400_advanced_mode.serialize_json but there's no description of what it does.

Could it be an issue with how the input is formatted?

Here I see the same SDK Version (2.29) for C# where the wrapper simply doesn't have access to Advanced Mode instructions. Though the examples available for python and closing of issues suggest it has been.
https://www.gitmemory.com/issue/IntelRealSense/librealsense/5117/545864982

> 
> 
> Is the Python script in the link below for applying the High Accuracy preset any help to you please?
> 
> [#2577 (comment)](https://github.com/IntelRealSense/librealsense/issues/2577#issuecomment-432137634)
> 
> An easy way to define your own custom JSON file is to set up your preferred settings in the Real sense Viewer and then export them as a JSON using an icon on the toolbar at the top of the options side-panel.

I have used this and ran into the exact same issue with `hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."`

hobbitsyfeet, 2019-11-04:
It would probably be useful to note that I tried loading the JSON and the IntelRealsense Viewer cannot load any of the presets again, even after restarting hardware, and the computer it's running on. 

I get a variety of errors including 
```
Backend in rs2_set_option(options:0CF31980, option:Visual Preset, value:3):
hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
```
as well as this when I try to set it again:
```
Backend in rs2_open_multiple(sensor:0CF31980, profiles:0E6D8D38, count:1):
hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
and when I try and return back to Custom preset:
```
When I try and return back to Custom Preset
```
UNKNOWN in rs2_open_multiple(sensor:0CF31980, profiles:0E6D8D38, count:1):
Could not set Media Type. Device may be locked
```

This is very odd because before I tried to save the preset and load it, none of this was happening in the viewer and I could access the presets just fine.

ev-mp, 2019-11-07:
@hobbitsyfeet, can you provide the json you're using and also post the camera info ?
`rs-enumerate-devices -s`

closed 2024-03-29 by MartyG-RealSense