realsenseai/librealsense #5173: How to get the corresponding 3D point of a given RGB pixel?

JimXu1989, 2019-11-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Operating System & Version |   {Win (8.1/10)  |                                     | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<28>.<?> }                          | 
| Language                            |  {python}                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I use D435i, which could align RGB image with depth image, and get the 3D points of the whole view.
In other issue , points.get_vertices() is used to get the 3D points of a ROI, while, is there difference between vertix and 3D points?

And what will happen if I just want 3D point of a given RGB pixel?

ev-mp, 2019-11-04:
@JimXu1989 hello,
`rs2::points` is a multi-plane container comprising of 3D vertices and (optionally) the corresponding texture coordinates. So vertex is a component of that class, unless I misunderstand the question.

>And what will happen if I just want 3D point of a given RGB pixel?

You can use `rs2_project_color_pixel_to_depth_pixel` API to find a corresponding depth pixel, and then convert that 2D pixel into a 3D vertex. #4930

JimXu1989, 2019-11-04:
> @JimXu1989 hello,
> `rs2::points` is a multi-plane container comprising of 3D vertices and (optionally) the corresponding texture coordinates. So vertex is a component of that class, unless I [...]

@ev-mp  Hi, thanks a lot for the reply!! Is rs2_project_color_pixel_to_depth_pixel has a python version??

ev-mp, 2019-11-04:
Currently it is not available in pyrealsense2. Will be fixed in next releases.
You can see the API usage in [unit-testing](https://github.com/IntelRealSense/librealsense/blob/b54f47ba78307be1f337f0b62dcbb4ee1613df34/unit-tests/unit-tests-live.cpp#L5065).

ev-mp, 2019-11-11:
@JimXu1989 , the missing python interface for that  function is being reviewed in the PR linked above. You can download it for evaluation.

JimXu1989, 2019-11-11:
> @JimXu1989 , the missing python interface for that function is being reviewed in the PR linked above. You can download it for evaluation.

Hi, thanks a lot for your reply!
Do you mean rs2_project_color_pixel_to_depth_pixel  is avaiable for  RealSense SDK  v2.30.0?

RealSenseCustomerSupport, 2019-11-21:
----------------------------------------------

Hi @JimXu1989,

You can review below PR which is not yet in librealsense 2.30.0.
https://github.com/IntelRealSense/librealsense/pull/5200/files

ev-mp, 2019-11-21:
The PR has been merged into `development` branch. Will be tagged with the next version, tentatively v2.31.0.

RealSenseCustomerSupport, 2019-12-26:
----------------------------------------------

Hi @JimXu1989,

You can review below PR which has been merged into librealsense 2.31.0.
https://github.com/IntelRealSense/librealsense/pull/5200/files

If there is nothing further, then we can close this thread.

RealSenseCustomerSupport, 2020-03-05:
----------------------------------------------

Hi @JimXu1989,

You can review the latest librealsense 2.33.1.

If there is nothing further, then we can close this thread.
**If we don’t hear from you in 7 days, this issue will be closed.**

closed 2020-03-17 by RealSenseCustomerSupport