realsenseai/librealsense #5174: Cross Compiling librealsense

zainmehdi, 2019-11-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 18.04 - Raspbian Buster |                                  | 
| Platform                                 | Raspberry Pi |                   |                      |              | 

### Issue Description
Is there any guide for cross compiling librealsense for raspbian. I have been trying to cross compile and build my qt project but it fails. I was successfully able to compile qt using [these](https://wiki.qt.io/Raspberry_Pi_Beginners_Guide) tutorials. But realsense isnt compiling well. I get "wrong file format error" Any help would be appreciated. Thanks

neilyoung, 2019-11-04:
There is a pretty good receipt by @patrickpoirier51 here https://github.com/IntelRealSense/librealsense/issues/4408. I personally compiled w/o bigger issues on RPI ZeroW (Raspbian Buster), PI3 (Ubuquity, Ubunutu 16.04), PI3B+ (Ubunutu) and Banana PI M2 Zero. For Raspbian #4408 rules apply, especially this "-latomic" thing. I would not recommend PIZW, since you will loose poses (not to talk about video frames).  The bigger spap file is the main thing you need and some time. It takes 8 hours to compile on PIZW, about 1.5 on Banana and the 3 series PIs. On 3B+ you will even be able to consume fisheye video up to 27 frames per second and poses up to 200 fps.

zainmehdi, 2019-11-04:
@neilyoung  Thanks alot for your response. Actually I was able to compile too on RPi3 successfully. What I am looking for is cross compilation where I can compile it on my host pc and then just run binaries on RPI. I was able to compile qt and I just copy the binaries now and they run well on RPi3 but I am having issues compiling librealsense since it has so many dependencies and I will have to cross compile all those dependencies in order to build it for arm processors. I was wondering if there was some official guide by intel fro this purpose as many people will be using it on embedded platforms and development time is quite alot if one tries to build and compile on them.

neilyoung, 2019-11-04:
Ah, sorry. Then I got that request wrong. Sorry.

dorodnic, 2019-11-11:
Hi @zainmehdi 
We are cross-compiling with Android toolchain in Travis - [.travis.yml](https://github.com/IntelRealSense/librealsense/blob/master/.travis.yml#L87). Maybe this can help. 
I may be mistaken, but the wrong file format for Android is handled by [`set(CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} -fPIE -pie")`](https://github.com/IntelRealSense/librealsense/blob/master/CMake/android_config.cmake#L17)

zainmehdi, 2019-12-03:
@neilyoung Hi. How were you able to get reasonable performance on Rpi zero. I am having a latency of atleast 3-4 seconds even with a swap size of 5gb. Can you provide any suggestions ?

neilyoung, 2019-12-03:
@zainmehdi I wasn't able to get that with the PIZ. That's why I abandoned it and went with a Banana PI Zero M2. The PI Zero was only capable of reaching about 20 fps pose frames (not video frames).

neilyoung, 2019-12-03:
@zainmehdi And btw: The swap is only required during compilation, which took very, very long.

zainmehdi, 2019-12-04:
@neilyoung  Thanks. I recently got banana pi and starting to test with it. Can you tell me which linux version you used. I dont have much time left so I am opting to not experiment with available versions.

neilyoung, 2019-12-04:
@zainmehdi Well, yes, that was an incredibly difficult story (for me). There is a verrrry loooong thread, which pretty much illustrates what I had to go through :)

But in the end it worked. One of the MOST IMPORTANT THINGS IS: Don't really use a "cheap" SD card (whatever cheap is). The Banana PI is very bitchy with "bad" SDs. 

The image I flashed in the end was this:

https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/releases/tag/v15

If you have an Ubuntu system, you should flash there. That ensures to have a really good system on the SD. 

Just informational (don't read it all): Here is the thread. The host there was very, very helpful, so don't hesitate to ask, if you have problems.

https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal/issues/29

closed 2019-12-10 by dorodnic