realsenseai/librealsense #5175: extrinsics from T265 pose to D435

vladHoloArch, 2019-11-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265} | 
| Firmware Version                   | (0.0.18.5715) | 
| Operating System & Version |   Win 10| 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | PC/Android |
| SDK Version                            |  2.23.0                       | 
| Language                            |  C#/unity                        | 
| Segment			|  AR/others                         | 

### Issue Description
Trying to implement AR using t265's pose and d435's rgb colorframe in unity, the pose I get doesn't correspond to 640x480 image received from d435.
The 3d model appears to be following the movement of the camera.
I've applied d435 color stream intrinsics to unity's camera but I have now idea how to coordinate t265's extrinsics with d435's color stream..?

ev-mp, 2019-11-04:
@vladHoloArch , there is a undergoing effort to add such tool to SDK. Follow https://github.com/IntelRealSense/librealsense/pull/4355 for details

ogoshen, 2019-11-04:
First of all, I suggest some sort of rigid mount like this one:
https://www.tinkercad.com/things/0FBAyD8ACOJ

There are some software tools that might help with calibration:
https://github.com/arpg/vicalib
https://github.com/ethz-asl/kalibr

But some measuring and eyeballing might do the trick.

As mentioned in T265's [datasheet](https://dev.intelrealsense.com/docs/tracking-camera-t265-datasheet):
> The center of tracking corresponds to the center location between the right and left imagers on the PCB.

From D400's [datasheet](https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet):

> The depth origin X-Y coordinates is the X-Y center of left imager

And the Z given by distance from front glass:

|Depth Camera|Camera Front Glass (Z’)|
|----------------|-------------------------|
|D415| -1.1mm|
|D435/D435i| -4.2mm|

So what you want to do, is build a hierarchy of transforms in Unity to model the extrinsics between these points.
(Assuming rigidity of the mount, and neglecting rotation differences).

So the parent transform is driven by T265's pose, and Unity's virtual camera is a child with an offset matching the points defined above.

vladHoloArch, 2019-11-04:
besides the extrinsics, could it be an issue of anchoring the 3d model to some spatial plane?

ogoshen, 2019-11-04:
I might have misunderstood the issue a bit.

> The 3d model appears to be following the movement of the camera.

This doesn't sound right... virtual objects should be anchored to the world, not following the camera.

vladHoloArch, 2019-11-04:
not following as if they were the child of the camera but unproportionate to the image from the real world provides by the rgb module from d435.
The setup is really simple, colorframe makes up the background texture using ArBackgroundRenderer and the t265 provides the position (I took the measurements you provided earlier into account prior to).
This rig most definitely does not provide a valid ar experience, as you can see I'm clueless about AR but as far a I understand the t265 should be sufficient and no anchoring is needed inside unity as there's no SLAM or any tracking done on unity's side, right?
I'm probably wrong but if I'm not, what am I missing?

ogoshen, 2019-11-04:
Right, so the setup sounds good to me.
Sounds like this might be a scaling issue. Your units should be in meters (the default Unity cube is 1m³) and taken into consideration both in the extrinsics and in objects in the scene.

vladHoloArch, 2019-11-04:
Right, so if the scaling is correct and unity's camera projection matrix = d435s rgb projection matrix.
Shouldn't the SLAM positioning be scaled too?

ogoshen, 2019-11-05:
No, pose is transformed from device coordinate system to Unity's here: 
https://github.com/IntelRealSense/librealsense/blob/81d469db173dd682d3bada9bd7c7570db0f7cf76/wrappers/unity/Assets/RealSenseSDK2.0/Scripts/RsPoseStreamTransformer.cs#L81
And should not be scaled, positional units are in meters.

RealSenseCustomerSupport, 2019-12-26:
----------------------------------------------

Is there anything further needed on this thread?

closed 2020-02-03 by RealSenseCustomerSupport

AndreV84, 2020-08-24:
how to modify it in a way so that it works with L515 instead of D4xx?