realsenseai/librealsense #5176: setting "RS2_OPTION_ENABLE_MOTION_CORRECTION" off after D435i imu calibration

jaehwan-pi, 2019-11-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   |  05.11.06.250) | 
| Operating System & Version |   Ubuntu 16.04 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.29.0                       | 
| Language                            |  {C }                          | 
| Segment			|  AR                          | 

### Issue Description
Hi, I'm trying to set an option "RS2_OPTION_ENABLE_MOTION_CORRECTION" OFF after D435i imu calibration.

In my code, I'm trying to do something like below.

std::vector<rs2::pipeline>  pipelines;
    rs2::context                ctx;            // Create librealsense context for managing devices
    cv::Mat intrinsic = cv::Mat::zeros(3,3, CV_64F);
    cv::Mat distCoeffs = cv::Mat::zeros(1,4, CV_64F);
    for (auto&& dev : ctx.query_devices())
    {
        rs2::pipeline pipe(ctx);
        rs2::config cfg;
        cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
        std::string devName= dev.get_info(RS2_CAMERA_INFO_NAME);

        if(!devName.compare("Intel RealSense D435I"))
        {
            std::cout<<dev.get_info(RS2_CAMERA_INFO_NAME) << " detected." << std::endl;
            cfg.enable_stream(RS2_STREAM_COLOR, 848, 480, RS2_FORMAT_BGR8, 30);
            cfg.enable_stream(RS2_STREAM_ACCEL, RS2_FORMAT_MOTION_XYZ32F, 250);
            cfg.enable_stream(RS2_STREAM_GYRO, RS2_FORMAT_MOTION_XYZ32F, 400);

            auto sensor = dev.query_sensors().front();
            sensor.set_option(RS2_OPTION_ENABLE_MOTION_CORRECTION, 0);

        }
        rs2::pipeline_profile profiles = pipe.start(cfg, callback);
        pipelines.emplace_back(pipe);
}

However the set_option function gives me the error message like below.

RealSense error calling rs2_set_option(options:0x27af010, option:Enable Motion Correction, value:0):
    object doesn't support option #29

How can I set this option correctly?

RealSenseCustomerSupport, 2019-11-15:
----------------------------------------------

Hi @jaehwan-pi,

You can refer to rs-sensor-control sample app to see how to configure the IMU option.
However, I can see another issue to control this. Will further check internally.

RealSenseCustomerSupport, 2019-12-23:
----------------------------------------------

Hi @jaehwan-pi

I've had a look at the latest releases of our librealsense as well as latest FW for the D435i and am not coming across the issue that you presented.  The "RS2_OPTION_ENABLE_MOTION_CORRECTION" is a SW setting.  Can you outline any further specifics to your situation or let us know if you're still seeing the problem?

Thanks

jaehwan-pi, 2019-12-23:
Thank you for the comment. Actually I raised this issue to deal the issue #5177. So if the issue #5177 is solved, this issue is no longer needed. I will close this issue, so please provide further information on #5177 if possible. Thank you.

closed 2019-12-23 by jaehwan-pi