realsenseai/librealsense #5178: T265: How to obtain pitch, roll and yaw in C++?

neilyoung, 2019-11-04:
There is a nice sample in python which shows, how to determine the Euler angles from the pose. Is there something similar for the C++ world or should I use the gyro stream directly, as shown in the `rs-motion` sample? What would you suggest?

MartyG-RealSense, 2019-11-04:
The subject of getting roll pitch and yaw from T265 is discussed in the link below.  Please read downwards from the linked-to comment to find user-contributed scripting on the subject.

https://github.com/IntelRealSense/librealsense/issues/3129

neilyoung, 2019-11-04:
@MartyG-RealSense Right, thanks. This is a pretty nice reference, thanks for providing it.

neilyoung, 2019-11-04:
But if I see atan2 I suspect an evil gimbal lock waiting around the corner :)) Maybe I'm not entirely through this thread... Reading on.

neilyoung, 2019-11-04:
Evaluated. Works. Thanks

MartyG-RealSense, 2019-11-04:
Awesome news - thanks so much for letting us know!

neilyoung, 2019-11-04:
@MartyG-RealSense Sorry, but I would like to take the opportunity to ask you something, maybe you have a clue. I was asking the a.m. question, because I would like to know the _yaw_ Euler angle, at which the AprilTag is detected. For this I was altering the `rs-pose-apriltag` sample. I checked the #3129 algorithm with the quaternion delivered in the camera pose and the yaw value is always plausible and follows my rotation around the camera's y axis (z in Euler's world). 

But doing a similar conversion with the derived 'world' frame (see also #5184) shows a little different result. Even though I found, that my assumption of having a more or less "stable" transformation matrix of the tag after detection, the derived yaw is not as expected. It is always about 45 degrees "off" and I don't really know, why. I suspect it could have to do with the fisheye distortion or with the FOV.

For example: Having the tag straight ahead gives a camera pose yaw of 0 (or 360) degrees as expected at start of the script. I would now also expect to see 360 for the tag's yaw, but I get 315. Same, if I have the tag in my back and translate/rotate a bit in order to get the tag then. The value is not 180 as expected, but 225. 

Any idea what that could be? Or am I on a completely wrong path?

Any pointer appreciated.
TIA

MartyG-RealSense, 2019-11-04:
This particular question is outside of my knowledge, unfortunately.  One of the Intel RealSense team members on this forum should be able to give good advice on this subject.  I do apologize.

neilyoung, 2019-11-04:
No issue. Thanks nevertheless for your efforts.

neilyoung, 2019-11-05:
@MartyG-RealSense I think I have to revert my assessment regarding the correctness of the supplied functions to determine pitch, roll and yaw from the pose's quaternion. I think the algorithm in the comment you quoted above seemed to have been correct, since I only checked the yaw value. Today I checked the rest and this is my version now:

```
    auto w = camera_pose->rotation.w;
    auto x = -camera_pose->rotation.z;
    auto y = camera_pose->rotation.x;
    auto z = -camera_pose->rotation.y;

    auto pitch =  -asin(2.0 * (x*z - w*y)) * 180.0 / M_PI;
    auto roll  =  atan2(2.0 * (w*x + y*z), w*w - x*x - y*y + z*z) * 180.0 / M_PI;
    auto yaw   =  atan2(2.0 * (w*z + x*y), w*w + x*x - y*y - z*z) * 180.0 / M_PI;

    std::stringstream ss;
    ss << "frame " << fn << " Pitch: " << pitch << " Roll: " << roll << " Yaw: " << yaw;
    std::cout << ss.str() << std::endl;
```

Please check my math, if you like. I'm referring to this paper. http://www.chrobotics.com/library/understanding-euler-angles

Looking at the air plane there you see, that in NED (north-east-down) or Euler coordinates, x is the north, y the east and z the down coordinate. 

In contrary this is the coordinate system of the T265, if pointing into the same direction as the air plane:

[image]

That again makes me think to apply this analogy:

|T265 Camera Pose  |              Euler/World/NED|
|--------------------|--------------------------|
|x                                  |       -z              |
|y                                  |      x                |
|z                                  |    -y |

Given this and the formulas for pitch, roll and yaw from the CH paper above (where they are using a, b, c and d instead of w, x, y and z) I came to the algorithm above.

The algorithm produces `yaw` with values between 0 and 360 degrees, if rotating right around the `y` axis of a forward pointing camera. Looks ok. It produces `pitch` values between 0 to 90 degrees, if pointing up and 0 to -90 degrees, if pointing down, rotating around `x` axis of the camera. Looks good to me too. And it produces `roll` values between 0 and 90 degrees, if rotating right around the `z` axis of the camera and 0 to -90 degrees, if rotating around z left-wards. All looks fine to me.

Does that make sense at all?

MartyG-RealSense, 2019-11-05:
I am not a T265 specialist, so I cannot offer an educated opinion on your figures, unfortunately. 
 @radfordi - a T265 expert - may be able to provide verification of your conclusions.

neilyoung, 2019-11-05:
@MartyG-RealSense Thanks nevertheless. You are the "good spirit" of this board, so sorry for bothering :)

MartyG-RealSense, 2019-11-05:
It's no bother at all, I promise.  No one person can know everything about every subject, so having a mix of people to call on helps to cover as many different bases as possible.  the important thing is knowing which person may hold which knowledge.   :)   Good luck!

neilyoung, 2019-11-05:
That alone is absolutely helpful. Keep on doing your great job.

neilyoung, 2019-11-06:
Update and correction to my post above: The `yaw` value goes from 0 to 180 and -180 to 0 again while rotating right. So ok. Any confirmation?

neilyoung, 2019-11-06:
OK, no further confirmation necessary. My algorithm is right. It gives exactly the same values, if I alter the `t265_rpy.py` example and adding these lines at the end of the `for` loop:

```
            w = data.rotation.w
            x = -data.rotation.z
            y = data.rotation.x
            z = -data.rotation.y

            pitch =  -m.asin(2.0 * (x*z - w*y)) * 180.0 / m.pi;
            roll  =  m.atan2(2.0 * (w*x + y*z), w*w - x*x - y*y + z*z) * 180.0 / m.pi;
            yaw   =  m.atan2(2.0 * (w*z + x*y), w*w + x*x - y*y - z*z) * 180.0 / m.pi;

            print("RPY * [deg]:[ {0:.7f}, {1:.7f}, {2:.7f}]".format(roll, pitch, yaw))

```

So instead of the some opaque matrix witchcraft (no attack) a maybe more easier to understand approach. 

q.e.d.

neilyoung, 2019-11-06:
Created a PR to the development branch. Added a new python sample. https://github.com/IntelRealSense/librealsense/pull/5198

MartyG-RealSense, 2019-11-06:
Thanks so much for your dedication to the community!

neilyoung, 2019-11-06:
You're welcome. I use this to gain more and more knowledge about this great technology. And since the support is that superior, it's time to pay back (or try at least) :))

neilyoung, 2019-11-08:
@MartyG-RealSense OT question: Do you offer kind of payed support? I'm having a use case idea which I would like to discuss with one of your experts (not in this board), since there are some open questions to me.

MartyG-RealSense, 2019-11-08:
Whilst there is an advanced tier of private RealSense support, it is usually only available to customers who will bring a significant level of 'return on investment' to Intel.  An example would be the intention to place an order for hundreds of thousands of RealSense cameras or camera boards.  Otherwise, this forum is the best place for T265 questions.

neilyoung, 2019-11-08:
Makes sense. Thanks

RealSenseCustomerSupport, 2019-11-21:
----------------------------------------------

Hi Neil. Would you say this particular ticket is resolved and can be closed?

radfordi, 2019-11-21:
@neilyoung provided his code in a now merged PR #5198.  Thanks @neilyoung!  After his removal of the library code we used, the conversion to C++ is nearly a direct copy of the python.  I'd say this is closed.

closed 2019-11-21 by radfordi

neilyoung, 2019-11-21:
Sorry, too late. Yes, absolutely. Can be closed :)

rakshith95, 2020-03-23:
Hello, didn't want to open a new issue for this, but is there any way to get the yaw heading from the T265 pose data, or will I have to get it by accessing the magnetometer?

neilyoung, 2020-03-23:
The above example shows how to obtain the yaw of the camera. But this will always be the yaw w.r.t. the current reference frame of the cam. That means: If you once start the cam pointing to geographical north, the yaw will be 0 then. If you do the same w/o special precautions then the result will also be 0 if you point to geographical south. A possible solution is to start mapping from a known geographical direction and store a reference point in the map. Once you are reloading the map you will get back the stored reference - now in the current reference frame of the cam - and by knowing the original heading you can calculate the geographical direction again. All shown and prepared menu like here https://neilyoung.serveblog.net

rakshith95, 2020-03-23:
> If you do the same w/o special precautions then the result will also be 0 if you point to geographical south.
> 
This is what I wanted to avoid/account for.

> A possible solution is to start mapping from a known geographical direction and store a reference point in the map. Once you are reloading the map you will get back the stored reference - now in the [...]
Is this possible with the T265? How do I go about doing this?

neilyoung, 2020-03-23:
It is all published https://neilyoung.serveblog.net. Just read, download the code. 

You don't even have to perfectly geo-reference your floor plan. 
- Draw a true to scale rectangle on a piece of paper and make a photo. You have your floor plan. 
- Go to the page linked above, "GEO-REFERENCING", stich your floor plan onto the map at (0,0), draw a rectangle around it, provide me the real world dimensions. 
- Place a start marker onto the map, let it point into the direction, into which you start mapping. 
- Download the python scripts, install and start it. 
- Remote control your camera from "REMOTE".

rakshith95, 2020-03-23:
Okay, thanks a lot. I'm planning on using it on a drone, so for now I think I'll take the drone heading, and use that

rakshith95, 2020-03-25:
Update in case someone has a similar necessity to obtain the heading: The T265 uses the [BMI055](https://www.bosch-sensortec.com/products/motion-sensors/imus/bmi055.html) IMU module, which does not have a magnetometer.

neilyoung, 2020-03-25:
Nice that you mention it, but that's not a news :)